thesofproject/sof #2690: [BUG][CML]"error: ipc error for 0x60010000 size 20" happened after booting up successfully.

sinahuang, 2020-04-02:
**Describe the bug**
Dmesg shows "error: ipc error for 0x60010000 size 20" after booting up successfully.

**To Reproduce**
1.Boot up system successfully
2.Check the dmesg.

**Reproduction Rate**
5/5

**Expected behavior**
Ipc error should't  happen after boot up successfully.

**Impact**
"error: ipc error for 0x60010000 size 20" happened after booting up successfully.
```
[    4.232752] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 2 dir 0
[    4.232753] sof-audio-pci 0000:00:1f.3: generating page table for 000000001426818a size 0x4b00 pages 5
[    4.232773] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[    4.232798] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[    4.232801] sof-audio-pci 0000:00:1f.3: period_bytes:0x12c0
[    4.232801] sof-audio-pci 0000:00:1f.3: periods:4
[    4.232818] sof-audio-pci 0000:00:1f.3: stream_tag 1
[    4.232821] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[    4.233341] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.233342] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[    4.233342] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[    4.233343]  SDW1-speakers: ASoC: hw_params FE failed -22
[    4.233344] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[    4.233345] intel-master sdw-master-1: sdw_deprepare_stream: failed -22
[    4.233345] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[    4.233346] intel-master sdw-master-2: sdw_deprepare_stream: failed -22
[    4.233347] sof-audio-pci 0000:00:1f.3: pcm: free stream 2 dir 0
[    4.233349] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[    4.233350] intel-master sdw-master-1: sdw_deprepare_stream: failed -22
[    4.233350] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[    4.233351] intel-master sdw-master-2: sdw_deprepare_stream: failed -22
[    4.233379] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.233591] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.233624] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.233883] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[    4.233886] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 2 dir 0
[    4.233887] sof-audio-pci 0000:00:1f.3: generating page table for 0000000086468a0e size 0x4b00 pages 5
[    4.233895] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
[    4.233915] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[    4.233917] sof-audio-pci 0000:00:1f.3: period_bytes:0x12c0

```

**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev  commit:b8464d14
FW: https://github.com/thesofproject/sof/commits/master commit:  5452ac5
Tplg: Same with FW branch,  sof-cml-rt711-rt1308-rt715.tplg
Platform:CML-H Laptop with codec ALC711, 2 x RT1308, RT715 in SDW mode.

[dmesg02.log](https://github.com/thesofproject/sof/files/4418817/dmesg02.log)

sinahuang, 2020-04-02:
Bisect result:
Caused by commit: https://github.com/thesofproject/sof/commit/f755277000b56a4a240809ba1a6bb52288c4895d
@libinyang , please help to check it.

sinahuang, 2020-04-02:
On ubuntu 20.04, error also can be reproduced if  ctrl+ c to stop any piple's playback.

sinahuang, 2020-04-07:
Checked this  issue with master(commit:b4e8b1e4) + #2954 and topic/sof-dev(commit:052e45d) + #1982 + #1943, issue can't be reproduced.

bardliao, 2020-04-07:
> Error can't be reproduced after two 1308 sync playback enabled. Tried with master(commit:b4e8b1e4) + #2594 
> 
> and topic/sof-dev(commit:052e45d)
> + [thesofproject/linux#1982](https://github.com/thesofproject/linux/pull/1982)
> + [thesofproject/linux#1943](https://github.com/thesofproject/linux/pull/1943)

So this should be a alh data param issue which will be fixed by  [thesofproject/linux#1982](https://github.com/thesofproject/linux/pull/1982) and [thesofproject/linux#1943](https://github.com/thesofproject/linux/pull/1943)

sinahuang, 2020-04-29:
Issue can't be seen with sof-dev(commit：8f1c6ad) +PR #2058 (note :PR2058 is for other blocked issue #2031) and master (commit:af38b20b), closed it.

closed 2020-04-29 by sinahuang