thesofproject/sof #2693: [BUG][APL-nocodec]"error: ipc error for 0x80010000 size 12" when do check-kmod-load-unload-after-playback.sh test

Liviali155, 2020-04-02:
**Describe the bug**
"error: ipc error for 0x80010000 size 12" when do check-kmod-load-unload-after-playback.sh test
After error occured,all the pipeline can not  work,reboot can recovery

**To Reproduce**
1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg
5../check-kmod-load-unload-after-playback.sh -l 50

**Reproduction Rate**
2/2

**Expected behavior**
There is o error occured when do check-kmod-load-unload-after-playback.sh test

**Impact**
Tried 2 times
1."Unable to install hw params" when aplay(0,3),dmesg show "error: ipc error for 0x80010000 size 12"
[image]
2.Caught error in dmesg at 36th when aplay via(0,5) ,error message is "error: ipc error for 0x80010000 size 12"
[image]

```
ubuntu@jf-apl-up2-nocodec-2:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofnocodec [sof-nocodec], device 0: Port0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 1: Port1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 3: Port3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 5: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 7: Media Playback 7 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

<details>
 <summary>dmesg_error.log</summary>

````
nel: [ 4573.186634] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
jf-apl-up2-nocodec-2 kernel: [ 4573.186640] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
jf-apl-up2-nocodec-2 kernel: [ 4573.186752] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
jf-apl-up2-nocodec-2 kernel: [ 4573.186758] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
jf-apl-up2-nocodec-2 kernel: [ 4573.186873] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x80010000 size 12
jf-apl-up2-nocodec-2 kernel: [ 4573.186881] sof-audio-pci 0000:00:0e.0: error: failed to set dai config for NoCodec-1
jf-apl-up2-nocodec-2 kernel: [ 4573.186885] sof-audio-pci 0000:00:0e.0: error: failed to restore pipeline after resume -22
jf-apl-up2-nocodec-2 kernel: [ 4573.186903] sof-audio-pci 0000:00:0e.0: pcm: open stream 3 dir 0
jf-apl-up2-nocodec-2 kernel: [ 4573.186906] sof-audio-pci 0000:00:0e.0: period min 192 max 16384 bytes
jf-apl-up2-nocodec-2 kernel: [ 4573.186907] sof-audio-pci 0000:00:0e.0: period count 2 max 16
jf-apl-up2-nocodec-2 kernel: [ 4573.186909] sof-audio-pci 0000:00:0e.0: buffer max 65536 bytes
jf-apl-up2-nocodec-2 kernel: [ 4573.187161] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
jf-apl-up2-nocodec-2 kernel: [ 4573.187164] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
jf-apl-up2-nocodec-2 kernel: [ 4573.187169] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
jf-apl-up2-nocodec-2 kernel: [ 4573.187170] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
jf-apl-up2-nocodec-2 kernel: [ 4573.187173] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
jf-apl-up2-nocodec-2 kernel: [ 4573.187174] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
jf-apl-up2-nocodec-2 kernel: [ 4573.187179] sof-audio-pci 0000:00:0e.0: pcm: hw params stream 3 dir 0
jf-apl-up2-nocodec-2 kernel: [ 4573.187183] sof-audio-pci 0000:00:0e.0: generating page table for 000000002f0aaa54 size 0xffc0 pages 16
jf-apl-up2-nocodec-2 kernel: [ 4573.187190] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
jf-apl-up2-nocodec-2 kernel: [ 4573.187205] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
jf-apl-up2-nocodec-2 kernel: [ 4573.187209] sof-audio-pci 0000:00:0e.0: period_bytes:0x3fc0
jf-apl-up2-nocodec-2 kernel: [ 4573.187210] sof-audio-pci 0000:00:0e.0: periods:4
jf-apl-up2-nocodec-2 kernel: [ 4573.187219] sof-audio-pci 0000:00:0e.0: stream_tag 1
jf-apl-up2-nocodec-2 kernel: [ 4573.187226] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
jf-apl-up2-nocodec-2 kernel: [ 4573.187279] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 20
jf-apl-up2-nocodec-2 kernel: [ 4573.187286] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 1
jf-apl-up2-nocodec-2 kernel: [ 4573.187290] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -22
jf-apl-up2-nocodec-2 kernel: [ 4573.187294]  Port3: ASoC: hw_params FE failed -22
jf-apl-up2-nocodec-2 kernel: [ 4573.187299] sof-audio-pci 0000:00:0e.0: pcm: free stream 3 dir 0
````
</details>

No sof error trace

**Environment**
Platform: APL UP2 with nocodec
Firmware: 5452ac56 https://github.com/thesofproject/sof/commits/master
Kernel: b8464d14 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: sof-apl-nocodec.tplg same as firmware

[dmesg.txt](https://github.com/thesofproject/sof/files/4420594/dmesg.txt)
[etrace.txt](https://github.com/thesofproject/sof/files/4420596/etrace.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4420597/slogger.txt)

Liviali155, 2020-04-07:
Changed firmware to 20200318(a133d5d5),issue can not be reproduced
Changed firmware to 20200330(1ce07dcf),issue can be reproduced

Liviali155, 2020-04-08:
Bisect result:Caused by firmware commit https://github.com/thesofproject/sof/commit/3073aa5fc4dbe3c5b42b3cca84692fd64e15f73a

@tlauda  please check it

Liviali155, 2020-04-13:
This issue also can be reproduced on sof v1.5-rc1
Kernel:4b4681eb Sof-dev
fw:5f7d4e2 https://github.com/thesofproject/sof/releases/tag/v1.5-rc1

mengdonglin, 2020-04-14:
Change priority to P1 since it blocks following SOF Linux CI tests

tlauda, 2020-04-14:
> Bisect result:Caused by firmware commit [3073aa5](https://github.com/thesofproject/sof/commit/3073aa5fc4dbe3c5b42b3cca84692fd64e15f73a)
> 
> @tlauda please check it

Sorry, but this piece of code is only executed when slave cores are active. APL is not even build with SMP enabled, so idc.c is compiled out. Your bisect is wrong.

Liviali155, 2020-04-15:
I am sorry,I will recheck it

Liviali155, 2020-04-17:
@tlauda  I recheck it
Seems the issue caused by https://github.com/thesofproject/sof/commit/1a01f7490f7b79834de081f214d329e4ce93442e

Tried on https://github.com/thesofproject/sof/commit/bc5f3a6cde0e253a8141f5b53571192c40d69818 8times, pass

tlauda, 2020-04-22:
@Liviali155 Still impossible. This commit just adds new function. Please verify again your bisect.

paulstelian97, 2020-04-22:
IPC errors seem to be heisenbugs... I think someone needs to decode the 0x80010000 message and see what IPC it is.

slawblauciak, 2020-04-28:
@paulstelian97 that's a DAI config message.

Liviali155, 2020-04-30:
Tried on https://github.com/thesofproject/sof/commit/6713d90c21004f45ce5e25708087b8c77a78ff2a 500 times pass
https://github.com/thesofproject/sof/commit/4e87899f25b0c859169c1dfd4dd9f9fd65d48ad2 failed

@ktrzcinx could you please help check it ,seems it is a regression caused by https://github.com/thesofproject/sof/commit/4e87899f25b0c859169c1dfd4dd9f9fd65d48ad2

ktrzcinx, 2020-04-30:
@Liviali155 https://github.com/thesofproject/sof/commit/4e87899f25b0c859169c1dfd4dd9f9fd65d48ad2 doesn't change sof behavior at all, it's just simple change in sof build system

Liviali155, 2020-05-09:
On lastest sof-dev(d43dc16c)+master(782066e),error "error: ipc error for 0x80010000 size 12" not seen ,but has another error "error: ipc timed out for 0x80010000 size 216",have creat another issue https://github.com/thesofproject/sof/issues/2921 to check,closed this issue

YvonneYang2, 2020-05-20:
Issue can be reproduced with latest sof-dev(commit:3914cd29) and master(commit: d9e3d4d4) on apl nocodec and apl I2S PCM512x platforms.

Reopen it.

tlauda, 2020-05-26:
@ranj063 @kv2019i I've noticed something weird in FW log (slogger.txt). After freeing and recreating the whole topology again every DAI config IPC is sent twice instead of once. Also IPC parameters for the faulty IPC are totally messed up. Could take a look into the kernel code?

tlauda, 2020-05-28:
@YvonneYang2 Could you check with this patch https://github.com/tlauda/sof/commit/e3f9299b4fa4af5ed0d003b18f4fbd032e851a36 and let me know whether it helps.

Liviali155, 2020-05-29:
> @YvonneYang2 Could you check with this patch [tlauda@e3f9299](https://github.com/tlauda/sof/commit/e3f9299b4fa4af5ed0d003b18f4fbd032e851a36) and let me know whether it helps.

Tried with  [tlauda@e3f9299](https://github.com/tlauda/sof/commit/e3f9299b4fa4af5ed0d003b18f4fbd032e851a36)
1st:./check-kmod-load-unload-after-playback.sh -l 1000,error not seen
2nd:./check-kmod-load-unload-after-playback.sh -l 1000,error not seen

kv2019i, 2020-05-29:
@tlauda wrote:
> @ranj063 @kv2019i I've noticed something weird in FW log (slogger.txt). After freeing and recreating the whole topology again every DAI config IPC is sent twice instead of once. Also IPC parameters for the faulty IPC are totally messed up. Could take a look into the kernel code?

Interesting. I cross-checked the matching attached dmesg and kernel is not sending the duplicated IPCs -> FW logs show more DAI_CONFIG IPCs than what kernel is sending.  I wonder how the stack corruption could be hit in this case... it's the same case repeated many times after all.

tlauda, 2020-05-29:
@kv2019i These multiple DAI_CONFIG IPCs are not directly responsible for stack corruption, but probably increase the probability of happening. In the past I've already observed stack being exceeded during DAI config execution and it seems that on cAVS 1.5 based platforms we need even more memory.
However it's interesting observation, that FW receives multiple IPCs and I'm pretty sure they need to somehow come from the kernel. @Liviali155 Could you capture dmesg from the test on some other platform e.g. ICL or CNL so we could compare whether IPCs there are also duplicated?

Liviali155, 2020-06-01:
> @kv2019i These multiple DAI_CONFIG IPCs are not directly responsible for stack corruption, but probably increase the probability of happening. In the past I've already observed stack being exceeded during DAI config execution and it seems that on cAVS 1.5 based platforms we need even more memory.
> However it's interesting observation, that FW receives multiple IPCs and I'm pretty sure they need to somehow come from the kernel. @Liviali155 Could you capture dmesg from the test on some other platform e.g. ICL or CNL so we could compare whether IPCs there are also duplicated?
> 
> Edit: Note that there is also a bug in the FW, where DAI config for SSP is set two times, but it's done in terms of one IPC execution (fix in #3001). Here there are clearly two IPCs for every config.

I tested on ICL,dmesg as below:
[dmesg.txt](https://github.com/thesofproject/sof/files/4709429/dmesg.txt)

Liviali155, 2020-06-01:
Tried with PR #2999,run "./check-kmod-load-unload-after-playback.sh -l 1000",error not seen

slawblauciak, 2020-06-01:
@Liviali155 that PR has been merged. Good to close?

Liviali155, 2020-06-02:
Closed for PR #2999 merged

closed 2020-06-02 by Liviali155