thesofproject/sof #2700: [BUG][CML]"error: ipc timed out for 0x60040000 size 12" after bootup

Liviali155, 2020-04-03:
**Describe the bug**
"error: ipc timed out for 0x60040000 size 12" after bootup,all the pipeline can not work

**To Reproduce**
1.sudo reboot
2."dmesg -w" to check dmesg

**Reproduction Rate**
10/10

**Expected behavior**
There is no error message in dmesg,all the pipeline can work

**Impact**
"error: ipc timed out for 0x60040000 size 12" after bootup,all the pipeline can not work

<details>
 <summary>sof_errortrace.log</summary>

````
   TIMESTAMP              DELTA C# COMPONENT          LOCATION                       CONTENT
[  5050198.958333] (  5050199.000000) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6080 ticks
[  5050262.656250] (       63.697918) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30072
[  5051762.812500] (     1500.156250) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 251 uS 6026 ticks
[  5051826.354167] (       63.541668) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30023
[  5053329.843750] (     1503.489624) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6086 ticks
[  5053394.010417] (       64.166664) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30102
[  5054894.062500] (     1500.052124) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 251 uS 6034 ticks
[  5054958.333333] (       64.270836) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30032
[  5056466.041667] (     1507.708374) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 257 uS 6182 ticks
[  5056530.572917] (       64.531250) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30187
[  5058033.281250] (     1502.708374) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6090 ticks
[  5058096.875000] (       63.593750) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30073
[  5059600.156250] (     1503.281250) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6085 ticks
[  5059664.166667] (       64.010414) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30092
[  5061163.802083] (     1499.635376) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 251 uS 6025 ticks
[  5061227.447917] (       63.645832) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30015
[  5062724.010417] (     1496.562500) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 247 uS 5949 ticks
[  5062787.552083] (       63.541668) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 29954
[  5064292.291667] (     1504.739624) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 254 uS 6115 ticks
[  5064355.833333] (       63.541668) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30111
[  5065869.531250] (     1513.697876) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 261 uS 6286 ticks
[  5065933.802083] (       64.270836) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30300
[  5067438.072917] (     1504.270874) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 254 uS 6113 ticks
[  5067502.395833] (       64.322914) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30114
[  5069010.312500] (     1507.916626) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 257 uS 6183 ticks
[  5069074.687500] (       64.375000) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30188
[  5070575.572917] (     1500.885376) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 252 uS 6060 ticks
[  5070639.010417] (       63.437500) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30038
[  5072144.322917] (     1505.312500) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 254 uS 6117 ticks
[  5072208.229167] (       63.906250) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30126
[  5073711.145833] (     1502.916626) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6082 ticks
[  5073774.947917] (       63.802082) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30084
[  5075278.229167] (     1503.281250) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6089 ticks
[  5075341.927083] (       63.697918) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30083
[  5076839.895833] (     1497.968750) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 249 uS 5984 ticks
[  5076903.437500] (       63.541668) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 29981
[  5078408.750000] (     1505.312500) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 255 uS 6121 ticks
[  5078473.020833] (       64.270836) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30139
[  5079971.927083] (     1498.906250) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 250 uS 6014 ticks
[  5080036.041667] (       64.114586) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30007
[  5081543.437500] (     1507.395874) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 257 uS 6173 ticks
[  5081608.125000] (       64.687500) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30184
[  5083110.052083] (     1501.927124) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6078 ticks
[  5083173.281250] (       63.229168) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30054
[  5084676.875000] (     1503.593750) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6084 ticks
[  5084740.520833] (       63.645832) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30088
[  5086244.166667] (     1503.645874) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 253 uS 6095 ticks
[  5086308.020833] (       63.854168) c0 SA                          src/lib/agent.c:67    WARN validate(), ll drift detected, delta = 30096
[  5087806.197917] (     1498.177124) c0 SCHEDULE_LL        /schedule/timer_domain.c:35    ERROR timer_report_delay(): timer 3 delayed by 249 uS 5984 ticks
[... 72 lines cut ...]
[    4.265235] sof-audio-pci 0000:00:1f.3: error: can't enter idle
[    4.265236] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[    4.265236] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:62
[    4.265238] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[    4.265239] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060325 SAR     0x00000000
[    4.265240] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe02f676 EPC3     0x00000000 EPC4    0x00000000
[    4.265241] sof-audio-pci 0000:00:1f.3: EPC5     0xbe02f676 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[... cut ...]
rename sof-cnl.ri to sof-cml.ri
tplg: [sof-hda-generic-2ch.tplg ]
Platform:[CML laptop with onboard codec ALC3204 in HDA mode]
PalseAudio Version: [13.99.1]
UCM2:https://github.com/thesofproject/alsa-ucm-conf/tree/master/ucm2/sof-hda-dsp
OS:ubuntu 20.04
[dmesg0403.log](https://github.com/thesofproject/sof/files/4426135/dmesg0403.log)
[sof_errortrace0403.log](https://github.com/thesofproject/sof/files/4426136/sof_errortrace0403.log)
[sof_trace0403.log](https://github.com/thesofproject/sof/files/4426138/sof_trace0403.log)

Liviali155, 2020-04-03:
Error not seen on ubuntu 18.04

Liviali155, 2020-04-03:
Related with https://github.com/thesofproject/sof/issues/2702

lgirdwood, 2020-04-03:
I'm suspecting this is related to Pulseadio on ubuntu 20.04 probing the PCM devices and the FW is getting in a mess. I guess this is not seen if you disable PA on Ubuntu 20.04 ?

Liviali155, 2020-04-03:
Bisect result:Caused by firmware commit https://github.com/thesofproject/sof/commit/5550a5f6768d52b9793f964e5bdd522452012db4

@tlauda  please check it

Liviali155, 2020-04-03:
> I'm suspecting this is related to Pulseadio on ubuntu 20.04 probing the PCM devices and the FW is getting in a mess. I guess this is not seen if you disable PA on Ubuntu 20.04 ?

Yes,after disable PA,error not seen

Liviali155, 2020-04-07:
Error not seen on latest sof/dev/ +master，closed it
Platform: CML laptop with onboard codec ALC3204 in HDA mode
Firmware: b4e8b1e4 https://github.com/thesofproject/sof/commits/master
Kernel: 052e45d7 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-hda-generic-2ch.tplg same as firmware

closed 2020-04-07 by Liviali155