thesofproject/sof #2712: [BUG] HP EliteBook 840 G6 SoF errors

leezu, 2020-04-04:
**Describe the bug**
On HP EliteBook 840 G6 with kernel 5.6.2 and linux-firmware 20200316, dmesg shows many sof related errors and the microphone does not work.

**To Reproduce**
Get a HP EliteBook 840 G6?

**Reproduction Rate**
Happens every time

**Expected behavior**
No errors in dmesg. Microphone works.

**Impact**
Microphone doesn't work.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.6.2 stable release
    * SOF: Not sure
2) Name of the topology file
    * Topology: dmesg says "Topology: ABI 3:11:0 Kernel ABI 3:12:0"
3) Name of the platform(s) on which the bug is observed.
    * Platform: HP EliteBook 840 G6

**Screenshots or console output**
```
[   35.313285] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   35.313683] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   35.366351] intel_rapl_common: Found RAPL domain package
[   35.366355] intel_rapl_common: Found RAPL domain core
[   35.366356] intel_rapl_common: Found RAPL domain uncore
[   35.366358] intel_rapl_common: Found RAPL domain dram
[   35.396313] iwlwifi 0000:3a:00.0: base HW address: d0:ab:d5:90:36:8e
[   35.410818] thermal thermal_zone13: failed to read out thermal zone (-61)
[   35.413063] iwlwifi 0000:3a:00.0 wlp58s0: renamed from wlan0
[   35.426509] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   35.426868] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   35.444869] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   35.445234] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   35.445461] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   35.445611] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   35.449878] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   35.483409] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   35.483411] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   35.486980] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   35.486983] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   35.489155] HDMI HDA Codec ehdaudio0D2: Max dais supported: 3
[   35.516396] input: SYNA3091:00 06CB:82F5 Mouse as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-8/i2c-SYNA3091:00/0018:06CB:82F5.0003/input/input17
[   35.516549] input: SYNA3091:00 06CB:82F5 Touchpad as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-8/i2c-SYNA3091:00/0018:06CB:82F5.0003/input/input18
[   35.516652] hid-multitouch 0018:06CB:82F5.0003: input,hidraw2: I2C HID v1.00 Mouse [SYNA3091:00 06CB:82F5] on i2c-SYNA3091:00
[   35.590105] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[   35.590108] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:12:0
[   35.590362] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   35.594116] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp1
[   35.594118] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp2
[   35.594120] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link iDisp3
[   35.594122] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Analog Playback and Capture
[   35.594124] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link Digital Playback and Capture
[   35.594126] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic01
[   35.594128] skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic16k
[   35.595062] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:12:0
[   35.595070] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   35.602389] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   35.602393] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   35.603161] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   35.603164] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   35.603943] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   35.603947] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   35.604743] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   35.604745] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[... 21 lines cut ...]
[   35.678209] skl_hda_dsp_generic skl_hda_dsp_generic: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[   35.678212] skl_hda_dsp_generic skl_hda_dsp_generic: Analog Codec DAI <-> Analog CPU DAI mapping ok
[   35.678215] skl_hda_dsp_generic skl_hda_dsp_generic: Digital Codec DAI <-> Digital CPU DAI mapping ok
[   35.678219] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC01 Pin mapping ok
[   35.678222] skl_hda_dsp_generic skl_hda_dsp_generic: dmic-hifi <-> DMIC16k Pin mapping ok
[   35.678247] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> DMIC48kHz 6 mapping ok
[   35.678261] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> DMIC16kHz 7 mapping ok
[   35.678281] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDA Analog 0 mapping ok
[   35.678306] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDA Digital 1 mapping ok
[   35.678317] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI1 3 mapping ok
[   35.678329] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI2 4 mapping ok
[   35.678339] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI3 5 mapping ok
[   35.678500] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi3 overwritten
[   35.678522] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi2 overwritten
[   35.678527] skl_hda_dsp_generic skl_hda_dsp_generic: ASoC: sink widget hifi1 overwritten
[   35.679417] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20
[   35.679486] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input21
[   35.679550] input: sof-hda-dsp HDMI/DP, pcm=18 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input22
[   35.679611] input: sof-hda-dsp HDMI/DP, pcm=19 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input23
[   35.679673] input: sof-hda-dsp HDMI/DP, pcm=20 Jack as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input24
```

[dmesg.txt](https://github.com/thesofproject/sof/files/4432642/dmesg.txt)

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

plbossart, 2020-04-04:
@leezu can you post the link to the results of alsa-info (or alsa-info.sh)?
We also need to know what distro you use, which version of alsa-lib is installed and if you see anything under /usr/share/alsa/ucm2/

Also have you tried to record directly from the hardware device for microphones? arecord -l will tell you what capture devices are available and arecord -Dhw0,x -c4 -fSE32_LE -r48000 should capture sound

leezu, 2020-04-05:
@plbossart thank you. Please find the alsa-info attached.

I'm on Ubuntu 18.04 with a self-compiled 5.6.2 kernel and the linux-firmware package from the Ubuntu 20.04 repository.[ `alsa-utils` is of version 1.1.3 ](https://packages.ubuntu.com/bionic/alsa-utils). `/usr/share/alsa/ucm2/` does not exist, but there is `/usr/share/alsa/ucm/`.

arecord lists the following devices

```
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC48kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

and `arecord -Dhw0,x -c4 -fS32_LE -r48000` (`fSE32_LE` you mentioned is invalid) for x in `x, 0, 1, 6, 7` all returns
```
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM hw0,7
arecord: main:788: audio open error: No such file or directory
```

One sidenote, prior to Linux 5.5 the microphone is [not detected at all on this platform](https://askubuntu.com/questions/1179045/built-in-microphone-not-detected-on-ubuntu-18-04-hp-elitebook-840-g6).

[alsa-info.txt](https://github.com/thesofproject/sof/files/4432769/alsa-info.txt)

plbossart, 2020-04-05:
@leezu you want to use card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC48kHz (*) []

so that's 

arecord -Dhw0,6 -c4 -fS32_LE -r48000

leezu, 2020-04-06:
Thanks for pointing out the missing `:` in my previous attempt. `arecord -Dhw:0,6 -c4 -fS32_LE -r48000` works well and records audio. Thus I think the problem is not with SoF or Alsa and we can close this issue? Thanks @plbossart!

I suppose manually upgrading PulseAudio may help to get the microphone working in other applications, or do you have any other recommendations?

plbossart, 2020-04-06:
Good progress @leezu, so I guess the next step is to figure out what UCM files will work for you.

Typically on Ubuntu18.04 the 'old' files will work, but we since did a bunch of card renames, etc, so you may need to 
a) upgrade your distro if you can
b) edit the UCM files

Can you share the tar file of what you see in /usr/share/alsa/ucm/sof*hda*dsp ? wildcards used on purpose since there are _ and - confusions.

leezu, 2020-04-07:
I have the following files:

```
/usr/share/alsa/ucm/sof-skl_hda_card/Hdmi1.conf
/usr/share/alsa/ucm/sof-skl_hda_card/Hdmi2.conf
/usr/share/alsa/ucm/sof-skl_hda_card/Hdmi3.conf
/usr/share/alsa/ucm/sof-skl_hda_card/HdmiHdac.conf
/usr/share/alsa/ucm/sof-skl_hda_card/HiFi.conf
/usr/share/alsa/ucm/sof-skl_hda_card/Mics.conf
/usr/share/alsa/ucm/sof-skl_hda_card/sof-skl_hda_card.conf
```

https://packages.ubuntu.com/bionic-updates/all/libasound2-data/filelist

plbossart, 2020-04-07:
@leezu understand. So it looks like you are caught in a hole

a) you cannot update
b) the UCM files you have are obsolete and no longer match what the kernel does. There is a break of ABI that happened with 5.5.x

So the only think I can suggest is to take you existing UCM files, and so a string replacement of 
sof-skl_hda_card by sof-hda-dsp (directory names, file names and inside files as well).

then play with alsa-ucm to see if you changes worked. then use pulseaudio and repeat.