thesofproject/sof #2735: [BUG][APL]"error: ipc timed out for 0x60070000 size 12" when do Pause/Release: simultaneous playback/capture  test

Liviali155, 2020-04-07:
Describe the bug
"error: ipc timed out for 0x60070000 size 12" when do Pause/Release: simultaneous playback/capture  test
After error occured,all the pipeline can not work,reboot can recovery

```
ubuntu@sh-apl-up2-pcm512x-02:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofpcm512x [sof-pcm512x], device 0: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 1: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 2: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 3: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 4: Media Playback 4 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
ubuntu@sh-apl-up2-pcm512x-02:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofpcm512x [sof-pcm512x], device 0: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 5: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 6: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
To Reproduce
1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
5../multiple-pause-resume.sh -r 25(https://github.com/thesofproject/sof-test/blob/master/test-case/multiple-pause-resume.sh)

Reproduction Rate
2/3

Expected behavior
There is o error occured when do do Pause/Release: simultaneous playback/capture  test

Impact
Connection timed out,dmesg show ""error: ipc timed out for 0x60070000 size 12" when do pause/release of aplay(0,0) and arecord (0,0)

[image]

<details>
 <summary>dmesgerror.log</summary>

````
32110] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 4
sh-apl-up2-pcm512x-02 kernel: [ 1094.433119] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x2014001e successful
sh-apl-up2-pcm512x-02 kernel: [ 1094.433137] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
sh-apl-up2-pcm512x-02 kernel: [ 1094.844684] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 3
sh-apl-up2-pcm512x-02 kernel: [ 1094.936041] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60070000 size 12
sh-apl-up2-pcm512x-02 kernel: [ 1094.936055] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
sh-apl-up2-pcm512x-02 kernel: [ 1094.936064] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
sh-apl-up2-pcm512x-02 kernel: [ 1094.936120] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
sh-apl-up2-pcm512x-02 kernel: [ 1094.936132] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000083 rirb 00
sh-apl-up2-pcm512x-02 kernel: [ 1094.936138] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
sh-apl-up2-pcm512x-02 kernel: [ 1094.936147] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
sh-apl-up2-pcm512x-02 kernel: [ 1094.936153] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
sh-apl-up2-pcm512x-02 kernel: [ 1094.936172]  Port5: ASoC: trigger FE cmd: 4 failed: -110
sh-apl-up2-pcm512x-02 kernel: [ 1094.936190] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 0 cmd 3
sh-apl-up2-pcm512x-02 kernel: [ 1094.936201] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
sh-apl-up2-pcm512x-02 kernel: [ 1094.936212] sof-audio-pci 0000:00:0e.0: error: trace IO error
sh-apl-up2-pcm512x-02 kernel: [ 1095.440015] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60060000 size 12
````
</details>

<details>
 <summary>sof_errortrace.log</summary>

````
    TIMESTAMP              DELTA C# COMPONENT          LOCATION                       CONTENT
[1094283701.770833] (             nan) c0 DMA                    src/drivers/dw/dma.c:1063  ERROR dw_dma_get_data_size(): xrun detected
[1094283706.406250] (        4.635417) c0 dai          6.39           src/audio/dai.c:648   ERROR dai_report_xrun(): overrun due to no data available
[1094283709.791667] (        3.385417) c0 dai          6.39           src/audio/dai.c:657   ERROR comp_overrun(): sink->free = 768, copy_bytes = 0
[1094283770.156250] (       60.364582) c0 PIPE                   src/audio/pipeline.c:860   ERROR pipeline_copy(): ret = -61, start->comp.id = 39, dir = 0
[1094283774.166667] (        4.010417) c0 PIPE         6.40      src/audio/pipeline.c:1040  ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
````
</details>

[dmesg.txt](https://github.com/thesofproject/sof/files/4442179/dmesg.txt)
[etrace.txt](https://github.com/thesofproject/sof/files/4442180/etrace.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4442181/slogger.txt)

Liviali155, 2020-04-07:
Bisect result:Caused by firmware commit https://github.com/thesofproject/sof/commit/8b7f1fafd20e1a0445bda7ab30190559b592e7e6

paulstelian97, 2020-04-07:
> Bisect result:Caused by firmware commit [8b7f1fa](https://github.com/thesofproject/sof/commit/8b7f1fafd20e1a0445bda7ab30190559b592e7e6)

Did you turn off the agent on that firmware? This commit switches to DMA scheduling which doesn't work well with the agent (currently).

Liviali155, 2020-04-09:
> > Bisect result:Caused by firmware commit [8b7f1fa](https://github.com/thesofproject/sof/commit/8b7f1fafd20e1a0445bda7ab30190559b592e7e6)
> 
> Did you turn off the agent on that firmware? This commit switches to DMA scheduling which doesn't work well with the agent (currently).

After turn off the agent，issue still can be reproduced

paulstelian97, 2020-04-09:
How many pipelines? It's possible we have an issue of overloading the system so you may need to also disable the low-power clock (if it applies to this platform, I'm not sure)

Liviali155, 2020-04-09:
> How many pipelines? It's possible we have an issue of overloading the system so you may need to also disable the low-power clock (if it applies to this platform, I'm not sure)

Error occured when do pause/release of 
 aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE -vv -i /dev/zero -q
arecord -D hw:0,0 -r 48000 -c 2 -f S16_LE -vv -i /dev/null -q

 I will try to disable the low power to check it

Liviali155, 2020-04-09:
It seems it has no low power clock item,I have no method to disable low-power clock
[image]

paulstelian97, 2020-04-09:
Hm, the particular topology you mention in the issue shouldn't be affected at all by that commit if I read it correctly. @plbossart @tlauda can you take a look?

The commit contents says you manually copy one of the topologies (removing the -master and -44100 suffixes if there). The one without such a suffix should remain identical and thus this commit shouldn't affect it.

plbossart, 2020-04-10:
that bisect result looks like a false positive to me?

Liviali155, 2020-04-10:
I retest it：
https://github.com/thesofproject/sof/commit/8b7f1fafd20e1a0445bda7ab30190559b592e7e6  fail
https://github.com/thesofproject/sof/commit/1ecf435d137864b1d1d1a95521611559444a8c72 pass

YvonneYang2, 2020-05-07:
Double checked, same result with @Liviali155 . Issue can not be reproduced with commit 1ecf435 after 1000 times test but can be reproduced with commit 8b7f1fa. Bisect result shows the root cause came from commit  8b7f1fa

YvonneYang2, 2020-05-09:
Tried to test it on latest sof-dev + revert commit 8b7f1fa , issue cannot be reproduced.

YvonneYang2, 2020-05-13:
Now with latest sof-dev+master, issue can be reproduced with same error sof-trace log but "ipc error for 0x60050000 size 12" in dmesg. 

dmesg:
```
kernel: [ 3865.028026] pcm512x i2c-104C5122:00: No SCLK, using BCLK: -2
kernel: [ 3865.043051] pcm512x i2c-104C5122:00: No SCLK, using BCLK: -2
kernel: [ 3865.552015] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60050000 size 12
kernel: [ 3865.552028]  Port5: ASoC: trigger FE cmd: 0 failed: -22
```
error sof-trace:
```
[3865052345.729167] (     4001.510498) c0 dw-dma                 src/drivers/dw/dma.c:1020  dw_dma_avail_data_size() size is 0!
[3865052358.750000] (       13.020833) c0 host         6.35          src/audio/host.c:382   host_copy_normal(): no bytes to copy
[3865054343.125000] (     1984.375000) c0 dw-dma                 src/drivers/dw/dma.c:1056  ERROR dw_dma_get_data_size(): xrun detected
[3865054346.406250] (        3.281250) c0 dai          6.39           src/audio/dai.c:649   ERROR dai_report_xrun(): overrun due to no data available
[3865054348.854167] (        2.447917) c0 dai          6.39           src/audio/dai.c:681   ERROR comp_overrun(): sink->free = 768, copy_bytes = 0
[3865054351.875000] (        3.020833) c0 pipe         6.40      src/audio/pipeline.c:736   pipe trigger cmd 8
[3865054356.406250] (        4.531250) c0 dai          6.39           src/audio/dai.c:620   dai_comp_trigger(), XRUN
[3865054359.739583] (        3.333333) c0 dw-dma                 src/drivers/dw/dma.c:406   dw_dma_stop(): dma 1 channel 0 stop
[3865054364.687500] (        4.947917) c0 ssp-dai      1.5   ......./intel/cavs/ssp.c:749   ssp_trigger() cmd 8
[3865054371.458333] (        6.770833) c0 pipe                   src/audio/pipeline.c:896   ERROR pipeline_copy(): ret = -61, start->comp.id = 39, dir = 0
[3865054374.479167] (        3.020833) c0 pipe         6.40      src/audio/pipeline.c:1086  ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[3865054377.708333] (        3.229167) c0 ll-schedule        ./schedule/ll_schedule.c:117   task complete 0xbe042380 pipe-task 
[3865054380.625000] (        2.916667) c0 ll-schedule        ./schedule/ll_schedule.c:120   num_tasks 1 total_num_tasks 1
[3865056344.739583] (     1964.114624) c0 dai          1.8       src/audio/pipeline.c:206   perf comp_copy peak plat 265 cpu 4464
[3865546583.281250] (   490238.531250) c0 pipe         6.40      src/audio/pipeline.c:736   pipe trigger cmd 0
[3865546589.583333] (        6.302083) c0 hda-dma            .../intel/cavs/hda-dma.c:623   hda-dmac: 4 channel 1 -> stop
[3865546595.520833] (        5.937500) c0 dai          6.39     src/audio/component.c:159   ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_STOP
[3865546599.166667] (        3.645833) c0 pipe                   src/audio/pipeline.c:757   ERROR pipeline_trigger(): ret = -22, host->comp.id = 35, cmd = 0
[3865546602.187500] (        3.020833) c0 ipc                       src/ipc/handler.c:470   ERROR ipc: comp 35 trigger 0x50000 failed -22
```

Test recipe:
kernel: topic/sof-dev (0773fe68)
fw: master (782066e0)

[dmesg_.txt](https://github.com/thesofproject/sof/files/4621188/dmesg_.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4621189/slogger.txt)

YvonneYang2, 2020-06-08:
Issue can still be reproduced with same error sof-trace log but "ipc error for 0x60050000 size 12" in dmesg with latest master and topic/sof-dev 

FW: master commit: 428804e1
Kernel: topic/sof-dev commit: 1af84d0e

Liviali155, 2020-07-24:
Issue still can be reproduced with latest sof-dev(e1a25077)+master(44a65d85)on apl pcm512

plbossart, 2020-07-30:
#3258 seems to be a duplicate, but contains additional information that point to the DW DMA handling being the problem?

ktrzcinx, 2020-09-03:
The issue can't be reproduced with RVP-APL platform, bisect shows commit with topology changes, so it doesn't look like FW issue. I can't handle this issue because of limited access to full-linux-stack apl platform.

mengdonglin, 2020-09-08:
@plbossart @lyakh  https://github.com/thesofproject/sof/pull/3289 has been merged. Can you help check if this issue can be reproduced on your UP2 with PCM512?

Suggest lower the priority to P2 as this topology is not a product one.

lyakh, 2020-09-08:
> @plbossart @lyakh #3289 has been merged. Can you help check if this issue can be reproduced on your UP2 with PCM512?
> 
> Suggest lower the priority to P2 as this topology is not a product one.

@mengdonglin @plbossart I've run a multiple-pause-resume.sh test with 25 repetitions (-r 25). The first attempt failed with a "device busy" error from aplay (or arecord). No errors in dmesg or in FW traces. The second attempt ran through successfully. So, I would say I cannot reproduce it. The log does seem to look a bit different from #3256 but it's possible that the switch to the timer scheduling domain has fixed this one too...

mengdonglin, 2020-09-08:
@lyakh Many thanks for the fix and double check! Let's close it.

closed 2020-09-08 by mengdonglin