thesofproject/sof #2749: [BUG][APL] "Input/output error" occurred when do multiple-pipeline-capture test.

sinahuang, 2020-04-09:
Describe the bug
"Input/output error" occurred when multiple-pipeline-capture test. After error occured,all the pipeline can not work,reboot can recovery.

Used ubuntu@sh-apl-up2-pcm512x-02:

```
ubuntu@sh-apl-up2-pcm512x-02:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofpcm512x [sof-pcm512x], device 0: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 1: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 2: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 3: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 4: Media Playback 4 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
ubuntu@sh-apl-up2-pcm512x-02:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofpcm512x [sof-pcm512x], device 0: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 5: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 6: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
To Reproduce
1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
5.. ./multiple-pipeline-capture.sh -c 4 -l 50 (https://github.com/thesofproject/sof-test/blob/master/test-case/multiple-pipeline-capture.sh)

Reproduction Rate
Test for 50 times, failed at 10th time.

Expected behavior
There is no error occured when do multiple-pipeline-capture test.

Impact
 "Input/output error" occurred when do multiple-pipeline-capture test.

[image]

Platform: APL UP2 with I2S PCM512x
Firmware:  939c14 https://github.com/thesofproject/sof/commits/master
Kernel: 77b49b3 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-apl-pcm512x.tplg same as firmware

[sof_error_trace9.log](https://github.com/thesofproject/sof/files/4454611/sof_error_trace9.log)
[sof_trace9.log](https://github.com/thesofproject/sof/files/4454612/sof_trace9.log)
[dmesg9.log](https://github.com/thesofproject/sof/files/4454613/dmesg9.log)

sinahuang, 2020-04-09:
Bisect result:Caused by firmware commit:https://github.com/thesofproject/sof/commit/8b7f1fafd20e1a0445bda7ab30190559b592e7e6

Liviali155, 2020-04-09:
related with https://github.com/thesofproject/sof/issues/2735

lgirdwood, 2020-04-09:
Seems related to slave SSP support which can be fixed for v1.6

lgirdwood, 2020-04-09:
@plbossart fyi.

YvonneYang2, 2020-05-27:
Issue cannot be reproduced with latest sof-dev(commit: 866e4d55) and master(commit: 6cfd3e5a) .
So close it.

Liviali155, 2020-05-28:
Error occured again with  latest sof-dev(0d22f66b)+Master(46897802) when run Pause/Release: simultaneous playback/capture - 100x in CI stress test
1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
5. ./multiple-pause-resume.sh -r 25

<details>
 <summary>dmesg_error.log</summary>

````
kernel: [ 4360.880045] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 3
kernel: [ 4360.880055] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 4360.880184] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60060000 size 12
kernel: [ 4360.880198] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20240000 successful
kernel: [ 4360.880205]  Port5: ASoC: trigger FE cmd: 3 failed: -22
kernel: [ 4360.880237] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 4
kernel: [ 4360.880781] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x2024001e successful
kernel: [ 4360.880788] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 4360.880882] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60070000 size 12
kernel: [ 4360.880887]  Port5: ASoC: trigger FE cmd: 4 failed: -22
kernel: [ 4360.881032] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 3
kernel: [ 4360.881037] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
kernel: [ 4360.881085] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60060000 size 12
kernel: [ 4360.881107] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20240000 successful
kernel: [ 4360.881110]  Port5: ASoC: trigger FE cmd: 3 failed: -22
kernel: [ 4360.881114] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 4
kernel: [ 4360.881725] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x2024001e successful
kernel: [ 4360.881737] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
kernel: [ 4360.881807] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60070000 size 12
kernel: [ 4360.881817]  Port5: ASoC: trigger FE cmd: 4 failed: -22
kernel: [ 4360.885463] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 4360.885481] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
````
</details>

<details>
 <summary>sof_error.log</summary>

````
[4344274348.750000] (        3.437500) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306372
[4344274351.875000] (        3.125000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1527 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[4344274355.364584] (        3.489583) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 536870912
[4344274358.385417] (        3.020833) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306372
[4344274366.145834] (        7.760417) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 671088640
[4344274369.583334] (        3.437500) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306372
[4344274372.552084] (        2.968750) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1527 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[4344274376.197917] (        3.645833) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 536870912
[4344274379.218750] (        3.020833) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306372
[4344274386.979167] (        7.760417) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 671088640
[4344274390.572917] (        3.593750) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306373
[4344274393.541667] (        2.968750) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1527 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[4344274397.031250] (        3.489583) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 536870912
[4344274400.052084] (        3.020833) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306373
[4344274407.812500] (        7.760417) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 671088640
[4344274411.406250] (        3.593750) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306373
[4344274414.375000] (        2.968750) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1527 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[4344274417.864584] (        3.489583) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 536870912
[4344274420.989584] (        3.125000) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306373
[4344274428.645834] (        7.656250) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 671088640
[4344274432.083334] (        3.437500) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306373
[4344274435.208334] (        3.125000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1527 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[4344274438.697917] (        3.489583) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 536870912
````
</details>

[dmesg.txt](https://github.com/thesofproject/sof/files/4693184/dmesg.txt)
[etrace.txt](https://github.com/thesofproject/sof/files/4693185/etrace.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4693186/slogger.txt)

Reopen it

lgirdwood, 2020-05-29:
DMIC is not getting emptied fast enough.  @mmaka1 @tlauda do we need more overhead for some DMA buffers (an extra period here) ?

```
[4344274542.760417] (        3.489583) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 536870912
[4344274545.885417] (        3.125000) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306375
[4344274553.645834] (        7.760417) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1523 dmic_irq_handler(), OUTSTAT0 = 671088640
[4344274557.083334] (        3.437500) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1524 dmic_irq_handler(), OUTSTAT1 = 805306375
[4344274560.208334] (        3.125000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1527 ERROR dmic_irq_handler(): full fifo A or PDM overrun
```

YvonneYang2, 2020-06-08:
Issue can still be reproduced as https://github.com/thesofproject/sof/issues/2749#issuecomment-635088125 said on latest topic/sof-dev(1af84d0e) and master(428804e1)

Liviali155, 2020-07-07:
On TGL chromebook  with codec 98357 in I2S mode also can reproduce this issue when do multiple-pipeline-playback test
1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=sof-tgl-max98357a-rt5682.tplg
5../multiple-pipeline-playback.sh -c 8 -l 300

Failed at 2th
[image]

<details>
 <summary>dmesg_error.log</summary>

````
[   85.543221] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[   85.543359] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60050000 size 12
[   85.543881] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   85.543893] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x340000 successful
[   85.543898]  Headset: ASoC: trigger FE cmd: 0 failed: -22
[   85.543900] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   85.543914]  Headset: ASoC: hw_free FE Headset
[   85.543918] sof-audio-pci 0000:00:1f.3: pcm: free stream 1 dir 1
[   85.543923] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[   85.544069] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x6003000
````
</details>

<details>
 <summary>sof_error.log</summary>

````
239583) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1537 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[177856006.875000] (        8.072917) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 536870912
[177856014.531250] (        7.656250) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 2147483648
[177856024.531250] (       10.000000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 671088640
[177856032.083333] (        7.552083) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 2147483648
[177856039.375000] (        7.291667) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1537 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[177856047.447917] (        8.072917) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 536870912
[177856055.052083] (        7.604167) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 2147483648
[177856065.052083] (       10.000000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 671088640
[177856072.395833] (        7.343750) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 2147483648
[177856079.843750] (        7.447917) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1537 ERROR dmic_irq_handler(): full fifo A or PDM overru
````
</details>

[dmesg0707.log](https://github.com/thesofproject/sof/files/4884233/dmesg0707.log)
[sof-logger_0707.log](https://github.com/thesofproject/sof/files/4884235/sof-logger_0707.log)

Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel: Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev (commit: 1b2f260)
+ PR#2208 for DP fix
+ PR#2184 for wov S0i2
+PR#2238 for max98373-sdw jack switch
+ codec hack commit to verify this captured data.
+ revert regression on USB-C 6a7c533 for volteer
+ revert d4b41f8bb
+ revert regression on PINCTRL cd0a323 for volteer
SOF: https://github.com/thesofproject/sof/tree/tgl-007-drop-stable v7.0-rc6,commit: a40b46c
+ PR#3116 for KWD trigger volume
Name of the topology file
Topology: sof-tgl-max98357a-rt5682.tplg
Name of the platform(s) on which the bug is observed.
Platform: TGL chromebook with codec 98357 in I2S mode

Liviali155, 2020-07-08:
Issue also can be reproduced on TGL  chromebook in  SDW mode with codec 98373 when run multipipeline with https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc7 ,commit: 33cfa90e

Issue can not be reproduced with FW:https://github.com/thesofproject/sof/tree/tgl-007-drop-stable   v7.0-rc6,commit: a40b46c1  + PR#3116 for KWD trigger volume

1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=sof-tgl-sdw-max98373-rt5682.tplg
5../multiple-pipeline-playback.sh -c 9 -l 300

Failed at 1th when arecord via dmic (0,4)
[image]

<details>
 <summary>sof_error.log</summary>

````
0 dai          3.17           src/audio/dai.c:736  dai_config() dai type = 4 index = 258
[145318319.010417] (   596156.125000) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 671088640
[145318343.906250] (       24.895834) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 2147483648
[145318367.552083] (       23.645834) c0 dmic-dai     2.0   src/drivers/intel/dmic.c:1537 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[145318392.864583] (       25.312500) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1533 dmic_irq_handler(), OUTSTAT0 = 671088640
[145318417.552083] (       24.687500) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1534 dmic_irq_handler(), OUTSTAT1 = 2147483648
[145318441.145833] (       23.593750) c0 dmic-dai     2.1   src/drivers/intel/dmic.c:1537 ERROR dmic_irq_handler(): full fifo A or PDM overrun
[145318773.958333] (      332.812500) c0 hda-dma            ..../intel/hda/hda-dma.c:790  ERROR hda_dma_link_check_xrun(): underrun detected
[145319395.520833] (      621.562500) c0 hda-dma            ..../intel/hda/hda-dma.c:790  ERROR hda_dma_link_check_xrun(): underrun detected
[145319989.843750] (      594.322937) c0 hda-dma            ..../intel/hda/hda-dma.c:790  ERROR hda_dma_link_check_xrun(): underrun detected
[145320307.447917] (      317.604156) c0 hda-dma            ..../intel/hda/hda-dma.c:790  ERROR hda_dma_link_check_xrun(): underrun detected
[145320430.104167] (      122.656250) c0 ll-schedule        /schedule/timer_domain.c:35   ERROR 
````
</details>

<details>
 <summary>dmesg_error.log</summary>

````
Volteer-rt5682-max98373-i2s kernel: [   72.400283] rt5682 sdw:0:25d:5682:0: rt5682_clock_config complete, clk_freq=4800000
Volteer-rt5682-max98373-i2s kernel: [   72.400824] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2034001e successful
Volteer-rt5682-max98373-i2s kernel: [   72.404536] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x24001e successful
Volteer-rt5682-max98373-i2s kernel: [   73.322150] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60040000 size 12
Volteer-rt5682-max98373-i2s kernel: [   73.322153] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
Volteer-rt5682-max98373-i2s kernel: [   73.322197] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
Volteer-rt5682-max98373-i2s kernel: [   73.322945] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
Volteer-rt5682-max98373-i2s kernel: [   73.322995] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0001f8f rirb 00
Volteer-rt5682-max98373-i2s kernel: [   73.323035] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
Volteer-rt5682-max98373-i2s kernel: [   73.323040] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
Volteer-rt5682-max98373-i2s kernel: [   73.323041] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
Volteer-rt5682-max98373-i2s kernel: [   73.323088] sof-audio-pci 0000:00:1f.3: error: trace IO error
Volteer-rt5682-max98373-i2s kernel: [   73.323093]  Headset mic: ASoC: trigger FE cmd: 1 failed: -110
Volteer-rt5682-max98373-i2s kernel: [   73.826167] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
Volteer-rt5682-max98373-i2s kernel: [   73.826210] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
Volteer-rt5682-max98373-i2s kernel: [   73.826213] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
Volteer-rt5682-max98373-i2s kernel: [   73.826985] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
Volteer-rt5682-max98373-i2s kernel: [   73.827032] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0001f8f rirb 00
Volteer-rt5682-max98373-i2s kernel: [   73.827033] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
Volteer-rt5682-max98373-i2s kernel: [   73.827039] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
Volteer-rt5682-max98373-i2s kernel: [   73.827040] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
Volteer-rt5682-max98373-i2s kernel: [   73.827619] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20340000 successful

````
</details>

[dmesg.txt](https://github.com/thesofproject/sof/files/4889717/dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4889719/slogger.txt)

ClarexZhou, 2020-07-09:
Error also seen on i2s 5682-max08373

Liviali155, 2020-07-13:
on TGL chromebook with codec 98357 in I2S mode
issue can be reproduced  in below scenerios

scenerio 1:
when do pause/release of speaker playbak&dmic capture
scenerio 2:
export TPLG=sof-tgl-max98357a-rt5682.tplg
./multiple-pause-resume.sh -r 100 (https://github.com/thesofproject/sof-test/blob/master/test-case/multiple-pause-resume.sh)

ClarexZhou, 2020-07-14:
move to P1 since TGL also failed.

Liviali155, 2020-07-14:
The issue of tgl has creat another issue https://github.com/thesofproject/sof/issues/3177 to track.this issue is only used to trace apl issue

Liviali155, 2020-07-24:
Issue still can be reproduced with latest sof-dev(e1a25077)+master(44a65d8)on apl pcm512

plbossart, 2020-07-29:
@Liviali155 I get very different results depending on which scripts are used on APL-Up2 w/ PCM512x

firmware: https://github.com/thesofproject/sof/pull/3245
kernel: topic/sof-dev b31892bc2404c4ecc59d863a1d5cadba6740be5b

export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
/multiple-pipeline-capture.sh -c 4 -l 50
-> no issues, seems to work fine (no idea what the -c 4 option means, removing it works fine as well). I could not make this test break with my setup.

./multiple-pause-resume.sh 
no IPC errors in dmesg but same dmic_irq_handler issue. 100% reproducible problem, this never worked on my side.

Can you give it a try on your side and try to narrow to a specific problem? This issue has too many details on different tests, we can't make progress if we don't focus on a common/easy to reproduce issue.

@singalsu @ranj063 @marc-hb  FYI if this rings a bell.

ClarexZhou, 2020-07-29:
@plbossart  Verified with PR #3245 , issue still can be reproduced.
[image]

plbossart, 2020-07-29:
@ClarexZhou there's got to be some device variation, I just reran the same test with 100 iterations and it works fine for me.

````
root@plb-UP2-DBG:~# ./sof-test/test-case/multiple-pipeline-capture.sh -l 100
...
2020-07-29 15:36:51 UTC [INFO] pipeline start sleep 0.5s for device wakeup
2020-07-29 15:36:56 UTC [INFO] checking pipeline status
2020-07-29 15:36:56 UTC [INFO] preparing sleep 5
2020-07-29 15:37:01 UTC [INFO] checking pipeline status again
2020-07-29 15:37:01 UTC [INFO] Test Result: PASS!
````

plbossart, 2020-07-29:
@ClarexZhou do you actually run with the latest sof-test code? I don't have the same traces at all, I don't see the [COMMAND] parts, so wondering if we are aligned on the tools?

ClarexZhou, 2020-07-30:
> @ClarexZhou do you actually run with the latest sof-test code? I don't have the same traces at all, I don't see the [COMMAND] parts, so wondering if we are aligned on the tools?

Thanks.  Error was reproduced because of not switching sof-test branch.  Error not seen after using latest Master sof-test.  Close.

closed 2020-07-30 by ClarexZhou