thesofproject/sof #2763: [EHL][BUG] ipc error for 0x60010000: GLB_STREAM_MSG: PCM_PARAMS when trigger DMIC capture

ClarexZhou, 2020-04-13:
**Describe the bug**
ipc error for 0x60010000: GLB_STREAM_MSG: PCM_PARAMS when trigger DMIC capture.
No effect for other pipline usage once error occurred

**To Reproduce**
arecord -Dhw:0,2 -fS16_LE -c4 -r48000 1.wav -vv -i

**Reproduction Rate**
3/3

**Expected behavior**
DMIC capture should works.

**Impact**
DMIC capture failed for "arecord: set_params:1403: Unable to install hw params" and ipc error in dmesg

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commits/topic/sof-dev, commit 4b4681e
    * SOF: https://github.com/thesofproject/sof/releases/tag/v1.5-rc1, commit: 5f7d4e2, using sof-tgl.ri rename to sof-ehl.ri
2) Name of the topology file
    * Topology: sof-ehl-rt5660.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: EHL RVP with RT5660 in I2S mode

**Screenshots or console output**
```

test@test-Elkhart-Lake-Embedded-Platform:~$ arecord -Dhw:0,2 --dump-hw-params
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "hw:0,2":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [64 128]
CHANNELS: 4
RATE: 48000
PERIOD_TIME: [250 42500]
PERIOD_SIZE: [12 2040]
PERIOD_BYTES: [192 16320]
PERIODS: [2 16]
BUFFER_TIME: [500 170500]
BUFFER_SIZE: [24 8184]
BUFFER_BYTES: [192 65472]
TICK_TIME: ALL
--------------------
arecord: set_params:1339: Sample format non available
Available formats:
- S16_LE
- S24_LE
- S32_LE
test@test-Elkhart-Lake-Embedded-Platform:~$ arecord -Dhw:0,2 -fS16_LE -c4 -r48000 1.wav -vv -i
Recording WAVE '1.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Channels 4
arecord: set_params:1403: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 64
CHANNELS: 4
RATE: 48000
PERIOD_TIME: 42500
PERIOD_SIZE: 2040
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 170500
BUFFER_SIZE: 8184
BUFFER_BYTES: 65472
TICK_TIME: 0
```

Dmesg:
```
[ 1018.058347] sof-audio-pci 0000:00:1f.3: pcm: open stream 2 dir 1
[ 1018.058351] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[ 1018.058353] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[ 1018.058354] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[ 1018.058596] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 2 dir 1
[ 1018.058599] sof-audio-pci 0000:00:1f.3: generating page table for 000000003c200025 size 0xffc0 pages 16
[ 1018.058610] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[ 1018.060933] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[ 1018.060938] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[ 1018.060939] sof-audio-pci 0000:00:1f.3: periods:4
[ 1018.060959] sof-audio-pci 0000:00:1f.3: stream_tag 2
[ 1018.060969] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1018.061159] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[ 1018.061163] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[ 1018.061166] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[ 1018.061170]  DMIC16kHz: ASoC: hw_params FE failed -22
[ 1018.061175] sof-audio-pci 0000:00:1f.3: pcm: free stream 2 dir 1
[ 1018.061396] sof-audio-pci 0000:00:1f.3: pcm: close stream 2 dir 1
[ 1018.869150] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1018.869191] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```
Error sof logger:
```
test@test-Elkhart-Lake-Embedded-Platform:~$ sudo sof-logger -l /etc/sof/sof*.ldc
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                       CONTENT
[130157201.510417] (130157200.000000) c0 dai          4.22           src/audio/dai.c:249   ERROR dai_verify_params(): pcm rate parameter does not match hardware rate
[130157213.437500] (       11.927083) c0 dai          4.22           src/audio/dai.c:390   ERROR dai_params(): pcm params verification failed.
[130157224.791667] (       11.354167) c0 PIPE                   src/audio/pipeline.c:469   ERROR pipeline_params(): ret = -22, host->comp.id = 18
[130157237.604167] (       12.812500) c0 IPC                       src/ipc/handler.c:302   ERROR ipc: pipe 4 comp 18 params failed -22
[157250381.041667] ( 27093144.000000) c0 dai          4.22           src/audio/dai.c:249   ERROR dai_verify_params(): pcm rate parameter does not match hardware rate
[157250392.343750] (       11.302083) c0 dai          4.22           src/audio/dai.c:390   ERROR dai_params(): pcm params verification failed.
[157250403.906250] (       11.562500) c0 PIPE                   src/audio/pipeline.c:469   ERROR pipeline_params(): ret = -22, host->comp.id = 18
[157250416.458333] (       12.552083) c0 IPC                       src/ipc/handler.c:302   ERROR ipc: pipe 4 comp 18 params failed -22
```
[soflogger_all.txt](https://github.com/thesofproject/sof/files/4469032/soflogger_all.txt)
[sof-logger_error.txt](https://github.com/thesofproject/sof/files/4469034/sof-logger_error.txt)
[dmesg_dmicCapture.txt](https://github.com/thesofproject/sof/files/4469036/dmesg_dmicCapture.txt)

ClarexZhou, 2020-04-13:
Error also seen on sof Master, commit 3808dfec

singalsu, 2020-04-23:
I'm missing something maybe, but why do you attempt to use -r 48000 for a 16 kHz PCM?

ClarexZhou, 2020-04-24:
Error also seen with -r 16000
```
test@test-Elkhart-Lake-Embedded-Platform:~$ arecord -Dhw:0,2 -fS16_LE -c4 -r16000 1.wav -vv -i
Recording WAVE '1.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Channels 4
Warning: rate is not accurate (requested = 16000Hz, got = 48000Hz)
         please, try the plug plugin
arecord: set_params:1403: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 64
CHANNELS: 4
RATE: 48000
PERIOD_TIME: 42500
PERIOD_SIZE: 2040
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 170500
BUFFER_SIZE: 8184
BUFFER_BYTES: 65472
TICK_TIME: 0
```

slawblauciak, 2020-04-27:
I think it's still trying to use the 48000 rate unless you do -Dplughw:0,2

slawblauciak, 2020-04-27:
Also, is DMIC supposed to support rates other than 16kHz on that platform?

ClarexZhou, 2020-04-27:
only using Dhw:0,2 directly, and dump info as below:
```

test@test-Elkhart-Lake-Embedded-Platform:~$ arecord -Dhw:0,2 --dump-hw-params
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "hw:0,2":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [64 128]
CHANNELS: 4
RATE: 48000
PERIOD_TIME: [250 42500]
PERIOD_SIZE: [12 2040]
PERIOD_BYTES: [192 16320]
PERIODS: [2 16]
BUFFER_TIME: [500 170500]
BUFFER_SIZE: [24 8184]
BUFFER_BYTES: [192 65472]
TICK_TIME: ALL
--------------------
arecord: set_params:1339: Sample format non available
Available formats:
- S16_LE
- S24_LE
- S32_LE
test@test-Elkhart-Lake-Embedded-Platform:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofehlrt5660 [sof-ehl-rt5660], device 0: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofehlrt5660 [sof-ehl-rt5660], device 1: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofehlrt5660 [sof-ehl-rt5660], device 2: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

slawblauciak, 2020-04-27:
It's a topology issue, the pipelines are not configured properly and have their stream parameters mismatched with DAI hardware configuration.

Also, a note to previous testing attempts: the hardware is configured only for 2 channels, so -c4 is also not going to work.

I'll prepare a PR shortly.

slawblauciak, 2020-04-27:
https://github.com/thesofproject/sof/pull/2886

ClarexZhou, 2020-04-28:
Verified and error can be fixed by PR #2886