thesofproject/sof #2764: [BUG][CML]Dmesg show "error: unknown DSP message 0xf0000000" when do Check-suspend-resume: 50x test

Liviali155, 2020-04-14:
Describe the bug
"error: unknown DSP message 0xf0000000" when do Check-suspend-resume: 50x test
Seems only a error message,has no impact on function

To Reproduce
1.sudo rebot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
5../check-suspend-resume.sh -l 50    

Reproduction Rate
1/3
failed at 33th

Expected behavior
No error message in dmesg

Impact
Dmesg show "error: unknown DSP message 0xf0000000" 

[image]

***Environment***
Kernel: https://github.com/thesofproject/linux/commits/topic/sof-dev, commit 4b4681e
SOF: https://github.com/thesofproject/sof/releases/tag/v1.5-rc1, commit: 5f7d4e2, using sof-cnl.ri rename to sof-cml.ri
Name of the topology file
sof-hda-generic-4ch.tplg
Name of the platform(s) on which the bug is observed.
Platform: CML RVP  with ALC700  in HDA

<details>
 <summary>dmesg_error.log</summary>

````
[ 1372.782453] Filesystems sync: 0.000 seconds
[ 1372.782803] Freezing user space processes ... (elapsed 0.002 seconds) done.
[ 1372.784864] OOM killer disabled.
[ 1372.784865] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[ 1372.785923] printk: Suspending console(s) (use no_console_suspend to debug)
[ 1372.945181] sof-audio-pci 0000:00:1f.3: error: unknown DSP message 0xf0000000
[ 1372.958850] e1000e: EEE TX LPI TIMER: 00000011
[ 1373.345738] ACPI: EC: interrupt blocked
[ 1373.448229] ACPI: Preparing to enter system sleep state S3
[ 1373.449297] ACPI: EC: event blocked
[ 1373.449297] ACPI: EC: EC stopped
[ 1373.449298] PM: Saving platform NVS memory
[ 1373.449310] Disabling non-boot CPUs ...
[ 1373.450764] smpboot: CPU 1 is now offline
[ 1373.452577] smpboot: CPU 2 is now offline
[ 1373.454384] smpboot: CPU 3 is now offline
````
</details>

[dmesg0414.log](https://github.com/thesofproject/sof/files/4473324/dmesg0414.log)

[sof-logger0414.log](https://github.com/thesofproject/sof/files/4473325/sof-logger0414.log)

Liviali155, 2020-04-14:
On master also can reproduce this issue
Tried 2times
1.pass
2.fail,fail at 26th
[image]

<details>
 <summary>dmesg_error.log</summary>

````
[ 1715.381842] Freezing user space processes ... (elapsed 0.001 seconds) done.
[ 1715.383503] OOM killer disabled.
[ 1715.383504] Freezing remaining freezable tasks ... (elapsed 0.000 seconds) done.
[ 1715.384496] printk: Suspending console(s) (use no_console_suspend to debug)
[ 1715.543932] sof-audio-pci 0000:00:1f.3: error: unknown DSP message 0xf0000000
[ 1715.557372] e1000e: EEE TX LPI TIMER: 00000011
[ 1715.944326] ACPI: EC: interrupt blocked
[ 1716.047841] ACPI: Preparing to enter system sleep state S3
[ 1716.048907] ACPI: EC: event blocked
[ 1716.048908] ACPI: EC: EC stopped
[ 1716.048909] PM: Saving platform NVS memory
[ 1716.048920] Disabling non-boot CPUs ...
[ 1716.050379] smpboot: CPU 1 is now offline
[ 1716.052574] smpboot: CPU 2 is now offline
[ 1716.055575] smpboot: CPU 3 is now offline
[ 1716.058114] smpboot: CPU 4 is now offline
[ 1716.059767] smpboot: CPU 5 is now offline
[ 1716.062102] smpboot: CPU 6 is now offline
[ 1716.063264] IRQ 122: no longer affine to CPU7
[ 1716.064279] smpboot: CPU 7 is now offline
[ 1716.064784] IRQ 17: no longer affine to CPU8
[ 1716.065829] smpboot: CPU 8 is now offline
[ 1716.066370] IRQ 135: no longer affine to CPU9
[ 1716.066375] IRQ 136: no longer affine to CPU9
[ 1716.067404] smpboot: CPU 9 is now offline
[ 1716.068006] IRQ 16: no longer affine to CPU10
[ 1716.069044] smpboot: CPU 10 is now offline
[ 1716.069682] IRQ 32: no longer affine to CPU11
````
</details>
[dmesg-master.log](https://github.com/thesofproject/sof/files/4473662/dmesg-master.log)
[sof-logger-master.log](https://github.com/thesofproject/sof/files/4473665/sof-logger-master.log)

linux (topic/sof-dev) | 053bce97
sof (master) | 3808dfec

Liviali155, 2020-04-14:
Changed fimware to 20200408(17f9296),error not seen
Changed firmware to 20200414(3808dfe),issue can be produced

sinahuang, 2020-05-07:
Error can be seen with today's sof-dev(commit:c05a3f86) and master(commit:6ce635), test for 1000 times and in the recent 14 days error also can't be reproduced. Closed it.

Liviali155, 2020-06-22:
Reopen for issue reproduced on latest sof-dev(c7b50d2d)+master(3d8a0ac8) when  run module reload
run check-kmod-load-unload.sh -l 50, failed at 49th
```
Removing snd_pcm
2020-06-21 23:43:26 UTC [REMOTE_INFO] checking for general errors after kmod unload with sof-kernel-log-check tool
2020-06-21 23:43:26 UTC [ERROR] Caught dmesg error
===========================>>
Jun 22 07:43:24 sh-cml-rvp-hda-01 kernel: [ 6066.466299] sof-audio-pci 0000:00:1f.3: error: unknown DSP message 0xf0000000
<<===========================
2020-06-21 23:43:26 UTC [REMOTE_ERROR] error found after kmod unload is real error, failing
2020-06-21 23:43:26 UTC [REMOTE_INFO] Test Result: FAIL!
```
[dmesg.txt](https://github.com/thesofproject/sof/files/4810888/dmesg.txt)

mengdonglin, 2020-09-01:
@xiulipan Do you think it's more reasonable to change this kernel message from an error to a warning? And as it's has no impact on audio functionality, can Linux CI tolerate this message in test?

xiulipan, 2020-09-01:
@mengdonglin I think we have refine on log filter and message in linux recently. Let track the CI result to see if this issue can be reproduced now.

mengdonglin, 2020-09-08:
@xiulipan besides CI,  I think this "error" message may cause users nervous unnecessarily.

slawblauciak, 2020-09-08:
@xiulipan please verify if we can close this issue.

kv2019i, 2020-09-08:
In bugscrub it was raised whether the issue is fixed by these:

ASoC: SOF: Intel: remove unnecessary waitq before loading firmware
  -> https://github.com/thesofproject/linux/pull/1875
ASoC: SOF: Intel: hda: remove unnecessary ROM IPC filter function
 -> https://github.com/thesofproject/linux/pull/1854

.. but, but, these were merged already in March, but bug was reported in April, so maybe this is still a separate issue. @xiulipan , are you still seeing the issue with current sof-dev?

xiulipan, 2020-09-08:
@kv2019i @aiChaoSONG @mengdonglin I did a quick search in our latest logs. The issue can still be seen on Aug.1 on APL, CML and BYT
It happen all with kmod reload randomly. Let's move this issue into Linux and try to debug with it. I think we may still have some issue with timing that missed this ROM status.

xiulipan, 2020-09-09:
@kv2019i @keyonjie @ranj063 I try to reproduce this issue on an CML platform. It seems the IPC comes after the CTX_SAVE message send. With full IPC CMD  `ipc rx: 0xffffffff` which seems to be related with PM gating?

```
[ 1118.665792] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1118.665882] sof-audio-pci 0000:00:1f.3: ipc rx: 0xffffffff: unknown GLB command
[ 1118.665887] sof-audio-pci 0000:00:1f.3: error: unknown DSP message 0xf0000000
[ 1118.665894] sof-audio-pci 0000:00:1f.3: ipc rx done: 0xffffffff: unknown GLB command
[ 1118.665943] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[ 1118.666011] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[ 1118.726839] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[ 1118.726849] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
```

keyonjie, 2020-09-21:
@xiulipan it looks this 0xffffffff rx IPC message is related to the CTX_SAVE IPC, can you debug to check if it is related the fact that we are invoking pm version helper sof_ipc_tx_message() which could potential send extra IPC in ->set_power_state()?

keyonjie, 2020-09-21:
we also observe similar issue on TGL when performing stress check-suspend-resume-500.sh.