thesofproject/sof #2766: sof errors on Lenovo Yoga C940 with cpu Cannon Lake  i9-9880H  [BUG]

aleksfadini, 2020-04-15:
Sound is working erratically

```
 dmesg | grep sof-audio-pci
[    4.570817] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.571127] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.571142] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.571357] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.765849] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.770549] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    4.813221] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    4.813223] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.815551] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    4.815554] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    4.939261] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[    4.939263] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:12:0
[    4.939550] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    4.944470] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:12:0
[    4.944474] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    4.945672] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    4.945674] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    4.946676] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    4.946680] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    4.947568] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    4.947570] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    4.948487] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    4.948488] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    4.948490] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    4.948491] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    4.948494] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    4.948496] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    4.949477] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    4.949479] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    4.950620] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    4.950622] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    4.951550] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    4.964808] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   12.078072] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   12.213026] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   33.018026] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   33.148565] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1063.565722] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1063.699750] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1305.411469] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1305.545349] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

Microphone not woring right either, tried a few fixes that worked fron Yoga c940 IceLake but not working for the Cannon Lake version.

**Environment**
    * Kernel: Linux GreatGentleGiant 5.6.4-arch1-1 #1 SMP PREEMPT Mon, 13 Apr 2020 12:21:19 +0000 x86_64 GNU/Linux

    * SOF: arch linux, sof-firmware 1.4.2-1

plbossart, 2020-04-15:
@aleksfadini the errors in the dmesg log are misleading, ignore them.

Can you please provide a link to the results of alsa-info ?

Also please try and capture at the hw level

arecord -Dhw:0,6 -c4 -r48000 -fS32_LE -d 10 mytest.wav
aplay -Dhw:0,0 mytest.wav

And please list what you have in /usr/share/alsa/ucm and /usr/share/alsa/ucm2.

Thanks!

aleksfadini, 2020-04-16:
Thank you for looking into this.
In reverse order:

```
 ls /usr/share/alsa/ucm
sof-skl_hda_card
```

```
 ls /usr/share/alsa/ucm2
broadwell-rt286  chtnau8824  GoogleNyan    SDP4430
broxton-rt298    chtrt5645   HDA-Intel     skylake-rt286
bytcht-cx2072x   chtrt5650   kblrt5660     sof-hda-dsp
bytcht-es8316    codecs      PandaBoard    tegraalc5632
bytcr-rt5640     DAISY-I2S   PandaBoardES  USB-Audio
bytcr-rt5651     DB410c      PAZ00         VEYRON-I2S
cht-bsw-rt5672   DB820c      platforms

```

Hw capture:
```

$ arecord -Dhw:0,6 -c4 -r48000 -fS32_LE -d 10 testFile.wav
Recording WAVE 'testFile.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Channels 4
arecord: set_params:1349: Channels count non available
```

```
$ aplay -Dhw:0,0 testFile.wav
Playing WAVE 'testFile.wav' : Signed 24 bit Little Endian in 3bytes, Rate 88200 Hz, Stereo
aplay: set_params:1343: Sample format non available
Available formats:
- S16_LE
- S24_LE
- S32_LE

```

alsa-info coming...

plbossart, 2020-04-16:
ok, let's retry, looks like you only have 2 mics. We should be able to verify this with the following command:

amixer -Dhw:0 info
You should see something in there that list cfg-dmic:2 ?

arecord -Dhw:0,6 -c2 -r48000 -fS32_LE -d 10 testFile.wav
aplay -Dhw:0,0 testFile.wav