thesofproject/sof #2775: [FEATURE] Add support for uint64_t in sof-logger

dbaluta, 2020-04-16:
**Is your feature request related to a problem? Please describe.**

We need to be able to log 64bit variables. 
```
uint64_t var;
trace_error(%I64d\n", var);
trace_error("%I64x\n", var);
```

First we will need to understand how intrusive the changes are, how hard is to implement this and if it is worth it.
For now we can always workaround this issues by printing upper and lower 32bits.

dbaluta, 2020-04-16:
Cc: @IulianOlaru249  @ymdatta @crazoes  feel free to look at this if you have some time. Just let a comment here so there won't be duplicate work.

ymdatta, 2020-04-16:
@dbaluta : I would like to work on this, can you assign it to me? Thanks.

paulstelian97, 2020-04-16:
It will be a major logger ABI change because the actual structures of the log messages will change -- the 4 parameter fields have to be changed from 32-bit integers to 64-bit integers and sof-logger has to be updated as well (which might prove tricky due to how 64-bit variables are passed on the stack differently from all others -- 16-bit and 8-bit integers are passed like 32-bit integers but 64-bit ones are different).

nayakaj, 2021-03-18:
hi
i'm new here can you help me to solve

dbaluta, 2021-03-18:
@nayakaj  just start with a simple tr_info call for example. Understand how that works first and then add support for 64 bits.

marc-hb, 2022-07-26:
The sof-logger is being deprecated, I think this should be closed.

paulstelian97, 2022-07-27:
> The sof-logger is being deprecated, I think this should be closed.

What are we using instead of it now?

lgirdwood, 2022-07-27:
@paulstelian97 trace/logging tools are in transition now to be fully Zephyr-ized. Good news is that there is a bunch of nice Zephyr tools (that do work with 64bits) bad news is that we have not completed the Zephyr log integration as of today. It's still WIP with target of v2.3

closed 2022-07-28 by dbaluta