thesofproject/sof #2776: [BUG][APL-nocodec]Warning error message ”error: DSP trace buffer overflow 4294967295 bytes“ occured in dmesg   when do Check-suspend-resume: 50x test

Liviali155, 2020-04-16:
*Describe the bug
Warning error message ”error: DSP trace buffer overflow 4294967295 bytes“ occured in dmesg   when do Check-suspend-resume: 50x test
Seems only a error message,has no impact on function,after error occured,all the pipeline can work normally

jf-cfl-rvp-hda-2

To Reproduce
1.sudo rebot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
5../check-suspend-resume.sh -l 50

Reproduction Rate
Tried 3 times
1.pass
2.Failed at 2/50th
3.Failed at 23/50th

Expected behavior
No error message in dmesg

Impact
Dmesg show "error: DSP trace buffer overflow 4294967295 bytes“

[image]

<details>
 <summary>dmesg_error.log</summary>

````
kernel: [ 7172.704382] PM: suspend entry (deep)
kernel: [ 7172.704823] Filesystems sync: 0.000 seconds
kernel: [ 7172.705210] Freezing user space processes ... (elapsed 0.001 seconds) done.
kernel: [ 7172.706955] OOM killer disabled.
kernel: [ 7172.706956] Freezing remaining freezable tasks ... (elapsed 0.000 seconds) done.
kernel: [ 7172.707917] printk: Suspending console(s) (use no_console_suspend to debug)
kernel: [ 7172.910933] sof-audio-pci 0000:00:1f.3: error: DSP trace buffer overflow 4294967295 bytes. Total messages -1
kernel: [ 7172.924390] e1000e: EEE TX LPI TIMER: 00000011
kernel: [ 7173.321782] ACPI: EC: interrupt blocked
kernel: [ 7173.348542] ACPI: Preparing to enter system sleep state S3
kernel: [ 7173.349733] ACPI: EC: event blocked
kernel: [ 7173.349733] ACPI: EC: EC stopped
kernel: [ 7173.349734] PM: Saving platform NVS memory
kernel: [ 7173.349739] Disabling non-boot CPUs ...
kernel: [ 7173.351193] smpboot: CPU 1 is now offline
kernel: [ 7173.353404] smpboot: CPU 2 is now offline
kernel: [ 7173.355604] smpboot: CPU 3 is now offline
kernel: [ 7173.357808] smpboot: CPU 4 is now offline
kernel: [ 7173.359636] smpboot: CPU 5 is now offline
kernel: [ 7173.362760] smpboot: CPU 6 is now offline
kernel: [ 7173.364426] smpboot: CPU 7 is now offline
kernel: [ 7173.365006] IRQ 122: no longer affine to CPU8
kernel: [ 7173.366039] smpboot: CPU 8 is now offline
kernel: [ 7173.366573] IRQ 140: no longer affine to CPU9
kernel: [ 7173.367669] smpboot: CPU 9 is now offline
kernel: [ 7173.368271] IRQ 124: no longer affine to CPU10
kernel: [ 7173.369300] smpboot: CPU 10 is now offline
kernel: [ 7173.369841] IRQ 138: no longer affine to CPU11
kernel: [ 7173.370882] smpboot: CPU 11 is now offline
         
````
</details>

CFL-S RVP with onboard codec ALC700 in HDA mode
Firmware: 7702288f https://github.com/thesofproject/sof/commits/master
Kernel: f9b87470 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ same as firmware

[dmesg.txt](https://github.com/thesofproject/sof/files/4486601/dmesg.txt)

tlauda, 2020-04-22:
@ranj063 @kv2019i @plbossart Any ideas when the kernel prints such error?

kv2019i, 2020-04-22:
@tlauda wrote:
> @ranj063 @kv2019i @plbossart Any ideas when the kernel prints such error?

This is print when FW sets the overflow message in the trace message:
```
struct sof_ipc_dma_trace_posn *posn
....
»       if (posn->overflow != 0)
»       »       dev_err(sdev->dev,
»       »       »       "error: DSP trace buffer overflow %u bytes. Total messages %d\n",
»       »       »       posn->overflow, posn->messages);
```

... this is called directly from the inbound IPC handler on the driver side.

ClarexZhou, 2020-04-30:
Error not occurred on sof-dev commit:8f1c6ad6 and master commit: af38b20b.
1st round 500 interactions: Pass
2nd round 500 interactions: Pass
3rd round 1k interactions: Pass

Keep monitor.

Liviali155, 2020-05-06:
Error not occured on sof-dev(c08157b6)+master(6ce635aa)
1st round 1000 interactions: Pass
2nd round 1000 interactions: Pass

ClarexZhou, 2020-05-07:
Close for error not seen

Liviali155, 2020-06-15:
Error reproduced on latest sof-dev(140f27c9)+master(a10b0ad2) when run module reload
1.sudo rebot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
5../check-kmod-load-unload.sh -l 50
6../check-kmod-load-unload.sh -l 50

Failed at 36th of step6

<details>
 <summary>dmesg_error.log</summary>

````
kernel: [ 6435.308504] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input434
kernel: [ 6435.308580] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input435
kernel: [ 6435.308648] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input436
kernel: [ 6435.308733] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input437
kernel: [ 6435.308939] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input438
kernel: [ 6436.615324] usbcore: deregistering interface driver snd-usb-audio
kernel: [ 6436.801230] sof-audio-pci 0000:00:1f.3: error: DSP trace buffer overflow 4294967295 bytes. Total messages -1
````
</details>
After error occured,can not catch sof-logger as below：

```
ubuntu@sh-cml-rvp-hda-01:~$ sudo sof-logger -l /etc/sof/sof-*.ldc
error: Unable to open in file /sys/kernel/debug/sof/etrace
```

[image]
[dmesg.txt](https://github.com/thesofproject/sof/files/4778974/dmesg.txt)

mengdonglin, 2020-08-11:
CI doesn't reproduce this error in recent daily test on CML Mantis. So closed this issue atm.

Test recipe.
Start Time: 2020-08-09 21:25:41 UTC
End Time: 2020-08-10 00:53:35 UTC
Kernel Branch: topic/sof-dev
Kernel Commit: 9724bf01
SOF Branch: master
SOF Commit:9f2a1abd

keqiaozhang, 2020-12-28:
CI caught this issue again on  CML.
https://sof-ci.sh.intel.com/#/result/planresultdetail/1371?model=CML_RVP_HDA&testcase=check-kmod-load-unload-50

xiulipan, 2020-12-29:
@keqiaozhang We already had an issue https://github.com/thesofproject/linux/issues/2576 to track this.

closed 2020-12-29 by xiulipan