thesofproject/sof #2783: [BUG][CML]ASRC]Warning message “error: ipc error for 0x60010000 size 20”occured  in dmesg  after bootup

Liviali155, 2020-04-17:
***Describe the bug***
Warning message “error: ipc error for 0x60010000 size 20”occured  in dmesg  after bootup
Just a error message,ALL the pipeline can work
```
ubuntu@cml-hda-mantis-1:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubuntu@cml-hda-mantis-1:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
***To Reproduce***
1.download sof-hda-asrc-2ch.tplg rename to sof-hda-generic-2ch.tplg
2.sudo reboot
3.dmesg -w

Reproduction Rate
5/5

***Expected behavior***
There is no error message in dmesg

***Impact***
“error: ipc error for 0x60010000 size 20”  in dmesg

<details>
 <summary>dmesg_error.log</summary>

````
[    2.843718] leds platform::micmute: Setting an LED's brightness failed (-19)
[    3.312184] random: crng init done
[    3.770392] leds platform::micmute: Setting an LED's brightness failed (-19)
[    3.770405] leds platform::micmute: Setting an LED's brightness failed (-19)
[    4.237954] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[    4.391044] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.391053] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.391057] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.391061]  HDA Analog: ASoC: hw_params FE failed -12
[    4.391976] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.391983] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.391987] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.391991]  HDA Analog: ASoC: hw_params FE failed -12
[    4.392871] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.392878] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.392882] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.392886]  HDA Analog: ASoC: hw_params FE failed -12
[    4.393740] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.393747] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.393752] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.393756]  HDA Analog: ASoC: hw_params FE failed -12
[    4.394595] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.394603] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.394607] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.394611]  HDA Analog: ASoC: hw_params FE failed -12
[    4.396120] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.396126] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.396130] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.396133]  HDA Analog: ASoC: hw_params FE failed -12
[    4.397025] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.397032] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.397036] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.397040]  HDA Analog: ASoC: hw_params FE failed -12
[    4.397877] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.397884] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.397889] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.397892]  HDA Analog: ASoC: hw_params FE failed -12
[    4.398741] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.398749] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.398753] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.398757]  HDA Analog: ASoC: hw_params FE failed -12
[    4.399564] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.399572] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.399577] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.399581]  HDA Analog: ASoC: hw_params FE failed -12
[    4.401712] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.401719] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.401724] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -1         
````
</details>

No sof error trace

***Environment***
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
Kernel: {f9b87470} /sof/dev
SOF: {7702288f} master
Name of the topology file
Topology: {sof-hda-asrc-2ch.tplg rename to sof-hda-generic-2ch.tplg}
Name of the platform(s) on which the bug is observed.
Platform: {CML laptop with onboard codec ALC3204 in HDA mode}

[dmesg0416.log](https://github.com/thesofproject/sof/files/4490843/dmesg0416.log)
[sof-logger0416.log](https://github.com/thesofproject/sof/files/4490844/sof-logger0416.log)

Liviali155, 2020-04-17:
This issue also can be reproduced on sof v1.5-rc2+PR2750 after download sof-hda-asrc-2ch.tplg rename to sof-hda-generic-2ch.tplg
Kernel:f9b87470 Sof-dev
fw: 66e20171 https://github.com/thesofproject/sof/releases/tag/v1.5-rc1+PR2750
UCM2(https://github.com/thesofproject/alsa-ucm-conf/tree/master/ucm2/sof-hda-dsp)

<details>
 <summary>sof_errortrace.log</summary>

````
    TIMESTAMP              DELTA C# COMPONENT          LOCATION                       CONTENT
[  5761845.052083] (  5761845.000000) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5762660.260417] (      815.208313) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5762672.083333] (       11.822917) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5762679.895833] (        7.812500) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5763570.625000] (      890.729187) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5763582.500000] (       11.875000) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5763590.156250] (        7.656250) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5771803.958333] (     8213.801758) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5771816.093750] (       12.135417) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5771823.906250] (        7.812500) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5772531.354167] (      707.447937) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5772542.864583] (       11.510417) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5772550.572917] (        7.708333) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5773299.062500] (      748.489563) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5773310.937500] (       11.875000) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5773318.437500] (        7.500000) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5773999.635417] (      681.197937) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5774011.510417] (       11.875000) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5774019.114583] (        7.604167) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5774799.375000] (      780.260437) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5774811.145833] (       11.770833) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5774818.697917] (        7.552083) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5775682.864583] (      864.166687) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5775694.843750] (       11.979167) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5775702.500000] (        7.656250) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5776481.093750] (      778.593750) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5776492.968750] (       11.875000) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5776500.781250] (        7.812500) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5777232.604167] (      731.822937) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5777244.270833] (       11.666667) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5777251.927083] (        7.656250) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5777948.333333] (      696.406250) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[  5777960.468750] (       12.135417) c0 PIPE                   src/audio/pipeline.c:583   ERROR pipeline_prepare(): ret = -12, dev->comp.id = 11
[  5777968.177083] (        7.708333) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 2 comp 11 prepare failed -12
[  5778733.750000] (      765.572937) c0 asrc         2.12     src/audio/asrc/asrc.c:654   ERROR asrc_prepare(), allocation fail for size 1660
[... cut ...]
[    4.550256] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[    4.550268] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[    4.550270] sof-audio-pci 0000:00:1f.3: period_bytes:0x540
[    4.550271] sof-audio-pci 0000:00:1f.3: periods:14
````
</details>

[dmesg0417.log](https://github.com/thesofproject/sof/files/4491337/dmesg0417.log)
[sof-errortrace0417.log](https://github.com/thesofproject/sof/files/4491340/sof-errortrace0417.log)
[sof-logger_0417.log](https://github.com/thesofproject/sof/files/4491341/sof-logger_0417.log)

slawblauciak, 2020-04-17:
Interesting, it would appear FW is unable to allocate 1660 bytes for frame processing in asrc.

singalsu, 2020-04-17:
Sounds like running out of 2 kB pool. I don't understand how the fail is not propagated and streaming aborted.

singalsu, 2020-04-17:
I checked from code the ASRC prepare() flow and the -ENOMEM should abort the playback. I noticed a minor mistake in the previous allocation fail handling (lines 622-630) and will do a fix PR that avoids free of null pointer by changing the goto into err instead of err_free_buf. But it won't help this issue.

lgirdwood, 2020-04-19:
@singalsu sound like we need to increase pool size here on CML

singalsu, 2020-04-20:
@lgirdwood  Yep, that should avoid the error. I'll test here, hope it happens with a CFL too that I can use.

Liviali155, 2020-04-21:
I tried on CI devices  CML laptop with onboard codec ALC3204 in HDA  which is ubuntu 18.04 with lastest sof/dev(f17b96ae) +master(504ec438),error not seen
But I used my own CML laptop with onboard codec ALC3204 in HDA mode which is ubuntu 20.04,issue can be reproduced
Seems this issue only occured on ubuntu20.04

Liviali155, 2020-04-22:
Tried with PR  https://github.com/thesofproject/sof/pull/2831/files,error not seen

Liviali155, 2020-04-22:
Error not seen on latest sof/dev(8dc1789e) +master(0befebb7),closed it

closed 2020-04-22 by Liviali155