thesofproject/sof #2805: [BUG]

dingesista, 2020-04-19:
**Describe the bug**
Thinkpad X1 Carbon 7th generation
Debian Unstable, Linux 5.5.0-2-amd64 Debian 5.5.17-1 (2020-04-15) x86_64 GNU/Linux
Has sound, but microphones are not detected.

**Expected behavior**
It would be great to have the mics working. 

**Impact**
A headset microphone works fine, so not all is lost.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Linux 5.5.0-2-amd64 Debian 5.5.17-1 (2020-04-15) x86_64 GNU/Linux
    * SOF: {SHA} I don't know, sorry
2) Name of the topology file
    * Topology: the ucm2 directory has: 
HDA-Capture-value.conf 
Hdmi.conf
HiFi.conf
sof-hda-dsp.conf 

3) Name of the platform(s) on which the bug is observed.
Debian unstable

**Screenshots or console output**
from syslog:

snd_hda_intel 0000:00:1f.3: dmic_detect option is deprecated, pass snd-intel-dspcfg.dsp_driver=1 option instead
snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])

and also:

snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
snd_hda_codec_realtek hdaudioC0D0:    inputs:
snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19

alsaucm listcards
ALSA lib utils.c:261:(uc_mgr_config_load) could not open configuration file /usr/share/alsa/ucm2/HDA-Intel/HDA-Intel.conf
alsaucm: error failed to get card list: No such file or directory

alsaucm -i
alsaucm: Interacive mode - 'q' to quit
alsaucm>> open sof-hda-dsp
ALSA lib ucm_cond.c:207:(if_eval_control_exists) cannot determine control device
ALSA lib main.c:962:(snd_use_case_mgr_open) error: failed to import sof-hda-dsp use case configuration -22
alsaucm: error failed to open sound card sof-hda-dsp: Invalid argument
^^^ error, try again

/usr/share/alsa/ucm2/ has sof-hda-dsp
/usr/share/alsa/ucm is empty bar a README.md

lgirdwood, 2020-04-19:
@dingesista can you paste the full dmesg. @plbossart @kv2019i @juimonen fyi

dingesista, 2020-04-20:
[dmesg.txt](https://github.com/thesofproject/sof/files/4501561/dmesg.txt)

dingesista, 2020-04-20:
[image]

juimonen, 2020-04-20:
@dingesista Don't know if you did it, but did you press F5 after getting this screen? To show the capture side also? alsamixer doesn't show you capture mixers by default...

dingesista, 2020-04-20:
[image]

I did not hit F5 "today"; but I've seen this. Pavucontrol (still) says: No input devices available. But when I plug in an smartphone headset, it will immediately show "Microphone (plugged in), and there is an indicator that starts to wiggle because of the background noise.

lgirdwood, 2020-04-20:
@dingesista Looks like the legacy HDA driver is being used here
```
[   14.845123] snd_hda_intel 0000:00:1f.3: dmic_detect option is deprecated, pass snd-intel-dspcfg.dsp_driver=1 option instead
[   14.845315] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   14.919514] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
[   14.919516] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   14.919517] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   14.919517] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   14.919517] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   14.919518] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19
```
Can you try and pass ```snd-intel-dspcfg.dsp_driver=1 ``` in kernel cmd line.

dingesista, 2020-04-20:
I added that to /etc/default/grub in this line: GRUB_CMDLINE_LINUX="<here>" and then ran update-grub and rebooted. I'm doing it wrong, or the option is not propagated to the module. See below

dmesg | grep snd
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-5.5.0-2-amd64 root=/dev/mapper/inauditus--vg-root ro snd-intel-dspcfg.dsp_driver=1 quiet
[    0.184251] Kernel command line: BOOT_IMAGE=/vmlinuz-5.5.0-2-amd64 root=/dev/mapper/inauditus--vg-root ro snd-intel-dspcfg.dsp_driver=1 quiet
[   13.307781] snd_hda_intel 0000:00:1f.3: dmic_detect option is deprecated, pass snd-intel-dspcfg.dsp_driver=1 option instead
[   13.324585] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   13.376044] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC285: line_outs=2 (0x14/0x17/0x0/0x0/0x0) type:speaker
[   13.376045] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   13.376046] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[   13.376047] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[   13.376047] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[   13.376048] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x19

pavucontrol shows "No input devices available"; alsamixer shows the four s/pdif like above, and there is nothing I can do with the auto mute options

kv2019i, 2020-04-20:
@lgirdwood 
> Can you try and pass `snd-intel-dspcfg.dsp_driver=1 ` in kernel cmd line.

Actually @dingesista this is wrong, this actually force the non-DSP kernel driver to be used and dmic will not be found. So either pass "snd-intel-dspcfg.dsp_driver=0" (for auto) or "snd-intel-dspcfg.dsp_driver=3" for SOF.

You can see https://github.com/thesofproject/linux/issues/1877 for various tips and tips on issues you can hit.

I'll move this bug to kernel as this is not related to firmware based on these logs.