thesofproject/sof #2806: [BUG] [v1.4.2] System hang when boot up with pulseaudio on one Nvidia GPUs laptop

ClarexZhou, 2020-04-20:
**Describe the bug**
On one Nvidia GPUs laptop, system will hang when system boot up with pulseaduio enaBled.

Error seen with fw 1.4.2 but not see with latest master and 1.5-rc1 fw.

**To Reproduce**
1. Pulseaduio is enabled
2. Boot up system

**Reproduction Rate**
3/3

**Expected behavior**
Boot up system successfully

**Impact**
System hang during boot up.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: Sof-dev 4b4681eb
    * SOF: https://github.com/thesofproject/sof-bin/tree/stable-v1.4.2/lib/firmware/intel/sof/v1.4.2, commit 2ef369d5
2) Name of the topology file
    * Topology: sof-hda-generic-4ch.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: CFL Laptop in HDA mode, selek  G3 3590, 

Can't  get logs for system hang during boot up.

ClarexZhou, 2020-04-20:
System also hang in below steps if disabled pulseaduio:
1. boot up system after disable pulseaudio
2. Change PGA in playback volume to max via alsamixer
3. 'aplay -Dhw:1,0 -f dat music.wac -vv -i',   -> System hang.   No error dmesg when system hang

lgirdwood, 2020-04-20:
@mengdonglin @fkaczmar @lbetlej I know that v1.5 is imminent,  but v1.4.x is shipping with all the 20.04 Ubuntu/Redhat releases. If we are good with v1.5 on Ubuntu/ RH 20.04 then we can ship this as the fix rather than a new v1.4.3 release (as either way we the distros have to update FW).

plbossart, 2020-04-20:
> @mengdonglin @fkaczmar @lbetlej I know that v1.5 is imminent, but v1.4.x is shipping with all the 20.04 Ubuntu/Redhat releases. If we are good with v1.5 on Ubuntu/ RH 20.04 then we can ship this as the fix rather than a new v1.4.3 release (as either way we the distros have to update FW).

@lgirdwood unrelated topic but can we make sure all firmwares are generated for 1.5, I don't want to regenerate the baytrail/broadwell/apl/cml once as I did last time.

mengdonglin, 2020-04-21:
@lrgirdwo @fkaczmar @jajanusz latest result shows this issue is very likely to be the system hang issue we previously fixed for customer.

jajanusz, 2020-04-21:
I checked the source code of https://github.com/thesofproject/sof/tree/v1.4.2 and it does not have fix for hangs that was merged in this PR: https://github.com/thesofproject/sof/pull/2326

mengdonglin, 2020-04-21:
Thank you! @jajanusz  Can this fix be applied to https://github.com/thesofproject/sof/tree/v1.4.2?

jajanusz, 2020-04-22:
But 1.4.2 is not aligned with master from Feb 21, there are just cherr-picks from that date, so it needs more changes.
Anyway I'll make PR with backport for branch 1.4-stable.

ClarexZhou, 2020-04-27:
Close for error is fixed on https://github.com/thesofproject/sof/commits/v1.4.3

closed 2020-04-27 by ClarexZhou