thesofproject/sof #2810: [BUG][CML][EQ]EQ test failed

Liviali155, 2020-04-20:
**Describe the bug**
EQ test failed
The pipeline aplay/arecord normally
Caused by 736762b3 https://github.com/thesofproject/sof/releases/tag/v1.5-rc3
```
ubuntu@cml-hda-mantis-1:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubuntu@cml-hda-mantis-1:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
**To Reproduce**
1.chmod +x sof-ctl 
2.sudo mv sof-ctl /usr/bin/
3.get sof-hda-generic-eq-2ch.tplg,then renamed to sof-hda-generic-2ch.tplg
4.reboot
5.cd sof-test
6.cd test-case
7.export TPLG=/lib/firmware/intel/sof-hda-generic-2ch.tplg
8../check-audio-equalizer.sh(https://github.com/thesofproject/sof-test/blob/master/test-case/check-audio-equalizer.sh)

**Reproduction Rate**
10/10

**Expected behavior**
EQ test pass

**Impact**
EQ test failed
[image]

Liviali155, 2020-04-20:
Issue also can be reproduced on master
sof/dev  202bac22
master  504ec438

Liviali155, 2020-04-21:
Without 736762b https://github.com/thesofproject/sof/releases/tag/v1.5-rc3,eq test is pass,result as below:
```
ubuntu@ubuntu-Inspiron-5590:~/sof-test/test-case$ ./check-audio-equalizer.sh 
2020-04-21 02:53:29 UTC [INFO] ./check-audio-equalizer.sh will use topology /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg to run the test case
2020-04-21 02:53:29 UTC [INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital,Media Playback,DMIC16k' in test case
2020-04-21 02:53:29 UTC [INFO] Run command to get pipeline parameters
2020-04-21 02:53:29 UTC [COMMAND] sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg -f type:playback -b pcm:'HDA Digital,Media Playback,DMIC16k' -s 0 -e
2020-04-21 02:53:29 UTC [INFO] 0 - total pipeline= 4
2020-04-21 02:53:29 UTC [INFO] eq_support= EQIIR1.0 EQFIR1.0
2020-04-21 02:53:29 UTC [INFO] 0. Get amixer control id for IIR and FIR
2020-04-21 02:53:29 UTC [INFO] IIR list 5 ./eqctl/eq_iir_bandpass.txt ./eqctl/eq_iir_bassboost.txt ./eqctl/eq_iir_flat.txt ./eqctl/eq_iir_loudness.txt ./eqctl/eq_iir_pass.txt
2020-04-21 02:53:29 UTC [INFO] FIR list 4 ./eqctl/eq_fir_flat.txt ./eqctl/eq_fir_loudness.txt ./eqctl/eq_fir_mid.txt ./eqctl/eq_fir_pass.txt
2020-04-21 02:53:29 UTC [INFO] [1/1] 1. Test IIR config list, IIR amixer control id=42
2020-04-21 02:53:29 UTC [COMMAND] sof-ctl -Dhw:0 -n 42 -s ././eqctl/eq_iir_bandpass.txt
Control size is 304.
Applying configuration "././eqctl/eq_iir_bandpass.txt" into device hw:0 control numid=42.
4607827,0,116,50331648,0,0,0,0,116,2,1,0,0,0,0,0,0,2,2,0,0,0,0,3316150158,2048164275,513807534,3267352229,513807534,0,16384,3867454526,1191025347,38870735,77741469,38870735,4294967292,24197
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 116 bytes
hdr: abi 3:0:0
4607827,0,116,50331648,0,0,0,0,116,2,1,0,0,0,0,0,0,2,2,0,0,0,0,3316150158,2048164275,513807534,3267352229,513807534,0,16384,3867454526,1191025347,38870735,77741469,38870735,4294967292,24197
2020-04-21 02:53:29 UTC [COMMAND] aplay -Dhw:0,0 -f S16_LE -c 2 -r 48000 /dev/zero -d 5
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
Warning: rate is not accurate (requested = 8000Hz, got = 48000Hz)
         please, try the plug plugin 
2020-04-21 02:53:32 UTC [COMMAND] sof-ctl -Dhw:0 -n 42 -s ././eqctl/eq_iir_bassboost.txt
Control size is 304.
Applying configuration "././eqctl/eq_iir_bassboost.txt" into device hw:0 control numid=42.
4607827,0,116,50331648,0,0,0,0,116,2,1,0,0,0,0,0,0,2,2,0,0,0,0,3227172081,2141520527,536653443,3221660410,536653443,0,16384,3260252783,2107733822,161646111,3961037800,172645501,4294967294,27910
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 116 bytes
hdr: abi 3:0:0
4607827,0,116,50331648,0,0,0,0,116,2,1,0,0,0,0,0,0,2,2,0,0,0,0,3227172081,2141520527,536653443,3221660410,536653443,0,16384,3260252783,2107733822,161646111,3961037800,172645501,4294967294,27910
2020-04-21 02:53:32 UTC [COMMAND] aplay -Dhw:0,0 -f S16_LE -c 2 -r 48000 /dev/zero -d 5
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
Warning: rate is not accurate (requested = 8000Hz, got = 48000Hz)
         please, try the plug plugin 
2020-04-21 02:53:33 UTC [COMMAND] sof-ctl -Dhw:0 -n 42 -s ././eqctl/eq_iir_flat.txt
Control size is 304.
Applying configuration "././eqctl/eq_iir_flat.txt" into device hw:0 control numid=42.
4607827,0,88,50331648,0,0,0,0,88,2,1,0,0,0,0,0,0,1,1,0,0,0,0,0,0,0,0,538145694,0,32690
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 88 bytes
hdr: abi 3:0:0
4607827,0,88,50331648,0,0,0,0,88,2,1,0,0,0,0,0,0,1,1,0,0,0,0,0,0,0,0,538145694,0,32690
2020-04-21 02:53:33 UTC [COMMAND] aplay -Dhw:0,0 -f S16_LE -c 2 -r 48000 /dev/zero -d 5
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
Warning: rate is not accurate (requested = 8000Hz, got = 48000Hz)
         please, try the plug plugin 
2020-04-21 02:53:35 UTC [COMMAND] sof-ctl -Dhw:0 -n 42 -s ././eqctl/eq_iir_loudness.txt
Control size is 304.
Applying configuration "././eqctl/eq_iir_loudness.txt" into device hw:0 control numid=42.
4607827,0,204,50331648,0,0,0,0,204,4,2,0,0,0,0,1,1,1,1,1,1,0,0,0,0,0,0,0,0,538145694,0,32690,3,3,0,0,0,0,3237960220,2130690484,297056159,3694463533,303476880,0,16384,3302357025,2064935920,245217319,3776455865,274003881,0,16384,4106268671,4130185751,69182517,4179658584,365641401,4294967292,25468
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 204 bytes
hdr: abi 3:0:0
4607827,0,204,50331648,0,0,0,0,204,4,2,0,0,0,0,1,1,1,1,1,1,0,0,0,0,0,0,0,0,538145694,0,32690,3,3,0,0,0,0,3237960220,2130690484,297056159,3694463533,303476880,0,16384,3302357025,2064935920,245217319,3776455865,274003881,0,16384,4106268671,4130185751,69182517,4179658584,365641401,4294967292,25468
2020-04-21 02:53:35 UTC [COMMAND] aplay -Dhw:0,0 -f S16_LE -c 2 -r 48000 /dev/zero -d 5
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 8000 Hz, Stereo
Warning: rate is not accurate (requested = 8000Hz, got = 48000Hz)
         please, try the plug plugin 
2020-04-21 02:53:37 UTC [COMMAND] sof-ctl -Dhw:0 -n 42 -s ././eqctl/eq_iir_pass.txt
[... 59 lines cut ...]
4607827,0,56,50331648,0,0,0,0,56,65538,0,0,0,0,4294967295,4294901764,0,0,0,0,16384,0
Success.
hdr: magic 0x00464f53
hdr: type 0
hdr: size 56 bytes
hdr: abi 3:0:0
4607827,0,56,50331648,0,0,0,0,56,65538,0,0,0,0,4294967295,4294901764,0,0,0,0,16384,0
2020-04-21 02:53:45 UTC [COMMAND] aplay -Dhw:0,0 -f S16_LE -c 2 -r 48000 /dev/zero -d 5
[... cut ...]
2020-04-21 02:53:47 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-21 02:53:47 UTC [INFO] Graph of /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg will be generated to /home/ubuntu/sof-test/logs/check-audio-equalizer/2020-04-20-22:53:29-25522
2020-04-21 02:53:47 UTC [INFO] Test Result: PASS!

```
[dmesg0421.log](https://github.com/thesofproject/sof/files/4507443/dmesg0421.log)
[eq_pass.txt](https://github.com/thesofproject/sof/files/4507444/eq_pass.txt)
[sof-logger0421.log](https://github.com/thesofproject/sof/files/4507445/sof-logger0421.log)

xiulipan, 2020-04-21:
@Liviali155 Please try with #2827

Liviali155, 2020-04-21:
Tried with PR #2827,error not seen

closed 2020-04-21 by lgirdwood