thesofproject/sof #2820: Commit: b393c4d1: Volume: Always launch ramp in beginning of copy() operations

lbetlej, 2020-04-20:
This is automatically generated issue to track if changes from commit b393c4d10585a7a29cf00e8408b64cf59c7c7fbb found as a result of comparison https://github.com/thesofproject/sof/compare/master...cml-010-hot-fix2 are upstreamed.

singalsu, 2020-04-20:
@lbetlej The description is very brief, I don't understand. Is there something for me to do for this and similar other automatic issues?

lbetlej, 2020-04-22:
@singalsu Correct, you are expected to check if changes from commit b393c4d are upstreamed and close the issue after verification. If they are not upstreamed then the issue remains open to indicate  that there is work left to do.Check #2682 for background on the process.

singalsu, 2020-04-22:
@lbetlej Thanks for explanation! This was also mentioned today in our team meet with @mengdonglin so we all now now what these are about.

singalsu, 2020-04-22:
This patch is in master.

closed 2020-04-22 by singalsu