thesofproject/sof #2828: [BUG] Sound output does not produce any output, neither directly from speakers nor via headphone jack

Enerccio, 2020-04-21:
**Describe the bug**
A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?
Please also read https://thesofproject.github.io/latest/contribute/process/bug-tracking.html for further information on submitting bugs.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

slawblauciak, 2020-04-21:
We're gonna need more information and a reproduction path for this.
Could you please upload the firmware trace when you encounter the issue again with the SOF logger?
And the entire dmesg dump would be helpful too.

Enerccio, 2020-04-21:
Will do, it was unexpected so what I did was reboot. These logs are from journalctl

Enerccio, 2020-04-21:
Where can I find the logger? I only have these:

```sudo pacman -Ql sof-firmware
sof-firmware /usr/
sof-firmware /usr/lib/
sof-firmware /usr/lib/firmware/
sof-firmware /usr/lib/firmware/intel/
sof-firmware /usr/lib/firmware/intel/sof-tplg/
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-da7219.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-demux-pcm512x.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-dmic-2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-dmic-4ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-dmic-a2ch-b2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-dmic-a2ch-b4ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-dmic-a2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-dmic-a4ch-b2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-dmic-b2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-eq-dmic.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-eq-pcm512x.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-keyword-detect.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-pcm512x-nohdmi.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-rt298.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-src-50khz-pcm512x.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-src-dmic.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-src-pcm512x.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-tdf8532.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-apl-wm8804.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-bdw-rt5640.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-byt-da7213.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-byt-nocodec.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-byt-rt5640.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-byt-rt5645.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-byt-rt5651.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-byt-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cht-max98090.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cht-nocodec.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cht-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cht-src-50khz-pcm512x.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-demux-rt5682-max98357a.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-demux-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-rt1011-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-rt5682-kwd.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-rt5682-max98357a.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-rt700.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-mono-rt715.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-rt711-rt1308-rt715.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cml-src-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cnl-nocodec.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-cnl-rt274.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-glk-da7219-kwd.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-glk-da7219.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-glk-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-idisp-2ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-idisp-4ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic-idisp.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hda-generic.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-hsw-rt5640.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-icl-dmic-4ch.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-icl-nocodec.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-icl-rt5682-kwd.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-icl-rt5682.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-icl-rt700.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-icl-rt711-rt1308-rt715-hdmi.tplg
sof-firmware /usr/lib/firmware/intel/sof-tplg/sof-icl-rt711-rt1308-rt715.tplg
[... 18 lines cut ...]
sof-firmware /usr/lib/firmware/intel/sof/debug/sof-icl.ldc
sof-firmware /usr/lib/firmware/intel/sof/debug/sof-imx8.ldc
sof-firmware /usr/lib/firmware/intel/sof/debug/sof-jsl.ldc
sof-firmware /usr/lib/firmware/intel/sof/debug/sof-kbl.ldc
sof-firmware /usr/lib/firmware/intel/sof/debug/sof-skl.ldc
sof-firmware /usr/lib/firmware/intel/sof/debug/sof-sue.ldc
sof-firmware /usr/lib/firmware/intel/sof/sof-apl.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-bdw.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-byt.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-cfl.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-cht.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-cml.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-cnl.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-hsw.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-icl.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-imx8.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-jsl.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-kbl.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-skl.ri
sof-firmware /usr/lib/firmware/intel/sof/sof-sue.ri
sof-firmware /usr/share/
sof-firmware /usr/share/licenses/
sof-firmware /usr/share/licenses/sof-firmware/
sof-firmware /usr/share/licenses/sof-firmware/LICENSE
```

lgirdwood, 2020-04-21:
@Enerccio I dont think that sof-logger is being shipped as an rpm/deb currently. It needs to be manually built atm. Please clone the SOF FW repo and run ./scripts/host-build-all.sh This will give you all the tooling.

Enerccio, 2020-04-22:
It happened again. Unfortunately, I didn't know what to do with sof-logger and hardlocked myself with save state options so I lost dmesg. But it seems it happens every time I leave computer idle for long time (night sleep). 

@lgirdwood Will check if it happens again today, but can you tell me what exactly I need to do with that sof-logger?

lgirdwood, 2020-04-22:
@Enerccio sof-logger details are here https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html
Fwiw, we have a v1.4.3 coming out this week (to fix a reported hang, could be the same as your issue) so lets retest with v1.4.3. v1.5 is also due out this week too, so will also be worth testing.

Enerccio, 2020-04-22:
Gotcha @lgirdwood, how do I check which ldc file I use though?

slawblauciak, 2020-04-23:
I believe it'd be sof-icl.ldc

Enerccio, 2020-04-24:
It happened again, unfortunately I can't use git logger:

```./sof-logger -l /usr/lib/firmware/intel/sof/debug/sof-icl.ldc -e -o trace.log
error: abi version in /sys/kernel/debug/sof/fw_version file does not coincide with abi version used by logger.
error: logger ABI Version is 4:2:0
error: version_file ABI Version is 3:11:0
```

maybe sof-fimware should be bundled with logger by default :/ 

here is the dmesg at least

[dmesg.log](https://github.com/thesofproject/sof/files/4526983/dmesg.log)

slawblauciak, 2020-04-28:
Ah, apologies. We neglected to mention that you'll need to build it from the right branch, since you're using 1.4.2 release.

@lgirdwood which release branch is appropriate for that version? Is it 1.4-stable?

semitom, 2020-05-05:
I'm also affected by this issue on my lenovo yoga c940 running arch linux kernel 5.6.10 using sof-firmware 1.4.2-1.

It is odd because the issue started between several reboots while I was trying to debug hanging shutdowns. Right after I rebooted without the `snd_hda_intel.dmic_detect=0` boot flag, the sound stopped working and journalctl got spammed with the mentioned messages.

```
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: error: failed to set dai config for iDisp3
May 05 16:09:32 laptop kernel: sof-audio-pci 0000:00:1f.3: ASoC: can't set iDisp3 Pin hw params: -110
May 05 16:09:32 laptop kernel:  iDisp3: ASoC: hw_params BE failed -110
May 05 16:09:32 laptop kernel:  HDMI3: ASoC: hw_params BE failed -110
```

Since then I have this sound issue around 50% of the boots with or without `snd_hda_intel.dmic_detect` kernel parameter set. The strangest part is that the issue of the hanging shutdowns is gone when the sound does not work, so 50% of the boots I have no sound and the other 50% of the time shutdown/reboot hangs.

I installed and ran the sof-logger, hopefully it will give some insight.
<details>
<summary>sof-logger output</summary>

```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         DMIC         107809827.500000 107809824.000000 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809840.208333        12.708333 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809851.875000        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809864.583333        12.708333 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809876.302083        11.718750 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809888.958333        12.656250 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809900.625000        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809913.281250        12.656250 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809924.947917        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809937.604167        12.656250 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809949.218750        11.614583 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809961.927083        12.708333 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809973.645833        11.718750 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809986.354167        12.708333 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107809997.968750        11.614583 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810010.572917        12.604167 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810022.239583        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810034.895833        12.656250 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810046.562500        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810059.166667        12.604167 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810070.833333        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810083.437500        12.604167 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810095.104167        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107810107.656250        12.552083 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
[... cut ...]
    0      1         DMIC         107847104.947917        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107847117.604167        12.656250 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107847129.270833        11.666667 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
    0      1         DMIC         107847142.031250        12.760417 rs/intel/cavs/dmic.c:1490	dmic_irq_handler() error: full fifo A or PDM overrrun
```
</details>

lgirdwood, 2020-05-07:
I think the v1.4.3 was tagged yesterday, it should be available in sof-bin after it's signed.

Enerccio, 2020-05-13:
I keep getting random crashes even when in use now:

```
[38381.077085] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[38381.077105] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[38381.077250] sof-audio-pci 0000:00:1f.3: error: can't enter idle
[38381.077250] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[38381.077251] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:50
[38381.077252] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[38381.077253] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060b25 SAR     0x00000000
[38381.077254] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d0a6 EPC3     0x00000000 EPC4    0x00000000
[38381.077254] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[38381.077255] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[38381.077256] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[38381.077256] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe04c0f0
[38381.077258] sof-audio-pci 0000:00:1f.3: 0xbe04c0f0: be00f8f0 be04c120 be052180 00000001
[38381.077259] sof-audio-pci 0000:00:1f.3: 0xbe04c0f4: 00000000 00000000 00000032 00000000
[38381.077259] sof-audio-pci 0000:00:1f.3: 0xbe04c0f8: 3f962000 951c1b92 000c0800 00000000
[38381.077260] sof-audio-pci 0000:00:1f.3: 0xbe04c0fc: 0dead000 00000000 d661c818 ffff9cf1
[38381.077261] sof-audio-pci 0000:00:1f.3: 0xbe04c100: c0e52044 ffffffff b961c780 ffff9cf1
[38381.077261] sof-audio-pci 0000:00:1f.3: 0xbe04c104: 00000004 00000000 00000000 00000000
[38381.077262] sof-audio-pci 0000:00:1f.3: 0xbe04c108: 00a4fbd0 ffffbd66 b961c780 ffff9cf1
[38381.077263] sof-audio-pci 0000:00:1f.3: 0xbe04c10c: 00000000 00000000 00000000 00000000
```

victortrac, 2020-05-15:
Exact same issue here.

- X1 Carbon, 7th Gen. 
- Linux carbon 5.6.11-arch1-1 #1 SMP PREEMPT Wed, 06 May 2020 17:32:37 +0000 x86_64 GNU/Linux
- sof-firmware 1.4.2-1

It seems to come and go randomly since the 5.6.x kernels. All sound breaks, including bluetooth.

lgirdwood, 2020-05-18:
@Enerccio @victortrac can you guys try v1.4.3 or v1.5. They have now been signed and can be found here
https://github.com/thesofproject/sof-bin

Please see the v1.4.3 and v1.5 branches.

timur-enikeev, 2020-06-22:
Hello. I have the same problem. Dell Inspirion 5391.

lspci item of sound card:

```
00:1f.3 Multimedia audio controller: Intel Corporation Device 02c8
        Subsystem: Dell Device 096e
        Flags: bus master, fast devsel, latency 64, IRQ 146
        Memory at a121c000 (64-bit, non-prefetchable) [size=16K]
        Memory at a1000000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [50] Power Management version 3
        Capabilities: [80] Vendor Specific Information: Len=14 <?>
        Capabilities: [60] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Kernel driver in use: sof-audio-pci
        Kernel modules: snd_hda_intel, snd_sof_pci
```
Ubuntu 20.04.

Is there any solution? It's very annoying problem. BTW this laptop had Ubuntu 18.04 preinstalled. But I don't work with the preinstalled version then can't check does it have this bug.

timur-enikeev, 2020-06-22:
@lgirdwood 

Hello. How to install the new version of sof? I see that there is different directory structure.

lgirdwood, 2020-06-23:
@timur-enikeev all sof fiirmware binaries are installed under
```/lib/firmware/intel/sof```
as per the layout in the sof-bin repo. The topologies are in
```/lib/firmware/intel/sof-tplg/```
You should just be able to follow the instructions in sof-bin. Please aslo provide the output of dmesg.

timur-enikeev, 2020-06-27:
@lgirdwood Hello. I have installed the 1.4.3 version, but there is the same problem. dmesg:

```[301980.862523] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[301980.862541] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[301980.862799] sof-audio-pci 0000:00:1f.3: error: can't enter idle
[301980.862804] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[301980.862808] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:50
[301980.862813] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[301980.862818] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060925 SAR     0x00000000
[301980.862823] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d486 EPC3     0x00000000 EPC4    0x00000000
[301980.862828] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[301980.862832] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[301980.862836] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[301980.862840] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe04f5b0
[301980.862848] sof-audio-pci 0000:00:1f.3: 0xbe04f5b0: be00fd90 be04f5e0 be062180 00000001
[301980.862853] sof-audio-pci 0000:00:1f.3: 0xbe04f5b4: 00000000 00000000 00000032 00000000
[301980.862858] sof-audio-pci 0000:00:1f.3: 0xbe04f5b8: dcd2df00 ce18ec61 97689818 ffff9146
[301980.862864] sof-audio-pci 0000:00:1f.3: 0xbe04f5bc: 0093fe70 ffffaa17 c0c605f9 ffffffff
[301980.862869] sof-audio-pci 0000:00:1f.3: 0xbe04f5c0: b0d16be8 ffffffff a11e2f80 ffff9146
[301980.862873] sof-audio-pci 0000:00:1f.3: 0xbe04f5c4: 00000000 00000000 00000000 00000000
[301980.862878] sof-audio-pci 0000:00:1f.3: 0xbe04f5c8: 00393b98 ffffaa17 0093ff48 ffffaa17
[301980.862883] sof-audio-pci 0000:00:1f.3: 0xbe04f5cc: 00000000 00000000 00000000 00000000
[301980.862887] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[310312.030105] SGI XFS with ACLs, security attributes, realtime, no debug enabled
```

And how to check, have I installed this version correctly? Can I receive the current version loaded?

timur-enikeev, 2020-06-29:
@lgirdwood  there is the same problem with 1.5 version.

I have installed the new version of SOF according to this instruction: https://wiki.debian.org/InstallingDebianOn/Thinkpad/X1 Carbon 7th Gen

lgirdwood, 2020-06-29:
@slawblauciak any comments ? Looks like FW is looping forever and agent is complaining.
@timur-enikeev can you share the full dmesg, we need to see the full SOF logs. Thanks.

timur-enikeev, 2020-06-29:
@lgirdwood Dmesg after the fail? Or in any time (if there weren't such fails since last boot)?

lgirdwood, 2020-06-30:
@timur-enikeev the full dmesg showing the SOF driver loading and operating. You can shorten it if you like by greping for "sof-" etc.

timur-enikeev, 2020-06-30:
@lgirdwood 
Current dmesg (without fail):

```
$ dmesg | grep sof-
[   16.528441] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   16.534190] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   16.534199] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[   16.534411] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   16.534417] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   16.534569] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   16.536160] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   16.544900] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   16.544901] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   16.567735] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   16.567736] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   16.660158] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:0-7e6cd
[   16.660161] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:10:0
[   16.660162] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   16.660374] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   16.666034] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:10:0
[   16.666037] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[   16.666047] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   16.667079] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   16.667083] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   16.670305] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   16.670309] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   16.671226] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   16.671229] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   16.672208] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   16.672212] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   16.672215] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   16.672218] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   16.672220] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   16.672223] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   16.673155] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   16.673158] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   16.674145] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   16.674147] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   16.675077] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   16.685626] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   16.804730] input: sof-hda-dsp Headphone Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input15
[   16.804806] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input16
[   16.804878] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[   16.804937] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[   22.163236] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   22.260506] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  111.199233] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  111.297926] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  129.623016] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  129.724259] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  591.195107] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  591.295093] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1357.290222] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1357.389268] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1742.622001] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1742.722732] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1766.858173] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1766.958901] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1821.158594] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1821.259404] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1866.394396] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1866.495387] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 2884.152374] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 2884.252322] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 3250.599246] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 3250.699946] sof-audio-pci 0000:00:1f.3: firmware boot complete
[... cut ...]
[ 4209.531729] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 5168.070736] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 5168.171473] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 5211.299212] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 5211.401457] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 5229.323350] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 5229.424612] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

lgirdwood, 2020-07-01:
@timur-enikeev thanks, I can see you are running v1.5 (which had an issue on some devices), can you try v1.5.1

https://github.com/thesofproject/sof-bin/tree/stable-v1.5.1

timur-enikeev, 2020-07-01:
@lgirdwood I will wait the next fail with the current version, and then install the 1.5.1

mikerogerz, 2020-07-11:
Still happening on 1.5.1, although less frequently:

`[172646.062276] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120`
`[172646.062299] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005`
`[172646.062546] sof-audio-pci 0000:00:1f.3: error: can't enter idle`
`[172646.062550] sof-audio-pci 0000:00:1f.3: error: trace point 00004000`
`[172646.062553] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:62`
`[172646.062557] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops`
`[172646.062562] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000`
`[172646.062567] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d1fe EPC3     0x00000000 EPC4    0x00000000`
`[172646.062570] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000`
`[172646.062574] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000`
`[172646.062578] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222`
`[172646.062581] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05a110`
`[172646.062587] sof-audio-pci 0000:00:1f.3: 0xbe05a110: 9e0ca000 00000000 00000000 000001a0`
`[172646.062592] sof-audio-pci 0000:00:1f.3: 0xbe05a114: 00000000 00000000 0000003e be064400`
`[172646.062597] sof-audio-pci 0000:00:1f.3: 0xbe05a118: 117dd600 ed92ef8a 64e3b738 ffff9303`
`[172646.062602] sof-audio-pci 0000:00:1f.3: 0xbe05a11c: 00000000 ffff9303 00000000 303a3030`
`[172646.062606] sof-audio-pci 0000:00:1f.3: 0xbe05a120: 00000078 00000000 64e3b6d8 ffff9303`
`[172646.062611] sof-audio-pci 0000:00:1f.3: 0xbe05a124: 65361358 ffff9303 6874ae40 ffff9303`
`[172646.062615] sof-audio-pci 0000:00:1f.3: 0xbe05a128: 620e6218 ffff9303 026ffaaf ffffa51b`
`[172646.062619] sof-audio-pci 0000:00:1f.3: 0xbe05a12c: 620e6220 ffff9303 826ff9f8 ffffa51a`
`[172646.062642] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00`
`[172646.062646] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000`
`[172646.062663] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003`
`[172646.062667] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers`
`[172646.062672] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture`
`[172646.062677] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110`

timur-enikeev, 2020-07-11:
As I see, I can't reproduce it with kernel 5.4.0 from ubuntu and sof 1.5. But with newer mainline kernels there is the same problem.

mikerogerz, 2020-07-11:
@timur-enikeev That's what I'm running, Ubuntu 20.04 with the 5.4.0 kernel, and sof 1.5.1. It was fine for a week or so and then happened again.

lgirdwood, 2020-07-14:
@slawblauciak fyi - any idea here - it looks like we are spinning somewhere if agent cant idle.

timur-enikeev, 2020-07-14:
```
# dmesg | grep sof-
[   14.140392] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[   14.141640] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[   14.141652] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[   14.144534] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   14.144542] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   14.145613] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   14.146411] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   14.154882] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   14.154884] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   14.181729] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   14.181731] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   14.266900] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:0-7e6cd
[   14.266903] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:10:0
[   14.266904] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   14.267149] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   14.416884] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:10:0
[   14.416886] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[   14.416891] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   14.425489] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   14.425492] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   14.426937] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   14.426939] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   14.427946] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   14.427949] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   14.428987] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   14.428989] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   14.428991] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   14.428993] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   14.428995] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   14.428997] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   14.430034] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   14.430037] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   14.432163] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   14.432165] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   14.433601] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   14.442491] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   14.566053] input: sof-hda-dsp Headphone Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input23
[   14.566085] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input24
[   14.566111] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input25
[   14.566137] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input26
[  738.387832] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  738.487844] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  867.796805] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  867.896911] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  954.661667] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  954.761115] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1378.750629] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1378.847870] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1628.555868] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1628.656674] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 1773.512590] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 1773.611913] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 6368.714903] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 6368.815068] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 6410.039466] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 6410.138202] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 6436.919836] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 6437.022647] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 6742.438909] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 6742.539503] sof-audio-pci 0000:00:1f.3: firmware boot complete
[ 6761.211390] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 6761.314198] sof-audio-pci 0000:00:1f.3: firmware boot complete
[... cut ...]
[263023.736852] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[263023.736854] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[263023.736865] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[263023.736868] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[263024.890104] sof-audio-pci 0000:00:1f.3: firmware boot complete
```
Last fail (sof 1.5.0, kernel 5.4.0-40-generic ubuntu)

slawblauciak, 2020-07-15:
@lgirdwood I'm afraid not, I don't know much about how agent works exactly.

lgirdwood, 2020-07-15:
@slawblauciak ok, can you add to scrub - agent panic means we have a while(1) {} condition somewhere in FW e.g. waiting on HW state change somewhere.
@timur-enikeev @mikerogerz @Enerccio is this always happening after resume from S3 i.e. opening laptop lid or can it happen at any time ? S3 use case has different PM flow that may mean FW is waiting for HW state change (that's not happening)

mikerogerz, 2020-07-15:
@slawblauciak In my case, there's no suspend/resume involved.. the laptop stays on 24/7.

lgirdwood, 2020-07-16:
@mikerogerz do you have audio playing 24/7. DSP enters D3 speculatively if no audio for 3 seconds to prolong laptop battery. This could still be PM related if we are not coming out of D3 cleanly.
@ranj063 @kv2019i do we have any simple to disable runtime PM in the driver (without disabling kernel wide).

plbossart, 2020-07-16:
> @mikerogerz do you have audio playing 24/7. DSP enters D3 speculatively if no audio for 3 seconds to prolong laptop battery. This could still be PM related if we are not coming out of D3 cleanly. [...]

yes we do, you can add this to /etc/modprobe.d/alsa-base.conf (or any other .conf file in that directory)

options snd-sof-pci sof_pci_debug=1

mikerogerz, 2020-07-16:
> @mikerogerz do you have audio playing 24/7. DSP enters D3 speculatively if no audio for 3 seconds to prolong laptop battery. This could still be PM related if we are not coming out of D3 cleanly. [...]

@lgirdwood That was something I noticed (although when I was back on stock sof in Ubuntu 20.04, where it was crashing constantly), and a workaround was to leave music on. When it crashed on sof 1.5.1 recently, it was a single alert sound (deleting a file in Gnome) that did it, and completely locked everything up. It hasn't happened ever when audio is already playing.

lgirdwood, 2020-07-17:
@mikerogerz ok, can you try and disable runtime PM as described by @plbossart above and let us know if that fixes things for you. This will let us know if it's PM specific on your device.

mikerogerz, 2020-07-17:
@lgirdwood Done, and thanks. Like I mentioned, since 1.5.1 it hasn't happened as often, but I'll keep everyone posted.

Enerccio, 2020-07-20:
> @slawblauciak ok, can you add to scrub - agent panic means we have a while(1) {} condition somewhere in FW e.g. waiting on HW state change somewhere.
> @timur-enikeev @mikerogerz @Enerccio is this [...]

For me it happens randomly and I barely close lid at all. It can just happen on it's own, usually when I want to play new source it will crash and not play it. Sometimes I can trigger it by plugging headphones or unplugging them.

lgirdwood, 2020-07-20:
@Enerccio ok, so these also sound like PM transitions, can you disable the DSP runtime PM as suggested by @plbossart and let us know it it fixes or otherwise.

Enerccio, 2020-07-20:
> @Enerccio ok, so these also sound like PM transitions, can you disable the DSP runtime PM as suggested by @plbossart and let us know it it fixes or otherwise.

`options snd-sof-pci sof_pci_debug=1` this?

plbossart, 2020-07-20:
> > @Enerccio ok, so these also sound like PM transitions, can you disable the DSP runtime PM as suggested by @plbossart and let us know it it fixes or otherwise.
> 
> `options snd-sof-pci sof_pci_debug=1` this?

yes, that's the best way to disable pm_runtime transitions.

Enerccio, 2020-07-21:
It crashed again even with `options snd-sof-pci sof_pci_debug=1`

[dmesg.txt](https://github.com/thesofproject/sof/files/4954645/dmesg.txt)

plbossart, 2020-07-21:
@lgirdwood do we really throw a panic for a timeout?

````c
	/* panic timeout */
	if (sa->panic_on_delay && delta > sa->panic_timeout)
		panic(SOF_IPC_PANIC_IDLE);
````

mikerogerz, 2020-07-21:
> It crashed again even with `options snd-sof-pci sof_pci_debug=1`
> 
> [dmesg.txt](https://github.com/thesofproject/sof/files/4954645/dmesg.txt)

Me too, just now (also with the debug option set). This was with no audio playing, and then a random sound triggered for a dialog in Gnome.

`[94981.550671]  Analog Playback and Capture: ASoC: hw_params BE failed -110`
`[94981.550676]  HDA Analog: ASoC: hw_params BE failed -110`
`[94982.062309] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120`
`[94982.062334] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005`
`[94982.062592] sof-audio-pci 0000:00:1f.3: error: can't enter idle`
`[94982.062596] sof-audio-pci 0000:00:1f.3: error: trace point 00004000`
`[94982.062600] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:62`
`[94982.062604] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops`
`[94982.062610] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000`
`[94982.062615] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d1fe EPC3     0x00000000 EPC4    0x00000000`
`[94982.062619] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000`
`[94982.062624] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000`
`[94982.062628] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222`
`[94982.062632] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05a110`
`[94982.062640] sof-audio-pci 0000:00:1f.3: 0xbe05a110: 9e0ca000 00000000 00000000 000001c0`
`[94982.062646] sof-audio-pci 0000:00:1f.3: 0xbe05a114: 00000000 00000000 0000003e be064400`
`[94982.062651] sof-audio-pci 0000:00:1f.3: 0xbe05a118: f7461b00 63f0ebba e45dceb8 ffff8a63`
`[94982.062656] sof-audio-pci 0000:00:1f.3: 0xbe05a11c: 00000000 ffff8a63 00000000 303a3030`
`[94982.062661] sof-audio-pci 0000:00:1f.3: 0xbe05a120: 00000078 00000000 e45dce58 ffff8a63`
`[94982.062666] sof-audio-pci 0000:00:1f.3: 0xbe05a124: ea197358 ffff8a63 eabb0a40 ffff8a63`
`[94982.062670] sof-audio-pci 0000:00:1f.3: 0xbe05a128: e50e2c18 ffff8a63 c1e8faaf ffff9830`
`[94982.062675] sof-audio-pci 0000:00:1f.3: 0xbe05a12c: e50e2c20 ffff8a63 41e8f9f8 ffff9830`
`[94982.062701] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00`
`[94982.062705] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000`
`[94982.062726] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003`
`[94982.062730] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers`
`[94982.062736] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture`
`[94982.062742] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110`
`[94982.062763]  Analog Playback and Capture: ASoC: hw_params BE failed -110`
`[94982.062768]  HDA Analog: ASoC: hw_params BE failed -110`

lgirdwood, 2020-07-22:
> @lgirdwood do we really throw a panic for a timeout?
> 
> ```c
> 	/* panic timeout */
> 	if (sa->panic_on_delay && delta > sa->panic_timeout)
> 		panic(SOF_IPC_PANIC_IDLE);
> ```

Yes we do - this is for **long** timeouts where we cant enter WFI state i.e. something unrecoverable has happened at IRQ runlevel. If we did not do this then we fail at next IPC, this is mean to provide more context so FW devs can see the stack and see where we are spinning or which IRQ is continually repeating.

lgirdwood, 2020-07-22:
@slawblauciak kernel PM is OFF here and it looks like playing a system sound is enough to trigger the agent. i.e we are spinning on some HW state transition forever (we shouldn't be) or is LL scheduler spinning .....

Enerccio, 2020-07-22:
is there a way to restart sof? Without reboot?

slawblauciak, 2020-07-22:
I wonder if Karol's DW DMA changes are in this version of the FW, they could've caused some FW hanging-for-too-long issues, but it's been a while and I kinda forgot what it all was exactly about. I'll have to check.

plbossart, 2020-07-22:
> is there a way to restart sof? Without reboot?

With the latest code, if you do a rmmod snd-sof-pci; modprobe snd-sof-pci that would reboot the DSP and re-install firmware. However we had a series of issues in previous versions so your mileage may vary. If you want to do things cleanly, try the scripts sof_remove.sh and sof_insert.sh from here: https://github.com/thesofproject/sof-test/tree/master/tools/kmod

timur-enikeev, 2020-07-26:
> > > @Enerccio ok, so these also sound like PM transitions, can you disable the DSP runtime PM as suggested by @plbossart and let us know it it fixes or otherwise.
> > 
> > 
> > `options snd-sof-pci sof_pci_debug=1` this?
> 
> yes, that's the best way to disable pm_runtime transitions.

Will it prevent this problem? I have added, rebooted, and caugth the same problem.

timur-enikeev, 2020-07-26:
>With the latest code, if you do a rmmod snd-sof-pci; modprobe snd-sof-pci that would reboot the DSP and re-install firmware.

With the latest code of what? Kernel or sof-frimware?
I have 1.5.1 sof and 5.4 kernel, but
```
# rmmod snd-sof-pci
rmmod: ERROR: Module snd_sof_pci is in use
```

mikerogerz, 2020-07-29:
Unsure if related, but happened again, this time while audio was playing. Was able to get audio back after using sof_remove.sh & sof_insert.sh as @plbossart recommended.

`[165784.806228] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120`
`[165784.806254] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005`
`[165784.806486] sof-audio-pci 0000:00:1f.3: error: can't enter idle`
`[165784.806490] sof-audio-pci 0000:00:1f.3: error: trace point 00004000`
`[165784.806494] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:62`
`[165784.806498] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops`
`[165784.806503] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000`
`[165784.806508] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d1fe EPC3     0x00000000 EPC4    0x00000000`
`[165784.806512] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000`
`[165784.806516] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000`
`[165784.806521] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222`
`[165784.806524] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05a110`
`[165784.806531] sof-audio-pci 0000:00:1f.3: 0xbe05a110: be05a140 00000001 be013760 00000001`
`[165784.806537] sof-audio-pci 0000:00:1f.3: 0xbe05a114: 00000000 00000000 0000003e be064400`
`[165784.806542] sof-audio-pci 0000:00:1f.3: 0xbe05a118: 74337f00 4755723c b3405738 ffff962c`
`[165784.806547] sof-audio-pci 0000:00:1f.3: 0xbe05a11c: 00000000 ffff962c 00000000 ffff962a`
`[165784.806552] sof-audio-pci 0000:00:1f.3: 0xbe05a120: 00000078 00000000 b34056d8 ffff962c`
`[165784.806557] sof-audio-pci 0000:00:1f.3: 0xbe05a124: eb019358 ffff962c 6445e110 ffff962c`
`[165784.806562] sof-audio-pci 0000:00:1f.3: 0xbe05a128: 81ddc018 ffff962d c13e3cc0 ffffa981`
`[165784.806567] sof-audio-pci 0000:00:1f.3: 0xbe05a12c: 81ddc020 ffff962d c13e3d08 ffffa981`
`[165784.806593] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00`
`[165784.806597] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000`
`[165784.806617] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003`
`[165784.806621] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers`
`[165784.806627] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture`
`[165787.810209] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76`
`[165787.810232] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005`
`[165787.810486] sof-audio-pci 0000:00:1f.3: error: can't enter idle`
`[165787.810489] sof-audio-pci 0000:00:1f.3: error: trace point 00004000`
`[165787.810492] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:62`
`[165787.810496] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops`
`[165787.810500] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000`
`[165787.810504] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d1fe EPC3     0x00000000 EPC4    0x00000000`
`[165787.810507] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000`
`[165787.810511] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000`
`[165787.810514] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222`
`[165787.810517] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe05a110`
`[165787.810523] sof-audio-pci 0000:00:1f.3: 0xbe05a110: be05a140 00000001 be013760 00000001`
`[165787.810528] sof-audio-pci 0000:00:1f.3: 0xbe05a114: 00000000 00000000 0000003e be064400`
`[165787.810532] sof-audio-pci 0000:00:1f.3: 0xbe05a118: 33acd600 80b2063e b3405738 ffff962c`
`[165787.810536] sof-audio-pci 0000:00:1f.3: 0xbe05a11c: 00000000 ffff962c 00000000 00000000`
`[165787.810540] sof-audio-pci 0000:00:1f.3: 0xbe05a120: 0000004c 00000000 b34056d8 ffff962c`
`[165787.810544] sof-audio-pci 0000:00:1f.3: 0xbe05a124: eb019018 ffff962c 00000001 00000000`
`[165787.810548] sof-audio-pci 0000:00:1f.3: 0xbe05a128: 00000001 0000004c 40010000 00000000`
`[165787.810552] sof-audio-pci 0000:00:1f.3: 0xbe05a12c: 00000000 00000000 00000000 00000000`
`[165787.810576] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00`
`[165787.810579] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000`
`[165787.810595] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003`
`[165787.810599] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend`

lgirdwood, 2020-08-04:
@slawblauciak can someone look at the stack trace and see whats preventing the timer scheduling correctly here. I would also say that we should have a FW register that the driver can set to enable this check and panic for debug otherwise we try and keep running (as XRUN handling or stream reset may be enough to recover in some circumstances).

balent, 2020-08-04:
Do you need more information or do you have enough to debug the issue?

I tried to build sof-logger but failed. Are there any instructions how to build it? I'm trying to run `./scripts/host-build-all.sh` but I'm getting error:

```
[100%] Linking C executable testbench
/usr/bin/cmake -E cmake_link_script CMakeFiles/testbench.dir/link.txt --verbose=1
/usr/bin/cc CMakeFiles/testbench.dir/testbench.c.o CMakeFiles/testbench.dir/alloc.c.o CMakeFiles/testbench.dir/common_test.c.o CMakeFiles/testbench.dir/file.c.o CMakeFiles/testbench.dir/ipc.c.o CMakeFiles/testbench.dir/schedule.c.o CMakeFiles/testbench.dir/ll_schedule.c.o CMakeFiles/testbench.dir/edf_schedule.c.o CMakeFiles/testbench.dir/panic.c.o CMakeFiles/testbench.dir/timer.c.o CMakeFiles/testbench.dir/topology.c.o CMakeFiles/testbench.dir/trace.c.o -o testbench  -Wl,-rpath,/home/robko/sof/tools/testbench/build_testbench/sof_ep/install/lib:/home/robko/sof/tools/testbench/build_testbench/sof_parser/install/lib: -ldl -lm sof_ep/install/lib/libsof.so sof_parser/install/lib/libsof_tplg_parser.so 
/usr/bin/ld: CMakeFiles/testbench.dir/topology.c.o:/home/robko/sof/tools/testbench/topology.c:24: multiple definition of `lib_table'; CMakeFiles/testbench.dir/testbench.c.o:/home/robko/sof/tools/testbench/testbench.c:20: first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/testbench.dir/build.make:273: testbench] Error 1
make[2]: Leaving directory '/home/robko/sof/tools/testbench/build_testbench'
make[1]: *** [CMakeFiles/Makefile2:157: CMakeFiles/testbench.dir/all] Error 2
make[1]: Leaving directory '/home/robko/sof/tools/testbench/build_testbench'
make: *** [Makefile:152: all] Error 2
```

I'm reproducing this issue with:

```
hw: X1 Carbon gen7
dist: Arch Linux
linux: 5.7.11.arch1-1
sof-firmware: 1.5.1-1
```

lgirdwood, 2020-08-05:
@ymdatta @singalsu can you guys report on the testbench build failure. IIRC, this should be part of CI now.

lgirdwood, 2020-08-05:
@balent I guess you are using HEAD ? There have been some recent testbench fixes, but any failures should have been caught by VI. Can you try and build tools from the stable v1.5 branch ?

balent, 2020-08-05:
> @balent I guess you are using HEAD ? There have been some recent testbench fixes, but any failures should have been caught by VI. Can you try and build tools from the stable v1.5 branch ?

I was able to run `../scripts/build-tools.sh` (from new dir) and build `sof-logger`.

The `./scripts/host-build-all.sh` is not working for me with `stable-v1.5` either. Same error.

What parameters should I use when running `sof-logger`? Which ldc file?

Thanks!

lgirdwood, 2020-08-06:
@balent the LDC files contain the debugging strings and are extracted from the ELF image as part of the FW binary build. i.e. you need to use the same LDC file that was produced as part of the FW build. The SOF LDC files are release with the binaries in SOF bin repo. Please find the v1.5 .1 LDC and binaries here
https://github.com/thesofproject/sof-bin/tree/stable-v1.5.1/lib/firmware/intel/sof/v1.5.1
The logger has some docs here
https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

[20 more replies]

closed 2021-06-10 by plbossart