thesofproject/sof #2842: [BUG][CML] sof fails during the initialization

hugh712, 2020-04-24:
**Describe the bug**
"sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110" occurred during system boot up, and sof failed to initialize. 

**To Reproduce**
1.System boot up.
2. Check dmesg

**Reproduction Rate**
100%

**Expected behavior**
sof can be loaded and initialized normally.

**Actual result**
Error message shows, and sof failed to initialize.

**dmesg**
[hp-sof-fail-dmesg.txt](https://github.com/thesofproject/linux/files/4527161/hp-sof-fail-dmesg.txt)

mengdonglin, 2020-04-24:
This issue happens on an HP CML laptop.  It seems like https://github.com/thesofproject/sof/issues/2136, which is caused by back ME. How to check on the ME side?

Dmesg for this issue:
[    9.030719] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    9.030975] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    9.030987] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    9.031154] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    9.031246] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    9.036542] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    9.084751] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    9.084752] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    9.087302] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    9.087303] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
...
[    9.387350] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    9.387356] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    9.388258] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    9.388259] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    9.577296] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.577297] Bluetooth: BNEP filters: protocol multicast
[    9.577301] Bluetooth: BNEP socket layer initialized
[    9.584861] NET: Registered protocol family 38
[    9.688598] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    9.688604] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    9.689193] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    9.689194] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    9.989211] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    9.989217] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    9.989232] sof-audio-pci 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -110
[    9.989248] sof-audio-pci 0000:00:1f.3: error: status = 0xffffffff panic = 0xffffffff
[    9.989251] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    9.989252] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   10.029072] iwlwifi 0000:00:14.3: Applying debug destination EXTERNAL_DRAM
[   10.039692] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   10.039694] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask f
[   10.039779] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110

cooper-chiou, 2020-04-24:
Some note from OSV in sof-logger failure.

We tried to run the sof-logger to collect the log, but it failed and reported "error: Unable to open in file /sys/kernel/debug/sof/trace", so it looks like this debugging too depends on the sof driver initialize without error and could work, but for our case, the sof fails during the initialization, so far we has not run the sof-logger successfully.

mengdonglin, 2020-04-24:
@hugh712 @cooper-chiou Please test with latest sof-dev kernel with SOF FW v1.4.2 binary.

hugh712, 2020-04-25:
@mengdonglin 

still doesn't work, here is the dmesg
[hp-sof-failed-dmesg-2.log](https://github.com/thesofproject/sof/files/4532468/hp-sof-failed-dmesg-2.log)

mengdonglin, 2020-04-26:
The same error pattern in the above dmesg of sof-dev

<details>
 <summary>dmesg</summary>

````
    7.957785] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    7.957787] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    7.957939] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 2
[snip]
[    8.260740] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    8.260746] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    8.562185] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    8.562195] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    8.863623] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    8.863635] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    8.863651] sof-audio-pci 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -110
[    8.863670] sof-audio-pci 0000:00:1f.3: error: status = 0xffffffff panic = 0xffffffff
[    8.863678] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    8.863678] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110

[    8.914219] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    8.914220] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask f
[    9.011671] iwlwifi 0000:00:14.3: FW already configured (0) - re-configuring
[    9.014520] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[    9.109750] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
````
</details>

RanderWang, 2020-04-26:
When error occurs, the ROM initialization is failed. It is very likely that the ROM is waiting for ME.
So changing FW or driver don't make effect.

mengdonglin, 2020-05-09:
Customer confirmed this issue cannot be reproduced after upgrading the whole system.

Maybe there is some hardware issue for this HW sample, or its BIOS doesn’t initialize the audio PCI device properly. And the issue seems to be fixed by upgraded system.

Output from lspci -v on a failed sample:
00:1f.3 Audio device: Intel Corporation Device 06c8 (prog-if 80)
       Subsystem: Hewlett-Packard Company Device 871b
       Flags: fast devsel, IRQ 255
       Memory at 4000100000 (64-bit, non-prefetchable) [disabled] [size=16K]
       Memory at 4000000000 (64-bit, non-prefetchable) [disabled] [size=1M]
       Capabilities: <access denied>
       Kernel modules: snd_hda_intel, snd_sof_pci

Output from lspci -v on another CML-H based laptop, on which SOF audio works well:
00:1f.3 Multimedia audio controller: Intel Corporation Comet Lake PCH cAVS
              Subsystem: xxxxxx
              Flags: bus master, fast devsel, latency 64, IRQ 181
              Memory at 609b118000 (64-bit, non-prefetchable) [size=16K]
              Memory at 609b000000 (64-bit, non-prefetchable) [size=1M]
              Capabilities: <access denied>
              Kernel driver in use: sof-audio-pci

closed 2020-05-09 by mengdonglin