thesofproject/sof #2879: firmware is not installed

sealse, 2020-04-24:
**Describe the bug**
On page https://bugzilla.kernel.org/show_bug.cgi?id=206085 I was told that it is necessary to install your firmware

cmake not worked

>cmake -DTOOLCHAIN=xtensa-hsw-elf -DROOT_DIR=`pwd`/../../xtensa-root/xtensa-hsw-elf ..
>-- Preparing Xtensa toolchain
>-- The C compiler identification is unknown
>-- The ASM compiler identification is GNU
>-- Didn't find assembler
>CMake Error at CMakeLists.txt:39 (project):
>  The CMAKE_C_COMPILER:
>
>    xtensa-hsw-elf-gcc
>
>  is not a full path and was not found in the PATH.
>
>  Tell CMake where to find the compiler by setting either the environment
>  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
>  the compiler, or to the compiler name if it is in the PATH.
>
>
>CMake Error at CMakeLists.txt:39 (project):
>  No CMAKE_ASM_COMPILER could be found.
>
>  Tell CMake where to find the compiler by setting either the environment
>  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
>  to the compiler, or to the compiler name if it is in the PATH.
>
>
>-- Configuring incomplete, errors occurred!
>See also "/home/sealse/Downloads/sof-1.4.3/build/CMakeFiles/CMakeOutput.log".
>See also "/home/sealse/Downloads/sof-1.4.3/build/CMakeFiles/CMakeError.log".
> 

**To Reproduce**
>cmake -DTOOLCHAIN=xtensa-hsw-elf -DROOT_DIR=`pwd`/../../xtensa-root/xtensa-hsw-elf ..

**Reproduction Rate**
all the time

**Expected behavior**
Firmware compiled

**Impact**
The sound does not work

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.4 - 5.7
    * SOF: I don't known
2) Name of the topology file
    * Topology:  I don't known
3) Name of the platform(s) on which the bug is observed.
    * Platform: Intel® Core™ i9-9980HK Processor

lgirdwood, 2020-05-07:
@sealse have you read the getting started guide ?
https://thesofproject.github.io/latest/getting_started/index.html
There are some dependencies that need installed.
Are you looking for FW binaries or tools ?

sealse, 2020-05-11:
Thank you very much for the answer, but I managed to solve the problem with
sound on my laptop by putting your library from bin, it’s probably not
relevant anymore, close the bug

closed 2020-05-12 by mengdonglin