thesofproject/sof #2880: [BUG][TGL][DSM] one additional cycle of single will be captured after playback ends

keqiaozhang, 2020-04-26:
**Describe the bug**
This issue happens when playback ends first. From the captured wav, there's one more cycle of single is captured. 

For example :
**To Reproduce**
1. enable DSM loopback
Run: 
[amp_enable.sh.txt](https://github.com/thesofproject/sof/files/4534963/amp_enable.sh.txt)
2. run playback(only for 3 seconds ) and capture at same time.
aplay -Dhw:0,0 -f s32_le -c 2 -r 48000  sine_48k_32bit.wav  -d 3 & arecord -Dhw:0,0 -f s32_le -c 4 -r 48000 capture.wav -d 5
3. check the wav file, form the snapshot you can see, ~16ms after playback, there's one more cycle of single is captured. 

[image]

**Reproduction Rate**
10/10

**Expected behavior**
no extra singles after playback 

**Environment**
Driver: fce7f934f + https://github.com/thesofproject/linux/pull/1971 + https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?h=for-next&id=914f674bec6efe42f9d6b036850a618fd1698290
Firmware:
SOF branch:  tgl-004-drop-stable Commit: 859110d
TPLG: sof-tgl-max98373-rt5682.tplg 

[dmesg.txt](https://github.com/thesofproject/sof/files/4534976/dmesg.txt)
[logger.txt](https://github.com/thesofproject/sof/files/4534977/logger.txt)

keqiaozhang, 2020-04-27:
Duplicated as https://github.com/thesofproject/sof/issues/2701

YvonneYang2, 2020-06-09:
Issue cannot be reproduced with test recipe as below:

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev commit:5cb79ee6f
Revert  6a7c533d4a1854f54901a065d8c672e890400d8a
Revert cd0a32371db73d0b50536a7ca4f036abddff0d1d+ https://github.com/thesofproject/linux/pull/2062 + hack commit https://github.com/thesofproject/linux/commit/6769df9d4d48cfae008a8192aa5b723bc8edb804
FW: https://github.com/thesofproject/sof/tree/tgl-006-drop-stable + #3006

[image]

lgirdwood, 2020-06-10:
Looks like 4ms of data coming 16ms after end of playback. Does this happens in SSP loopback mode ? i.e. is it a codec issue or a FW issue.

sinahuang, 2020-06-24:
Error also can be reproduced on chrome book with codec 98373 in I2S mode.

lgirdwood, 2020-06-24:
@tlauda is it possible that DMA continues to drain pipeline (or parts thereof) ?

Liviali155, 2020-07-08:
On TGL chromebook in  SDW mode with codec 98373  FW:https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc7 ,commit: 33cfa90e
The wave of capture file as below:
[image]
[sof-logger0708.log](https://github.com/thesofproject/sof/files/4888647/sof-logger0708.log)
[dmesg0708.log](https://github.com/thesofproject/sof/files/4888650/dmesg0708.log)
[capture.zip](https://github.com/thesofproject/sof/files/4888652/capture.zip)

Liviali155, 2020-07-09:
Error still can be reproduced with https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc8 ,commit: 2057409 on TGL chrome laptop I2S with codec 98357,the same  phenomenon with   FW:https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc7 ,commit: 33cfa90

ClarexZhou, 2020-07-27:
Error still seen on https://github.com/thesofproject/sof/tree/releases/tgl/v7.0-rc10

yongzhi1, 2020-10-19:
Hi, @Liviali155 , can you verify the same with https://github.com/thesofproject/sof/releases/tag/releases%2Ftgl%2Fv10.0-rc1 and cherry-pick https://github.com/thesofproject/sof/commit/20fbbe77e47421dae? In my test, the last channel is always 0. Thanks!!

lgirdwood, 2020-10-20:
@slawblauciak @bkokoszx request for TGL release v10

WYL-12, 2020-12-01:
On TGL-Delbin-I2S-02 , using https://sof-ci-storage.sh.intel.com/release/sof/releases/tgl/v11.0-rc3/c6249fff6493_chrome_release/ ,issue can be reproduced in below steps:
1.enable DSM loopback
Run:
amp_enable.sh.txt
2.run playback(only for 3 seconds ) and capture at same time.
aplay -Dhw:0,0 -f s32_le -c 2 -r 48000 sine_48k_32bit_10s_stereo_l5_r2.wav
 -d 3 & arecord -Dhw:0,0 -f s32_le -c 2 -r 48000 capture.wav -d 5
3.check the wav file, form the snapshot you can see, ~16ms after playback, there's one more cycle of single is captured.
[image]

closed 2020-12-01 by WYL-12