thesofproject/sof #2883: [BUG][TGL][DSM] old/wrong data at the beginning of released stream

bkokoszx, 2020-04-27:
**Describe the bug**
This issue happens when playback pipeline is stopped and release several times. At the beginning of released stream appears about 1 ms of old/wrong data.
[image]

**To Reproduce**
1. Enable smart amp loopback:
   run [amp_enable.sh.txt](https://github.com/thesofproject/sof/files/4538917/amp_enable.sh.txt)
2. Run capture:
   arecord -Dhw:0,0 -f s32_le -r 48000 -c 4 -vv rec.wav
3. While running capture, start playback:
   aplay -Dhw:0,0 -fS32_LE -c 2 -r 48000 -vv -i sine_48_2_32.wav
4. Stop and release playback pipeline several times by "space" key.
5. Check rec.wav wave.

**Reproduction Rate**
~1/10

**Expected behavior**
Proper data at the beginning of released stream.

**Environment**
1) Driver: fce7f934f + thesofproject/linux#1971 + https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?h=for-next&id=914f674bec6efe42f9d6b036850a618fd1698290
2) Topology file: sof-tgl-max98373-rt5682.tplg
3) Firmware: tgl-005-drop-stable Commit: 277a971

paulstelian97, 2020-04-27:
Does the wrong data fit well at the end of the previous stream?

WYL-12, 2020-12-01:
On TGL-Delbin-I2S-02 , using https://sof-ci-storage.sh.intel.com/release/sof/releases/tgl/v11.0-rc3/c6249fff6493_chrome_release/ ,issue can be reproduced in below steps:
1.Enable smart amp loopback:
run amp_enable.sh.txt
2.Run capture:
arecord -Dhw:0,0 -f s32_le -r 48000 -c 2 -vv rec.wav
3.While running capture, start playback:
aplay -Dhw:0,0 -f s32_le -c 2 -r 48000 -vv -i sine_48k_32bit_10s_stereo_l5_r2.wav
4.Stop and release playback pipeline several times by "space" key.
5.Check rec.wav wave.
[image]

plbossart, 2021-06-10:
@bkokoszx @slawblauciak @lgirdwood this also looks like an old issue without any update? isn't this fixed already?

lgirdwood, 2021-06-11:
Yes, I think it's fixed. Closing.

closed 2021-06-11 by lgirdwood