thesofproject/sof #2885: [BUG][SDW] Wrong data after release in sdw_alh_loopback_stream_operations

jdrzaszcz, 2020-04-27:
**Description**
When running simultaneous playback and capture on SdW interface on internal loopback with pause/release on playback stream after release at the beginning comes wrong data.
This problem doesn't occur on SSP interface.

Example recorded file:
[image]

**To Reproduce**
Slim Driver python test from file:
test_07_sdw_alh_loopback_stream_operations.py
e.g. TestSdwAlhLoopbackStreamOperations24000Hz16b16b8ch

**Reproduction Rate**
~30 %

**Expected behavior**
Proper data at the beginning of released stream.

**Environment**
WHL RVP
Commit: 60a5a96bbbf0889d8b5918b563d46ce1624c2492 on SOF master branch

 **Example log for topology with volume component and without this component in attachment**
[2020_04_27_14_08_39.zip](https://github.com/thesofproject/sof/files/4539520/2020_04_27_14_08_39.zip)
[2020_04_27_14_31_45.zip](https://github.com/thesofproject/sof/files/4539586/2020_04_27_14_31_45.zip)

slawblauciak, 2020-08-10:
@jdrzaszcz isn't this the same issue you've been noticing when testing the dai groups?

plbossart, 2020-08-10:
I don't think we clear the PDI fifos for a pause. I think we keep the data as is, so don't push anything out, but the capture stream continues so when the first stream resumes it's not aligned with the capture.

We do clear the PDI when starting playback, it was added in https://github.com/thesofproject/linux/pull/1887/commits/01cc461ba4f57d8dcfdf250d38adce9fb657d189

I think no matter what we do, you should expect a discontinuity on the capture path, unless the sine wave period matches the SSP period (4kHz).

jdrzaszcz, 2020-08-11:
@slawblauciak No, it's different issue.

plbossart, 2021-06-23:
@slawblauciak is this issue still relevant and isn't it a 'diagnostic driver' tag that's needed?

closed 2025-07-04 by github-actions[bot]