thesofproject/sof #2888: [BUG][JSL] Firmware load failure on new JSL platform

keqiaozhang, 2020-04-28:
**Describe the bug**
We received one new JSL platform weeks ago and codec ALC700 in HDA mode as default. I tried the latest kernel driver and firmware on it, but DSP firmware failed to boot with error code "0x0000002c". 

**To Reproduce**
check dmesg after system is booted.

**Reproduction Rate**
10/10
**Expected behavior**
sof can load successfully.

**Impact**
firmware boot failure.

**Environment**
1) kernel: sof-dev/8f1c6ad6
2) firmware: master/60a5a96b
3) JSL platform HDA mode

**dmesg**
```
[   67.010572] sof-audio-pci 0000:00:1f.3: couldn't bind with audio component
[   67.010591] sof-audio-pci 0000:00:1f.3: init of i915 and HDMI codec failed
[   67.012779] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   67.033164] sof-audio-pci 0000:00:1f.3: codec #2 probe error, ret: -2
[   67.033622] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 1
[   67.033625] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   67.034225] sof-audio-pci 0000:00:1f.3: DMICs detected in NHLT tables: 4
[   70.107132] sof-audio-pci 0000:00:1f.3: error: cl_copy_fw: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[   70.107147] sof-audio-pci 0000:00:1f.3: error: load fw failed ret: -110
[   70.107180] sof-audio-pci 0000:00:1f.3: error: status = 0x0000002c panic = 0x00000000
[   70.107188] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   70.107192] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[   70.158164] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[   70.158172] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask 3
[   70.158379] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110
```

ranj063, 2020-04-28:
@keqiaozhang seems like 2 issues here:
1. HDA codec probe failed.
2. probably an issue with FW signing?

mengdonglin, 2020-04-28:
@keqiaozhang Can you file another issue to track HDA codec probe failure?

kv2019i, 2020-04-28:
@mengdonglin wrote:
> @keqiaozhang Can you file another issue to track HDA codec probe failure? And we can focus on audio DSP boot failure in this issue.

Ack. This seems like a separate problem. This is HDMI codec init failing:
```
[   67.033164] sof-audio-pci 0000:00:1f.3: codec #2 probe error, ret: -2
[   67.033622] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 1
```

.. and with current driver, this will not block driver probe anymore. HDMI will just be disabled, but other codecs should be usable.

tlauda, 2020-04-28:
`0x2C` indicates signature verification failure. It looks like you have wrong FW signature.

xiulipan, 2020-04-29:
@tlauda You are right, the FW load issue is gone with debug key FW.
@keqiaozhang with `i915.force_probe=4e51` the HDA HDMI codec probe is also fine on the board.
Please double check and close the issue

keqiaozhang, 2020-04-29:
Thanks @xiulipan. This issue is solved after using debug key FW

closed 2020-04-29 by keqiaozhang