thesofproject/sof #2889: [DUT][JSL] random error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read

yongzhi1, 2020-04-28:
Describe the bug
On JSL-Rev2 volume rvp board with ES1 silicon. After flashing coreboot, in first few cold boot cycle(using power switch for coldboot) audio works and then it gets into this error condition.

To Reproduce
1.System boot up.
2. Check dmesg:

[   31.202419] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   31.215853] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff97c9ada0)
[   31.237736] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   31.280547] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 4
[   31.292929] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   31.300464] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   31.592949] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS red
[   31.608656] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[   31.623407] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   31.630936] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   31.923397] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS red
[   31.939625] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[   31.952005] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   31.959523] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   32.252408] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS red
[   32.266729] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[   32.275870] sof-audio-pci 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -110
[   32.286157] sof-audio-pci 0000:00:1f.3: error: status = 0xffffffff panic = 0xffffffff
[   32.295169] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[   32.302350] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110

Reproduction Rate
100% on some boards

Expected behavior
sof can be loaded and initialized normally.

Actual result
Error message shows, and sof failed to initialize.

dmesg
Attached a shorter log from https://partnerissuetracker.corp.google.com/issues/151015807
[Audio.log](https://github.com/thesofproject/linux/files/4542657/Audio.log)

mengdonglin, 2020-04-28:
@yongzhi1 Will flashing coreboot solve this issue?

yongzhi1, 2020-04-28:
Sure @mengdonglin, thanks for the pointer, will check with @sathyap-chrome.

sathyap-chrome, 2020-04-28:
@mengdonglin the issue in CML was due to flashing DCI enabled Coreboot and later switching back to another CB. But in that case, all boots failed until reflashed. 

Here the scenario is working in some boots and not working on others.
But will check once from coreboot team

yongzhi1, 2020-04-28:
For #2136 , the error happened at load phase, here .load_firmware is passed, but .run failed:

[   32.275870] sof-audio-pci 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -110

mengdonglin, 2020-05-04:
@yongzhi1 Can you please use the debug card to dump the HW status? It will help us to check with HW team where  the HW gets stuck.

yongzhi1, 2020-05-08:
Thanks @mengdonglin , we are still waiting for the validation results, for the HW status dump, do you mean some HD-A audio registers?

yongzhi1, 2020-05-26:
Hi, @mengdonglin @xiulipan , the issue was not reported recently with latest BKC, so let me close it.

closed 2020-05-26 by yongzhi1

yongzhi1, 2020-07-27:
Thanks @ranj063, the issue was not reproducible with #2950 

localhost /sys/kernel/debug/pmc_core # suspend_stress_test -c 5
Running 5 iterations with:
  suspend: 5-10 seconds
  wake: 5-10 seconds
  backup_rtc: false
  errors_fatal: true
  fw_errors_fatal: true
  bugs fatal:  true
  warnings fatal:  false
  crcs fatal:  true
  suspend command: powerd_dbus_suspend --delay=0
  wakeup source to ignore: none
  record_dmesg_dir: 
  pm_print_times:  true
tlsdated stop/waiting
Suspend     1 of 5: sleep for  8 seconds... wake for  5 seconds...
Suspend     2 of 5: sleep for  5 seconds... wake for  6 seconds...
Suspend     3 of 5: sleep for  5 seconds... wake for  9 seconds...
Suspend     4 of 5: sleep for  7 seconds... wake for  5 seconds...
Suspend     5 of 5: sleep for  9 seconds...
tlsdated start/running, process 4526

Finished 5 iterations.
Premature wakes: 0
Late wakes: 0
Suspend failures: 0
Wakealarm errors: 0
Firmware log errors: 0
s0ix errors: 0

slp_s0_residency_usec also increases as expected.