thesofproject/sof #2890: [DUT][ICL] random error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read

xiulipan, 2020-04-28:
random dsp error message will appear on ICL RVP when FW need to be loaded(boot/runtime_pm_resume/s3_resume)

<details>
 <summary>dmesg example</summary>

```
kernel: [  409.510390] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
kernel: [  409.510423] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
kernel: [  409.510428] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
kernel: [  409.510435] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
kernel: [  409.510981] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
kernel: [  409.510987] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
kernel: [  409.511140] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
kernel: [  409.811376] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x6000010 timedout
kernel: [  409.811379] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
kernel: [  409.811403] sof-audio-pci 0000:00:1f.3: unknown ROM status value 06000010
kernel: [  409.811426] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
kernel: [  409.811447] sof-audio-pci 0000:00:1f.3: unknown ROM status value 06000010
kernel: [  409.811456] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
kernel: [  409.811464] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
kernel: [  409.811469] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
kernel: [  409.811472] sof-audio-pci 0000:00:1f.3: iteration 0 of Core En/ROM load failed: -110
kernel: [  409.811493] sof-audio-pci 0000:00:1f.3: Error code=0xffffffff: FW status=0xffffffff
kernel: [  409.812051] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
kernel: [  409.812083] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
kernel: [  409.812088] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
kernel: [  409.812095] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
kernel: [  409.812887] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x80000000 successful
kernel: [  409.812909] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
kernel: [  409.813063] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
kernel: [  409.887374] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5000001 successful
kernel: [  409.897918] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x5 successful
kernel: [  409.898383] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x140000 successful
kernel: [  409.898395] sof-audio-pci 0000:00:1f.3: Firmware download successful, booting...
kernel: [  409.905888] sof-audio-pci 0000:00:1f.3: ipc rx: 0x70000000: FW_READY
kernel: [  409.905893] sof-audio-pci 0000:00:1f.3: ipc: DSP is ready 0x70000000 offset 0x81000
kernel: [  409.905901] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x70000000: FW_READY
kernel: [  409.905990] sof-audio-pci 0000:00:1f.3: firmware boot complete
```
</details>

previous discussion in https://github.com/thesofproject/sof-test/pull/16
@bardliao @ranj063

mengdonglin, 2020-04-28:
Transfer to SOF repo from linux issue #1703, to group with other DSP boot failure.

Liviali155, 2020-04-28:
On icl run  multiple-pipeline-playback also can reproduce this issue with latest sof/dev(8f1c6ad6) +master(de2dcbbc)

After error occured ,all the pipeline can work

1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
5../multiple-pipeline-playback.sh -l 50

***Reproduction rate***
Run 50times,failed at 44th

[image]

<details>
 <summary>dmesg_error.log</summary>

````
   [  777.984804] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[  777.984834] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000      
````
</details>

[dmesg0428.log](https://github.com/thesofproject/sof/files/4544319/dmesg0428.log)

[sof-logger0428.log](https://github.com/thesofproject/sof/files/4544320/sof-logger0428.log),

tlauda, 2020-04-28:
`0x6000010` indicates that ROM is still waiting for response from CSE regarding IMR allocation. What is the value of the timeout? It looks like timing issue.

Liviali155, 2020-04-29:
> On icl run multiple-pipeline-playback also can reproduce this issue with latest sof/dev(8f1c6ad6) +master([de2dcbb](https://github.com/thesofproject/sof/commit/de2dcbbc2658c67ed7ab761ea4a52a3c3a3672f8))
> 
> After error occured ,all the pipeline can work
> 
> 1.sudo reboot
> 2.cd sof-test
> 3.cd test-case
> 4.export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
> 5../multiple-pipeline-playback.sh -l 50
> 
> _**Reproduction rate**_
> Run 50times,failed at 44th
> 
> [image]
> 
> dmesg_error.log
> [dmesg0428.log](https://github.com/thesofproject/sof/files/4544319/dmesg0428.log)
> 
> [sof-logger0428.log](https://github.com/thesofproject/sof/files/4544320/sof-logger0428.log),

Add the log complete
[dmesg042.log](https://github.com/thesofproject/sof/files/4549651/dmesg042.log)
[sof-logger0429.log](https://github.com/thesofproject/sof/files/4549652/sof-logger0429.log)

Liviali155, 2020-05-22:
On latest sof-dev(2de2b2ac)+master(11ee3b87 )  run
./check-kmod-load-unload-after-playback.sh -l 1000
error not seen

Liviali155, 2020-05-26:
On 20200526 for-next(e0b2db35)+master(136b48ce) ,error occured again when run Check Runtime PM status
```
2020-05-25 22:59:15 UTC [REMOTE_INFO] aplay killed
2020-05-25 22:59:15 UTC [REMOTE_INFO] wait dsp power status to become suspended
2020-05-25 22:59:17 UTC [REMOTE_INFO] dsp suspended in 3s
2020-05-25 22:59:17 UTC [REMOTE_INFO] runtime status: suspended
2020-05-25 22:59:17 UTC [ERROR] Caught dmesg error
===========================>>
[  492.621937] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
<<===========================
2020-05-25 22:59:17 UTC [REMOTE_ERROR] Catch error in dmesg
2020-05-25 22:59:20 UTC [REMOTE_INFO] Test Result: FAIL!
```
[dmesg.txt](https://github.com/thesofproject/sof/files/4679657/dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4679658/slogger.txt)

Liviali155, 2020-07-08:
On 20200708  for-next(cce047aa)+master(adc8cc22) ,error occured again when run module reload test manually
1."sudo reboot" to reboot system
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-hda-generic-4ch.tplg
5../check-kmod-load-unload.sh  -l 50

Failed at 28/50
[image]

<details>
 <summary>dmesg_error.log</summary>

````
[  360.465179] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf010e0e successful
[  360.765342] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x6000010 timedout
[  360.765345] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[  360.765371] sof-audio-pci 0000:00:1f.3: unknown ROM status value 06000010
[  360.765378] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[  360.765400] sof-audio-pci 0000:00:1f.3: unknown ROM status value 06000010
[  360.765409] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  360.765417] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[  360.765423] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask f
[  360.765426] sof-audio-pci 0000:00:1f.3: iteration 0 of Core En/ROM load failed: -110
[  360.765447] sof-audio-pci 0000:00:1f.3: Error code=0xffffffff: FW status=0xffffffff
[  360.765815] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0f successful
[  360.765832] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f0f0e successful
[  360.765838] sof-audio-pci 0000:00:1f.3: unstall/run core: core_mask = 1
[  360.765845] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 1 : core_mask 1
````
</details>

[dmesg0708.log](https://github.com/thesofproject/sof/files/4888099/dmesg0708.log)

Liviali155, 2020-07-08:
On 20200708 sof-dev(ef5a473d)+master(adc8cc22),error not seen  when run module reload test  50 times manually

slawblauciak, 2020-08-10:
@xiulipan can you please verify if you can still see the error?

xiulipan, 2020-08-11:
@slawblauciak Yes, we can still see this issue on for-next branch. I think this maybe related to some kernel patches not up-streamed. Will double check .

mengdonglin, 2020-09-01:
Close as recent CI daily test with sof-dev kernel and FW master branch cannot reproduce this issue

closed 2020-09-01 by mengdonglin