thesofproject/sof #2891: [BUG][SSP] Additional data after release in test_12_ptp_simple_stream_operations

jdrzaszcz, 2020-04-28:
**Describe the bug**
When running simultaneous playback and capture on SSP interface on internal loopback with pause/release on playback stream after release at the beginning some additional data. The wave looks good in every interval , but when I cut the silence, the waves don't match.

Example recorded file:
[image]

**Topology**
```
                pipe_plb
                +--------------------+
+----------+    | +------+   +-----+ |   +----+
| Host plb |----->|Buffer|-->| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+    |  |
                pipe_cap                  |  |
                +--------------------+    |  |
+----------+    | +------+   +-----+ |   +----+
| Host cap |<-----|Buffer|<--| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+
```
**To Reproduce**
Ru slim driver python tests from file:
test_12_ptp_simple_stream_operations.py
e.g. TestPtpSimpleStreamOperations24000Hz16b16b4ch
Cut the silence between intervals.

**Reproduction Rate**
100%

**Expected behavior**
After cutting out the silence, wave should looks smooth.

**Environment**
APL RVP
TGL RVP
Commit: de2dcbb on SOF master branch

**Example log for in attachment**
[2020_04_28_08_19_50.zip](https://github.com/thesofproject/sof/files/4544272/2020_04_28_08_19_50.zip)

lgirdwood, 2020-05-07:
@jdrzaszcz do you mean pause and release here ? i.e. the stream is not aligned after pause release cycle ?

jdrzaszcz, 2020-05-08:
@lgirdwood Yes, that's right. For me, the additional data looks like it has been copied  from some buffer again.

lgirdwood, 2020-05-08:
Looks like 16 samples are being repeated which is incidentally the SSP FIFO size. IIRC, there are SSP register for trailing bytes and this may need to be cheked.

ktrzcinx, 2020-10-07:
It looks like this issue can't be reproduced on latest master branch (3c073b478fb9f), @jdrzaszcz can you confirm ?

lgirdwood, 2020-10-07:
@ktrzcinx good to know, could be the recent FIFO draining PR from @keyonjie

jdrzaszcz, 2020-10-07:
I checked on commit: 9e9aeadd7fe4b140bb1bdd97737c9e144570d15d and it looks like this issue has been fixed. I will perform more test and than close it.

jdrzaszcz, 2020-10-09:
@ktrzcinx @lgirdwood I performed more test with script on this commit 9e9aeadd7fe4b140bb1bdd97737c9e144570d15d and there are still additional data in recorded file but not with every pause/release action. 
[image]

This also heppen on top of the master commit: e5d17f3c10f3d170d3aa57460c124cf9cf77a1d2.
[image]

lgirdwood, 2020-10-09:
It's difficult to know if we are repeating samples (from say SSP or DMA FIFOs)  or we are skipping ahead to the next wave ? @jdrzaszcz I guess you have seen more results so can you comment ? Can you confirm we are stopping after DMA transfer has completed ? (to rule out stale DMA FIFO data).

jdrzaszcz, 2020-10-12:
@lgirdwood In this test case we are playing one wave with sending  StreamTrigPause and StreamTrigRelease ipc on playback stream.
As I can see in the recorded file, after sending StreamTrigRelease , the last few samples from the previously of the recorded wave are repeated.
After sending StreamTrigPause /StreamTrigRelease to FW, we only sends trigger and we do not interfere with DMA on the host side.

lgirdwood, 2020-10-12:
@jdrzaszcz ok are you saying you think it might be HDA DMA instead of DW DMA ?  I guess a simple test would be to add a simple DEBUG PCM checker in host.c (to say check that every sample has a value of +1 from the previous sample) and send a wav file of samples that increment. This would prove or otherwise if it was HDA or DW DMA.

jdrzaszcz, 2020-10-16:
@lgirdwood I don't know which DMA is causing the issue. Unfortunately, to prepare FW with  such a PCM, I will need some developer assistance. @ktrzcinx could You prepare FW with this DEBUG PCM checker?

jdrzaszcz, 2020-10-21:
@lgirdwood @ktrzcinx I have done some tests with FW prepared by Karol and I have bad news.
To test this, I use rising wave which means the previous value should be smaller than next value.
When I run 48 audio formats for 3 of them there were some duplicate values in FW logs (above condition wasn't True).
But the tests almost always fails.
Example:

[image]

But when this condition was met, there were still some repeated samples after trigger release.
[image]

As I understand this means, that the problem is on both sides HDA DMA and DW DMA, but much more often this problem occurs on DW DMA (Dai) side.

lgirdwood, 2020-10-21:
Thanks @jdrzaszcz very useful information. Can you confirm the probability at playback buffer and capture buffer below
```
                pipe_plb
                +--------------------+
+----------+    | +------+   +-----+ |   +----+
| Host plb |----->|Buffer|-->| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+    |  |
                pipe_cap                  |  |
                +--------------------+    |  |
+----------+    | +------+   +-----+ |   +----+
| Host cap |<-----|Buffer|<--| Dai |---->|SSPx|
+----------+    | +------+   +-----+ |   +----+
                +--------------------+
```
I'm trying to understand if the issues we see on capture are just due to downstream propagation of the issue on playback or introduced after ?
I am suspicious we are repeating data from SSP of HDA DMA FIFOs (since we are puasing on DW DMA transfer boundaries) ?

ktrzcinx, 2020-10-22:
@lgirdwood in this test implementation only playback pipeline is paused/resumed, capture pipeline is permanently at active state, so shouldn't introduce any distortions. After excluding host dma, we may suspect dai dma and ssp. SSP cleans-up FIFO after pause, so should have any trashes after restart on release. Dw dma doesn't have good alibi.

lgirdwood, 2020-10-22:
> @lgirdwood in this test implementation only playback pipeline is paused/resumed, capture pipeline is permanently at active state, so shouldn't introduce any distortions. After excluding host dma, we [...]

Good work and findings here.  Do you think this can be solved by flushing FIFO or using the trailing bytes bits in SSP ?

jdrzaszcz, 2020-10-23:
I run test with trigger pause/release on playback pipeline with probe connected buffers.  The buffer data in the playback pipeline looks, fine there is no distortion in the captured data to the wave file.
This confirms that there is some problem with DW DMA.

lgirdwood, 2020-10-29:
@jdrzaszcz good find - so it looks like we may need to invoke trailing bytes or flushing in pause or resume. @keyonjie fyi.

ktrzcinx, 2020-10-30:
I found that after resuming dw-dma without full dw-dma stop->start (rather by resetting SUSPEND bit) resolves issue with those additional data:

[image]

The problem is, that this solution (at least in my implementation) sometimes has a problem with cycling LLP start:

[image]

And logs from this situation:

```
[   4031422] (       995) c0 dai          1.1   dai_copy(), dir: 0 copy_bytes= 0x40, frames= 8, first sample value 1760936590
[   4031427] (         5) c0 dai          1.1   dai_dma_cb() dir 0 bytes 64
[   4032422] (       995) c0 dai          1.1   dai_copy(), dir: 0 copy_bytes= 0x40, frames= 8, first sample value 2147483646
[   4032427] (         5) c0 dai          1.1   dai_dma_cb() dir 0 bytes 64
[   4033363] (       935) c0 pipe         1.3   pipe trigger cmd 2
[   4033369] (         6) c0 dai          1.1   dai_comp_trigger_internal(), PAUSE 3
[   4033416] (        48) c0 dw-dma             dw_dma_pause(): dma 0 channel 0 pause, LLI 0xbe0da060
[   4033421] (         5) c0 ssp-dai      1.0   ssp_trigger() cmd 2
[   4033425] (         4) c0 ssp-dai      1.0   ssp_pause(), TX
[   5045925] (   1012500) c0 pipe         1.3   pipe trigger cmd 3
[   5045931] (         6) c0 dai          1.1   dai_comp_trigger_internal(), RELEASE 4
[   5045936] (         5) c0 dw-dma             dw_dma_release(): dma 0 channel 0 release, LLP 0xbe0da000
[   5045940] (         5) c0 dw-dma             dw_dma_release(): ctrl_lo 0x10104925 ctrl_hi 0xc0020000 cfg_lo 0x80000323 cfg_hi 0x20
[   5045945] (         5) c0 dw-dma             dw_dma_release(): LLI CH_EN 0x1 SAR 0xbe084f38 DAR 0x77010 LLP 0xbe0da020
[   5045949] (         4) c0 dw-dma             dw_dma_release(): LLI LLP 0xbe0da000
[   5045959] (        11) c0 dw-dma             dump_llp(): cfg_lo 0x80000023 cfg_hi 0x20 ctrl_lo 0x10104925 ctrl_hi 0xc0000040
[   5045964] (         5) c0 dw-dma             dump_llp(): LLI CH_EN 0x1 SAR 0xbe084f78 DAR 0x77010 LLP 0xbe0da040
[   5045969] (         5) c0 dw-dma             dump_llp() 0x0x9e0da000:   SAR 0xbe084f00 	DAR 0x77010 	LLI->LLP 0xbe0da020
[   5045973] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0000040
[   5045978] (         5) c0 dw-dma             dump_llp() 0x0x9e0da020:   SAR 0xbe084f40 	DAR 0x77010 	LLI->LLP 0xbe0da040
[   5045983] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0000040
[   5045988] (         5) c0 dw-dma             dump_llp() 0x0x9e0da040:   SAR 0xbe084f80 	DAR 0x77010 	LLI->LLP 0xbe0da060
[   5045993] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0000040
[   5045998] (         5) c0 dw-dma             dump_llp() 0x0x9e0da060:   SAR 0xbe084fc0 	DAR 0x77010 	LLI->LLP 0xbe0da000
[   5046003] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0000040
[   5046007] (         5) c0 dw-dma             dump_llp() 0x0x9e0da000:   SAR 0xbe084f00 	DAR 0x77010 	LLI->LLP 0xbe0da020
[   5046012] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0000040
[   5046017] (         5) c0 ssp-dai      1.0   ssp_trigger() cmd 3
[   5046021] (         4) c0 ssp-dai      1.0   ssp_release(), TX
[   5046423] (       402) c0 dai          1.1   dai_copy(), dir: 0 copy_bytes= 0x40, frames= 8, first sample value 601295421
[   5046429] (         5) c0 dai          1.1   dai_dma_cb() dir 0 bytes 64
[   5047423] (       994) c0 dai          1.1   dai_copy(), dir: 0 copy_bytes= 0x40, frames= 8, first sample value 987842477
[   5047428] (         5) c0 dai          1.1   dai_dma_cb() dir 0 bytes 64
[   5048423] (       995) c0 dai          1.1   dai_copy(), dir: 0 copy_bytes= 0x40, frames= 8, first sample value 1374389534
[   5048427] (         5) c0 dai          1.1   dai_dma_cb() dir 0 bytes 64
[   5049422] (       995) c0 dw-dma             ERROR dw_dma_get_data_size(): xrun detected
[   5049429] (         7) c0 dw-dma             dump_llp(): cfg_lo 0x80000223 cfg_hi 0x20 ctrl_lo 0x10104925 ctrl_hi 0xc0020040
[   5049434] (         5) c0 dw-dma             dump_llp(): LLI CH_EN 0x0 SAR 0xbe084f40 DAR 0x77010 LLP 0xbe0da040
[   5049438] (         5) c0 dw-dma             dump_llp() 0x0x9e0da000:   SAR 0xbe084f00 	DAR 0x77010 	LLI->LLP 0xbe0da020
[   5049443] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0020040
[   5049448] (         5) c0 dw-dma             dump_llp() 0x0x9e0da020:   SAR 0xbe084f40 	DAR 0x77010 	LLI->LLP 0xbe0da040
[   5049453] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0020040
[   5049458] (         5) c0 dw-dma             dump_llp() 0x0x9e0da040:   SAR 0xbe084f80 	DAR 0x77010 	LLI->LLP 0xbe0da060
[   5049462] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0020040
[   5049467] (         5) c0 dw-dma             dump_llp() 0x0x9e0da060:   SAR 0xbe084fc0 	DAR 0x77010 	LLI->LLP 0xbe0da000
[   5049472] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0020040
[   5049477] (         5) c0 dw-dma             dump_llp() 0x0x9e0da000:   SAR 0xbe084f00 	DAR 0x77010 	LLI->LLP 0xbe0da020
[   5049482] (         5) c0 dw-dma             dump_llp(): sstat 0x0 	dstat 0x0 	ctrl_lo 0x10104925 	ctrl_hi 0xc0020040
[... cut ...] .1   ERROR dai_report_xrun(): underrun due to no data available
[   5049492] (         5) c0 dai          1.1   ERROR comp_underrun(): dev->comp.id = 1, source->avail = 64, copy_bytes = 0
[   5049499] (         7) c0 pipe         1.3   pipe trigger cmd 8
[   5049504] (         5) c0 hda-dma            hda-dmac: 5 channel 0 -> stop
[   5049511] (         7) c0 dai          1.1   dai_comp_trigger_internal(), XRUN
[   5049516] (         4) c0 dw-dma             dw_dma_stop(): dma 0 channel 0 stop``

lgirdwood, 2020-10-30:
@jdrzaszcz I wonder if it's missing the SSP FIFO handshake. i.e for playback do we need to manually fill SSP FIFO with 0s so that it will assert a new FIFO watermark handshake to DMA (and restart DAM) ?

slawblauciak, 2020-11-04:
@jdrzaszcz can you please confirm that this PR is effective?

jdrzaszcz, 2020-11-04:
@slawblauciak Hi, I checked if this PR #3576 fixes this issue and on WHL and ICL platforms tests pass 10 playback iterations for each audio formats.

kfrydryx, 2023-01-31:
Not occurring any more

closed 2023-01-31 by kfrydryx