thesofproject/sof #2901: [BUG] FW / Topologies for SoundWire

perexg, 2020-04-30:
**Describe the bug**
The 1.4.2 firmware and topology files cannot be used with the latest kernel compiled from topic/sof-dev branch.

**Expected behavior**
No errors at boot.

**Impact**
Development showstopper.

**Environment**
Dell Precision 5750 - i5-10400H / RT1308 / RT711? (cannot 100% confirm, but I see related errors in dmesg)

plbossart, 2020-04-30:
@perexg there is no plan to be backwards compatible with older topologies for SoundWire, we changed the machine driver so you have to get the newer topologies.  
Can you confirm which SKU this platform is, just to check if you are using the 2 or 4-speaker version?

perexg, 2020-04-30:
Ok, topology loaded, sof-soundwire card available. I cannot get Speakers working (using the sof-soundwire ucm in your repo). Headphones seems working, The links seems to be initialized three times:

````
[    8.886417] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    8.886611] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    9.732007] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    9.732278] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    9.739192] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    9.804369] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 4
[    9.928174] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[    9.928176] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:16:0
[   11.143935] sof_sdw sof_sdw: info: override BE DAI link SDW0-Playback
[   11.143936] sof_sdw sof_sdw: info: override BE DAI link SDW0-Capture
[   11.143937] sof_sdw sof_sdw: info: override BE DAI link SDW1-Playback
[   11.143937] sof_sdw sof_sdw: info: override BE DAI link SDW3-Capture
[   11.143938] sof_sdw sof_sdw: info: override BE DAI link iDisp1
[   11.143938] sof_sdw sof_sdw: info: override BE DAI link iDisp2
[   11.143939] sof_sdw sof_sdw: info: override BE DAI link iDisp3
[   11.143941] sof_sdw sof_sdw: snd_soc_register_card failed -517
[   11.183556] sof_sdw sof_sdw: info: override BE DAI link SDW0-Playback
[   11.183558] sof_sdw sof_sdw: info: override BE DAI link SDW0-Capture
[   11.183559] sof_sdw sof_sdw: info: override BE DAI link SDW1-Playback
[   11.183559] sof_sdw sof_sdw: info: override BE DAI link SDW3-Capture
[   11.183560] sof_sdw sof_sdw: info: override BE DAI link iDisp1
[   11.183561] sof_sdw sof_sdw: info: override BE DAI link iDisp2
[   11.183561] sof_sdw sof_sdw: info: override BE DAI link iDisp3
[   11.183565] sof_sdw sof_sdw: snd_soc_register_card failed -517
[   11.183748] sof_sdw sof_sdw: info: override BE DAI link SDW0-Playback
[   11.183748] sof_sdw sof_sdw: info: override BE DAI link SDW0-Capture
[   11.183749] sof_sdw sof_sdw: info: override BE DAI link SDW1-Playback
[   11.183750] sof_sdw sof_sdw: info: override BE DAI link SDW3-Capture
[   11.183750] sof_sdw sof_sdw: info: override BE DAI link iDisp1
[   11.183751] sof_sdw sof_sdw: info: override BE DAI link iDisp2
[   11.183752] sof_sdw sof_sdw: info: override BE DAI link iDisp3
[   11.183754] sof_sdw sof_sdw: snd_soc_register_card failed -517
[   11.183779] sof_sdw sof_sdw: info: override BE DAI link SDW0-Playback
[   11.183780] sof_sdw sof_sdw: info: override BE DAI link SDW0-Capture
[   11.183780] sof_sdw sof_sdw: info: override BE DAI link SDW1-Playback
[   11.183781] sof_sdw sof_sdw: info: override BE DAI link SDW3-Capture
[   11.183782] sof_sdw sof_sdw: info: override BE DAI link iDisp1
[   11.183782] sof_sdw sof_sdw: info: override BE DAI link iDisp2
[   11.183783] sof_sdw sof_sdw: info: override BE DAI link iDisp3
[   11.184660] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:16:0 Kernel ABI 3:16:0
[   11.231463] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   11.367384] sof_sdw sof_sdw: rt711-aif1 <-> SDW0 Pin2 mapping ok
[   11.367387] sof_sdw sof_sdw: rt711-aif1 <-> SDW0 Pin3 mapping ok
[   11.367402] sof_sdw sof_sdw: multicodec <-> multicpu mapping ok
[   11.367404] sof_sdw sof_sdw: rt715-aif2 <-> SDW3 Pin2 mapping ok
[   11.367406] sof_sdw sof_sdw: intel-hdmi-hifi1 <-> iDisp1 Pin mapping ok
[   11.367409] sof_sdw sof_sdw: intel-hdmi-hifi2 <-> iDisp2 Pin mapping ok
[   11.367410] sof_sdw sof_sdw: intel-hdmi-hifi3 <-> iDisp3 Pin mapping ok
[   11.367420] sof_sdw sof_sdw: snd-soc-dummy-dai <-> Headphone 0 mapping ok
[   11.367425] sof_sdw sof_sdw: snd-soc-dummy-dai <-> Headset mic 1 mapping ok
[   11.367610] sof_sdw sof_sdw: snd-soc-dummy-dai <-> SDW1-speakers 2 mapping ok
[   11.367615] sof_sdw sof_sdw: snd-soc-dummy-dai <-> Microphones 4 mapping ok
[   11.367619] sof_sdw sof_sdw: snd-soc-dummy-dai <-> HDMI1 5 mapping ok
[   11.367623] sof_sdw sof_sdw: snd-soc-dummy-dai <-> HDMI2 6 mapping ok
[   11.367626] sof_sdw sof_sdw: snd-soc-dummy-dai <-> HDMI3 7 mapping ok
[   11.394536] input: sof-soundwire Headset Jack as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input27
[   11.394763] input: sof-soundwire HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input28
[   11.394799] input: sof-soundwire HDMI/DP,pcm=6 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input29
[   11.394834] input: sof-soundwire HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sof_sdw/sound/card0/input30
````

Found a little component string issue (extra ',' after cfg-spk:4):

````
# amixer -c 0 info
Card hw:0 'sofsoundwire'/'Intel Soundwire SOF'
  Mixer name	: 'Intel Kabylake HDMI'
  Components	: 'HDA:8086280b,80860101,00100000 cfg-spk:4, cfg-amp:2 hs:rt711 spk:rt1308 mic:rt715'
  Controls      : 73
  Simple ctrls  : 43
````

plbossart, 2020-04-30:
> SKU Number: 0990

ok, so that's the 4 speaker version, which means there are 2 RT1308 amps on links 1 and 2.

plbossart, 2020-04-30:
> Ok, topology loaded, sof-soundwire card available. I cannot get Speakers working (using the sof-soundwire ucm in your repo). 

The 4 speaker conf is still WIP with @libinyang. I don't have such a device so I am not sure if the https://github.com/thesofproject/alsa-ucm-conf/pull/34 is correct. 

> Headphones seems working, The links seems to be initialized three times:

That's not normal. Maybe linked to HDMI, I don't see a reason why we would do this. All SDW DAIs are available on startup.

> Found a little component string issue (extra ',' after cfg-spk:4):

Nice catch. @RanderWang do you remember where this was set?

>   Components	: 'HDA:8086280b,80860101,00100000 cfg-spk:4, cfg-amp:2 hs:rt711 spk:rt1308 mic:rt715'
``

libinyang, 2020-05-01:
> 
> The 4 speaker conf is still WIP with @libinyang. I don't have such a device so I am not sure if the [thesofproject/alsa-ucm-conf#34](https://github.com/thesofproject/alsa-ucm-conf/pull/34) is correct.
> 

There are 2 tiny issue in https://github.com/thesofproject/alsa-ucm-conf/pull/34. 
One is we should use "PGA 3.1 3" instead of "PGA3.0 3" in RT1308-2.conf (4 speaker tplg are using 'PGA3.0 3 Master Playback Volume'). I think may be we can refine the tplg to fix it? 
The other issue is: in HiFi.conf, we should include RT1308.conf instead of RT1308-1.conf. 

BTW: I have a the patches in my own repo: https://github.com/libinyang/alsa-ucm-conf/commits/sdw_ucm2_singleprofile/ucm2. This works.

perexg, 2020-05-01:
Regarding topologies: It seems ugly to me to expose all (non-functional) controls. I know, it's nice for debugging / looking for something, but it's really confusing for the end users.

Also, if the functionality is TIED to some path (like headphones or so), the appropriate controls should be renamed (do not use PGA 3.1 3, use headphones / speaker names or so..).

plbossart, 2020-05-01:
@perexg what makes you think those controls are not functional?
Or did you mean that they are not intended for end-users? That bridge was crossed many moons ago when we started seeing codecs and DSPs with hundreds of controls, with the understanding that some higher intelligence in userspace (UCM, HAL, etc) would configure them.

That said for the topologies I agree the PGA x.y.z names are horrible. I've been trying to request a human-readable alias but no one supported this move, and I don't know enough of the topology parsing to do this on my own while doing my regular day job...

perexg, 2020-05-01:
I mean controls without the usage / connection. I understand that the user space controls are created using the topology files and those topology files describe the real hardware paths (you renamed this PGA controls to Dmic for the digital microphone input on DSP). I assume that the topology knows about the purpose of the path (thus the associated controls), doesn't? Perhaps, I don't see something in the whole picture here.

plbossart, 2020-05-01:
The controls added by topology are only for the DSP graph, and it's not always possible to know what the usage will be. see for example some topologies, there is no information at the topology level that tells us what each pipeline is used for.

[image]

perexg, 2020-05-01:
Ok, I see. The components string contains only detected chips on the bus? So there is no information related to ALH0x102.OUT and the real hardware connected to this output in the driver / topology at all?

ranj063, 2020-05-01:
> The controls added by topology are only for the DSP graph, and it's not always possible to know what the usage will be. see for example some topologies, there is no information at the topology level that tells us what each pipeline is used for.

@plbossart but the kcontrols are added in topology. The graph above doesnt show the controls assicated with the widgets but it would be fairly easy to add them to the graph.

plbossart, 2020-05-01:
> Ok, I see. The components string contains only detected chips on the bus? So there is no information related to ALH0x102.OUT and the real hardware connected to this output in the driver / topology at all?

For DAI, the topology specifies the dai_id (e.g. ALHx102 means link 1 PDI 2) and the dailink_id, but the information for the dailink is part of the machine driver. It's one of the issues with the topology, it doesn't describe the connections outside of the DSP, which is both good and bad.
Good because e.g. with the same Baytrail topology using SSP2 and dailink0 we can deal with different platforms, with the machine driver configuring the actual platform routing.
Bad because the information is in two places and it's not self-explanatory.
We are really missing a 'platform-description'

juimonen, 2020-05-04:
> That said for the topologies I agree the PGA x.y.z names are horrible. I've been trying to request a 
> human-readable alias but no one supported this move, and I don't know enough of the topology 
> parsing to do this on my own while doing my regular day job...

@plbossart @perexg In HDA topologies we use mainly the HDA created mixer elements and related controls -> that's why we get "proper" names to most volumes (Dmic being an exception). In HDA topologies you can very well adjust the volume from "PGA_dadadidaa" and it has an effect, but we prefer using the codec volume. And as we have the codec volume, it doesn't make too much sense for the user to use the adjust the volume in the middle of the pipeline in the DSP. The volume component in these cases works mainly as format converter and the volume setting is mainly for pipeline internal levels (so maybe some default UCM setting is set, not for user).

OTH Dmic volume is crucial, because there is no "codec volume" to change. The DSP volume is the only one we can have... And I guess this can be the case in some SDW topologies.

So we have 2 kinds of volumes:
1) pipeline internal "balancing/conversion" volumes not meant too much for user (but for which we might still need the controls)
2) pipeline volumes, which really are the only/best way to change the volume

Concerning the naming, what would we call for example the pga in HDA joint speaker/headset path? HeadsetSpeakerPipelineIntermediateVolume? Would that be clear for the user?

perexg, 2020-05-05:
> In HDA topologies we use mainly the HDA created mixer elements and related controls -> that's why we get "proper" names to most volumes

Yes, I know this. But the HDA driver does the mapping job (to pretty names / functional names) in this case. I understand that this mapping is completely missing in ASoC (replaced by the hardware related descriptors). The hardware related descriptors are really confusing for the end users (including the non-unified names - strings) and all GUI mixer applications are ruled out to be understandable. We need to find a way to handle this.

My point is that if the driver knows something about the route, the control name should be set appropriately.

plbossart, 2020-06-22:
@perexg I suggest we close this bug for now, the discussion completely drifted from the initial issue and re-reading this I am not sure what the ask is?

perexg, 2020-06-22:
The original question was about the signed firmware files and compatible topologies for SoundWire hardware. If the SOF binary 1.5 release is fine, then you can close this bug.

The CONFIG_SOUNDWIRE_INTEL is not selectable from the sound/ tree in 5.8-rc2 (still) nor Mark's for-5.9 ASoC branch, so my hands are limited to add the SoundWire support to our distributions.

plbossart, 2020-06-22:
@perexg there's only 10-odd patches or so before we can add the build, should be soonish now that all infrastructure patches are behind us.

closed 2020-06-22 by plbossart