thesofproject/sof #2908: [BUG]  Interrupts from PMC are not masked on cavs platfroms

lbetlej, 2020-05-05:
**Describe the bug**
Bit #5 in REG_IRQ_IL2MD(0): 0xb00050 is not set. Expectation is that the bit will be set for all DSP cores. 
Since in a default mask bit #5 is set  (#define REG_IRQ_IL2MD_ALL	0x03F181F0) there is a need to investigate how the bit is cleared.

slawblauciak, 2020-08-10:
@abonislawski any update?

abonislawski, 2020-09-21:
@slawblauciak looks like there is no request to unmask this bit and it is cleared from the beginning.
REG_IRQ_IL2MD_ALL for some reason cannot set it to 1

lgirdwood, 2020-09-21:
@abonislawski @slawblauciak what are the potential side effects here ? is this a blocker for v1.6 ?

plbossart, 2021-06-10:
@lbetlej @slawblauciak @lgirdwood this is a year-old request with the last update back in September, can we close?

lgirdwood, 2021-06-11:
@andyross @nashif fyi
@mwasko @slawblauciak whats the impact here, and can we close if fixed.

closed 2025-07-04 by github-actions[bot]