thesofproject/sof #2911: [BUG] no sound kernel 5.6.10. sof-cnl-ri

syrofoam, 2020-05-06:
**Describe the bug**
Hardware: HP EliteBook 830 G6. 
Comes only up with dummy output in Pulsemixer and a fake soundcard in Alsamixer.

**To Reproduce**
Boot kernel 
5.6.10-zen & 5.6.10 normal same problem in both kernels.

**Reproduction Rate**
100%.

**Expected behavior**
Working sound when i boot my laptop.

**Impact**
I like music & sound.
none, I can boot LTS kernel 5.4.38-LTS and everything works as normal.

**Environment**
Same problem with Suse Thumbleweed & Archlinux.

**Screenshots or console output**
[image]
\`
[    3.787978] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.788518] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.788533] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.788859] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.790768] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.795052] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.854180] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.854185] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.854970] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-cnl.ri failed with error -2
[    3.854975] sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-cnl.ri failed err: -2
[    3.854977] sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -2
[    3.855571] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -2
\`

plbossart, 2020-05-06:
@syrofoam the 'official' firmware release can be found at https://github.com/thesofproject/sof-bin/tree/stable-v1.4.2
copy this and use the go.sh script to install all the simlinks.

alternatively create a symlink from sof-cnl.ri to sof-cfl.ri

syrofoam, 2020-05-06:
Downloaded the firmware release:
New dmesg output:

/'    3.789915] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.791502] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.791555] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.792396] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.795574] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.800079] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.868359] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.868367] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.877737] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    3.877740] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    4.177948] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    4.177962] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    4.180685] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    4.180688] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    4.480961] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    4.480968] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    4.481472] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    4.481476] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    4.781441] sof-audio-pci 0000:00:1f.3: error: cl_dsp_init: timeout HDA_DSP_SRAM_REG_ROM_STATUS read
[    4.781454] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
[    4.781473] sof-audio-pci 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -110
[    4.781506] sof-audio-pci 0000:00:1f.3: error: status = 0xffffffff panic = 0xffffffff
[    4.781513] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
[    4.781515] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware -110
[    4.832482] sof-audio-pci 0000:00:1f.3: error: hda_dsp_core_reset_enter: timeout on HDA_DSP_REG_ADSPCS read
[    4.832487] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask f
[    4.832697] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -110'/

The firmware files inside intel/sof
sof-apl.ri -> v1.4.2/intel-signed/sof-apl-v1.4.2.ri
sof-bdw.ri -> v1.4.2/sof-bdw-v1.4.2.ri
of-byt.ri -> v1.4.2/sof-byt-v1.4.2.ri
sof-cfl.ri -> v1.4.2/intel-signed/sof-cnl-v1.4.2.ri
sof-cht.ri -> v1.4.2/sof-cht-v1.4.2.ri
sof-cml.ri -> v1.4.2/intel-signed/sof-cnl-v1.4.2.ri
sof-cnl.ri -> v1.4.2/intel-signed/sof-cnl-v1.4.2.ri
sof-glk.ri -> v1.4.2/intel-signed/sof-apl-v1.4.2.ri
sof-icl.ri -> v1.4.2/intel-signed/sof-icl-v1.4.2.ri

plbossart, 2020-05-06:
@syrofoam this seems to be a nasty issue with firmware authentication.
In the past we had a different binary for CFL but I can't find it.
@lgirdwood @jajanusz can you me find the 1.3.1 binary for CFL, i'd like to test a theory that this is a problem with the PV bits.

lgirdwood, 2020-05-07:
@mengdonglin do you have the 1.3.1 binary for CFL ?

olividir, 2020-05-11:
I think I have similar issue to this. 
```
[olividir@HP ~]$ sudo alsa-info.sh --no-upload
[sudo] password for olividir: 
alsactl: save_state:1595: No soundcards found...
cat: /tmp/alsa-info.PZPATTaiL0/alsactl.tmp: No such file or directory

Your ALSA information is in /tmp/alsa-info.txt.ve15WCG8qT

[olividir@HP ~]$ dmesg | grep audio
dmesg: read kernel buffer failed: Operation not permitted

```
I did the dmesg with sudo,,, which didn't really do anything. 
Some other stuff I decided to try:
```
[olividir@HP ~]$ lspci | grep Audio
00:1f.3 Audio device: Intel Corporation Cannon Lake PCH cAVS (rev 10)
[olividir@HP ~]$ cat /proc/asound/devices
 33:        : timer
[olividir@HP ~]$ lspci -vv | grep HD
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 630 (Mobile) (prog-if 00 [VGA controller])
	DeviceName: Intel(R) UHD Graphics 630
	Subsystem: Hewlett-Packard Company UHD Graphics 630 (Mobile)
[olividir@HP ~]$ sudo journalctl -b0 | grep snd_hda
[sudo] password for olividir: 
maí 11 18:00:50 HP kernel: snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040380
maí 11 18:00:50 HP kernel: snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[olividir@HP ~]$ aplay -lL
null
    Discard all samples (playback) or generate zero samples (capture)
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
speex
    Plugin using Speex DSP (resample, agc, denoise, echo, dereverb)
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Default ALSA Output (currently PulseAudio Sound Server)
aplay: device_list:274: no soundcards found...

```
I just got kernel 5.6.11 and those are reads from that version.

mengdonglin, 2020-05-12:
> @mengdonglin do you have the 1.3.1 binary for CFL ?

@lgirdwood @plbossart 1.3.1 is not for public release. I can share with you offline

mengdonglin, 2020-05-12:
@olividir Can you please share the output of lspci -v?
This error was previously reported on another HP laptop. We compared its lspci output with another working device,  it seems its BIOS doesn't initialize the audio PCI device properly and so the PCI memory is disabled.  Please find more info in https://github.com/thesofproject/sof/issues/2842#issuecomment-626136239

@plbossart @lgirdwood, @jajanusz doesn't think PV bits can cause status = 0x00000000. If this is an PV bits issue, the status should be 0x0000002c

olividir, 2020-05-25:
@mengdonglin
Updating the BIOS on my machine sorted the sound out. I am running Kernel 5.6.xx with sound now. Just wanted to let you know.

lgirdwood, 2020-05-25:
@olividir ok thanks, we can close now.

closed 2020-05-25 by lgirdwood

syrofoam, 2020-05-25:
I did blacklist snd_hda_intel, blacklist snd_soc_skl. Rebooted. Disabled microphone in bios. Rebooted again, now it works here too.