thesofproject/sof #2913: TGL: S0IX entry is failing while WOV pipeline is running with DSP in D0IX

jairaj-arava, 2020-05-07:
**Describe the bug**
On Tiger Lake with WOV running and the DSP in D0IX, we observed S0IX entry is getting failed.

**To Reproduce**
1. Flash s0ix enabled coreboot and EC on the Tiger Lake board with default DB (Max98357a).
2. Boot the device.
3. Flash the latest Google kernel with  TGL S0IX enable kernel patch.
4. Make sure WOV KWD is working during S0 with the below command 
**arecord -Dhw:0,100 -M -N -c 2 -f S16_LE -r 16000 --buffer-size=68000 tmp.wav -d 10 -vvv &**
5. Make sure S0IX is working on your system without running WOV using **suspend_stress_test -c 1**
6. Now run the command in step#4 and keep the system to suspend using **powerd_dbus_suspend**.

**Reproduction Rate**
100%

**Expected behavior**
System should enter into S0IX with WOV running and should wake up upon detecting the keyworrd(tap/loud noise near the DMIC).

**Impact**
With WOV running, system is not entering S0IX and waking up by itself upon issuing the suspend command.

**Environment**
TGL with V0.4/V0.5 release.

jairaj-arava, 2020-05-07:
ranj063 , suggested some firmware changes as below, but still we see the issue.
https://github.com/thesofproject/sof/pull/2189/commits/dfb7dca19bc8abf487628e5add16c60c6fb7cf1e

ranj063, 2020-05-07:
@mmaka1 Do you think this failure could be the result of not enabling LPRO or LPS in TGL? I enabled LPRO in the config but that didnt fix the problem. Enabling LPS resulted in an FW build failure

jairaj-arava, 2020-05-08:
Hi,

We had a debug session with our S0IX team using ITP & below are our findings:
1.  **HDA PLL** is **shutdown** during normal (without wov) S0IX and S0IX functionality is working good.
2. On the other hand with WOV + S0IX, we see **HDA PLL** is **Active** and S0IX is failing (residency counter is not increasing).

Attached the kernel logs and sof-logger.
[soix_wov_not_workingLatest.txt](https://github.com/thesofproject/sof/files/4597043/soix_wov_not_workingLatest.txt)
[before_suspend_soflogger.txt](https://github.com/thesofproject/sof/files/4597045/before_suspend_soflogger.txt)
[after_suspend_soflogger.txt](https://github.com/thesofproject/sof/files/4597046/after_suspend_soflogger.txt)

keyonjie, 2020-05-08:
> @mmaka1 Do you think this failure could be the result of not enabling LPRO or LPS in TGL? I enabled LPRO in the config but that didnt fix the problem. Enabling LPS resulted in an FW build failure

LPS on TGL is not implemented yet, it doesn't compile with it.

keyonjie, 2020-05-08:
From the driver point of view, maybe worth to double confirm if EM2.L1SEN is enabled in this scenario, on CML days, this block us to enter S0I3 for quite long time.

The bit should be 1 if the related kernel commits are all there, I only want it to be confirmed.

mmaka1, 2020-05-08:
@ranj063 LPS is not enabled on TGL, but before we start investigating the PM settings, need to find out what is going on in this failing scenario. There are some unusual activities logged, not just incorrect pwr setup.

wwittbrx, 2020-05-08:
@jairaj-arava 

Could you provide commit hashes for FW, topology, logger, kernel and the patch? Coreboot version would be helpful too. We could try and reproduce the issue on our device.

mmaka1, 2020-05-08:
@jairaj-arava Pls also note that the .ldc file must come from the same build as the fw .ri file.

jairaj-arava, 2020-05-08:
Hi,

Today we had a debug session and below are some additional findings:

1. We ruled out the suspect that the traces  **src/audio/host.c:784   host_copy(): no bytes to copy** in the firmware logs is not causing any issue with S0IX.
2. We enabled LPR0 and seen the same behavior.

Below is the kernel code base we used for our experiments:
clone: **https://chromium.googlesource.com/chromiumos/third_party/kernel**
Checkout to branch: **chromeos-5.4** 
One S0IX enable patch in TGL (shared in the mail thread).

Attached the logs from the session:
[suspend_logs_0508.txt](https://github.com/thesofproject/sof/files/4601309/suspend_logs_0508.txt)
[WOV_SOIX_FAIL_Firmware_Logs.txt](https://github.com/thesofproject/sof/files/4601310/WOV_SOIX_FAIL_Firmware_Logs.txt)

keyonjie, 2020-05-09:
@jairaj-arava, from the kernel log, it looks the system has not entered s2idle at all, you didn't do any waking up here, right?
```
[  204.918366] xhci_hcd 0000:00:14.0: pci_pm_suspend_noirq+0x0/0x2e1 returned 0 after 11981 usecs
[  204.918508] PM: noirq suspend of devices complete after 35.464 msecs
[  204.921334] PM: suspend-to-idle
[  204.921338] PM: resume from suspend-to-idle //Keyon: it resume right after the s2idle done.
[  204.923649] pci 0000:00:00.0: calling pci_pm_resume_noirq+0x0/0x121 @ 2908, parent: pci0000:00
```

jairaj-arava, 2020-05-09:
@keyonjie 
Yes with S0IX+ WOV, we didn't wake up the system and it was waking by itself (though the logs shows suspend of devices complete after 35.464 msecs  / resume of devices complete after 427.128 msecs).

To rule out the same, earlier we forcefully made the DSP to D0I3 and with out running any WOV path, we entered into normal S0IX cycling (suspend_stress_test -c 1). We saw the system is not entered into S0IX (the residency counter is not incrementing).

@ranj063 , please correct me if I missed any.

keyonjie, 2020-05-12:
> @keyonjie
> Yes with S0IX+ WOV, we didn't wake up the system and it was waking by itself (though the logs shows suspend of devices complete after 35.464 msecs / resume of devices complete after [...]

OK, this means it is not the WoV itself making the problem, have you ever checked the L1SEN bit when entering S0ix cycle? 

> 
> @ranj063 , please correct me if I missed any.

jairaj-arava, 2020-05-13:
Hi,

We had a ITP debug session on the Volteer Proto 1 platform and below are our findings:

**ITP Results:**

  **only S0IX:**
                                             We see both HDA PLL and ACRO (Audio Calibrated Ring Oscillator) are off and the S0IX functionality is working good.
[image]

 **With WOV + S0IX :**
                                           We see in fail case both HDA PLL & ACRO are active and also the reason for S0IX blocking is ACRO as per ITP is as shown below:
[image]
          
Also we had a debug session with @keyonjie and from kernel we see the behavior of below (L1SEN & D0I3) registers which is desired during WOV + S0IX:

localhost /sys/kernel/debug/pmc_core # dmesg | grep jai
[  117.640310] **jai before setting D0I3 val=0x4
[  117.640312] **jai before  settingi snd_sof_dsp_read L1SEN val=0x4804000
[  117.640318] **jai after setting D0I3 val=0x4
[  117.640320] **jai after settingi snd_sof_dsp_read L1SEN val=0x4806000.

jairaj-arava, 2020-05-27:
@mrajwa  has provided us a firmware image with LPRO selected as default clock source. With this we had a ITP debug session and we see ACRO is off or shutdown during S0IX + WOV but still S0IX is failed with WOV.

But we found some differences as below from today’s debug session. We see main_divider, other clocks in the clock tree and all the power domains (HAD_PGD*) are ON during S0IX+WOV. These all could be the reason for S0IX failure as per S0IX team.

Is there any way that we can turn this clocks off during S0IX+WOV?

[image]
[image]
[image]

abonislawski, 2020-06-17:
@sathya-nujella it is enabled in tigerlake_chrome override defconfig

Liviali155, 2020-07-17:
Error not seen  with https://github.com/thesofproject/sof/commits/releases/tgl/v7.0-rc10 ,commit: fb801c8e5e3090406955d82e06c20fcdbfd302b4   on 
  Chrome OS
TGL chromebook in  I2S mode  with codec 98357a，closed it

closed 2020-07-17 by Liviali155