thesofproject/sof #2921: [BUG][APL-nocodec]"error: ipc timed out for 0x80010000 size 216" when do check-kmod-load-unload-after-playback.sh test

Liviali155, 2020-05-09:
**Describe the bug**
"error: ipc timed out for 0x80010000 size 216" when do check-kmod-load-unload-after-playback.sh test
After error occured,all the pipeline can not  work,reboot can recovery

**To Reproduce**
1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-nocodec.tplg
5../check-kmod-load-unload-after-playback.sh -l 50

**Reproduction Rate**
1. failed at 15/50
2.pass 50/50
3.failed at 33/50

**Expected behavior**
There is o error occured when do check-kmod-load-unload-after-playback.sh test

**Impact**

1."Unable to install hw params" when aplay(0,0),dmesg show "error: ipc timed out for 0x80010000 size 216"
[image]

```
ubuntu@sh-apl-up2-nocodec-04:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofnocodec [sof-nocodec], device 0: Port0 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 1: Port1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 3: Port3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 5: Port5 (*) []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: sofnocodec [sof-nocodec], device 7: Media Playback 7 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

<details>
 <summary>dmesg_error.log</summary>

````
[ 2957.018542] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 2957.018717] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[ 2957.018729] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[ 2957.525432] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x80010000 size 216
[ 2957.525457] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
[ 2957.525465] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
[ 2957.525509] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
[ 2957.525518] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
[ 2957.525521] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[ 2957.525527] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[ 2957.525531] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
[ 2957.525542] sof-audio-pci 0000:00:0e.0: error: failed to set dai config for NoCodec-5
[ 2957.525546] sof-audio-pci 0000:00:0e.0: error: failed to restore pipeline after resume -110
[ 2957.525584] sof-audio-pci 0000:00:0e.0: pcm: open stream 0 dir 0
[ 2957.525588] sof-audio-pci 0000:00:0e.0: period min 192 max 16384 bytes
[ 2957.525591] sof-audio-pci 0000:00:0e.0: period count 2 max 16
[ 2957.525594] sof-audio-pci 0000:00:0e.0: buffer max 65536 bytes
[ 2957.525696] sof-audio-pci 0000:00:0e.0: error: trace IO error
[ 2957.525975] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[ 2957.525979] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[ 2957.525985] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[ 2957.525988] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[ 2957.525992] sof-audio-pci 0000:00:0e.0: rate_min: 48000 rate_max: 48000
[ 2957.525995] sof-audio-pci 0000:00:0e.0: channels_min: 2 channels_max: 2
[ 2957.526003] sof-audio-pci 0000:00:0e.0: pcm: hw params stream 0 dir 0
[ 2957.526009] sof-audio-pci 0000:00:0e.0: generating page table for 00000000713a9ddc size 0xffc0 pages 16
[ 2957.526018] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[ 2957.526035] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
````
</details>

No sof error trace

**Environment**
Platform: APL UP2 with nocodec
Firmware: 782066e0 https://github.com/thesofproject/sof/commits/master
Kernel: d43dc16c https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology file: sof-apl-nocodec.tplg same as firmware

[dmesg0509.log](https://github.com/thesofproject/sof/files/4602971/dmesg0509.log)
[sof-logger0409.log](https://github.com/thesofproject/sof/files/4602972/sof-logger0409.log)

Liviali155, 2020-05-09:
This issue occured from 20200508
Chnaged FW to 20200507( # 6ce635aa),error not seen ,but has issue #2693

Liviali155, 2020-05-11:
I retest it ,seems this issue occured from commit (2c9bb63)@Tomasz Lauda,please help to check it

tlauda, 2020-05-20:
@Liviali155 That patch shouldn't have any functional impact. Could you verify again to make sure it's this specific patch that caused it?

tlauda, 2020-05-26:
@Liviali155 Ping.

Liviali155, 2020-05-26:
> @Liviali155 Ping.

@YvonneYang2  is verifying it,Hi Yvonne,do you have result now

YvonneYang2, 2020-05-26:
> > @Liviali155 Ping.
> 
> @YvonneYang2 is verifying it,Hi Yvonne,do you have result now

The bisect result shows the commit ba88351 caused the issue. 
Tried ba88351 can reproduced the issue. 
But tried it's last commit 6ce635a , it will always reproduce issue https://github.com/thesofproject/sof/issues/2693 while not #2921.

YvonneYang2, 2020-05-27:
@dbaluta Could you please have a look to see if the commit could cause the issue? thanks~

Liviali155, 2020-05-28:
On glk also occured this issue with latest sof-dev(0d22f66b)+master(46897802) when run Check kmod load/unload in CI stress test
1.sudo reboot
2.cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-glk-da7219.tplg
5../check-kmod-load-unload.sh -l 50(https://github.com/thesofproject/sof-test/blob/master/test-case/check-kmod-load-unload.sh)

<details>
 <summary>dmesg_error.log</summary>

````
[ 6114.483058] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x30030000: GLB_TPLG_MSG: COMP_CONNECT
kernel: [ 6114.483065] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [ 6114.983234] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x80010000 size 216
kernel: [ 6114.983238] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
kernel: [ 6114.983240] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
kernel: [ 6114.983282] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
kernel: [ 6114.983286] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
kernel: [ 6114.983288] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
kernel: [ 6114.983291] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
kernel: [ 6114.983292] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
kernel: [ 6114.983295] sof-audio-pci 0000:00:0e.0: error: failed to set dai config for SSP1-Codec
kernel: [ 6114.983297] sof-audio-pci 0000:00:0e.0: error: failed to restore pipeline after resume -110
kernel: [ 6114.983309] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
kernel: [ 6115.487233] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x40010000 size 76
kernel: [ 6115.487236] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
kernel: [ 6115.487238] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
kernel: [ 6115.487279] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
kernel: [ 6115.487284] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
kernel: [ 6115.487285] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
kernel: [ 6115.487288] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
kernel: [ 6115.487289] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
kernel: [ 6115.487291] sof-audio-pci 0000:00:0e.0: error: -110 failed to prepare DSP for device removal
kernel: [ 6115.487329] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
kernel: [ 6115.541224] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
kernel: [ 6115.541236] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
kernel: [ 6115.541238] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
kernel: [ 6115.541247] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
kernel: [ 6115.541251] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
kernel: [ 6115.541254] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 1
kernel: [ 6115.541269] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
kernel: [ 6115.541273] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x303 successful
````
</details>

[dmesg.txt](https://github.com/thesofproject/sof/files/4693228/dmesg.txt)

tlauda, 2020-05-28:
@Liviali155 Could you check with this patch https://github.com/tlauda/sof/commit/e3f9299b4fa4af5ed0d003b18f4fbd032e851a36 and let me know whether it helps.

Liviali155, 2020-05-29:
> @Liviali155 Could you check with this patch [tlauda@e3f9299](https://github.com/tlauda/sof/commit/e3f9299b4fa4af5ed0d003b18f4fbd032e851a36) and let me know whether it helps.

Tried with  [tlauda@e3f9299](https://github.com/tlauda/sof/commit/e3f9299b4fa4af5ed0d003b18f4fbd032e851a36)
1st:./check-kmod-load-unload-after-playback.sh -l 1000,error not seen
2nd:./check-kmod-load-unload-after-playback.sh -l 1000,error not seen

closed 2020-06-01 by tlauda

Liviali155, 2020-06-01:
Tried with PR #2999,run "./check-kmod-load-unload-after-playback.sh -l 1000",error not seen

PR #2999 (2020-06-01): task: increase default stack size for EDF tasks
Increases default stack size for EDF tasks. Some of the
IPC operations are very stack heavy, so let's make sure
they won't exceed stack.

Fixes #2921 & #2693.