thesofproject/sof #2929: [BUG] Firmware panic in FIR EQ fir_cmd_set_data() assert

singalsu, 2020-05-12:
**Describe the bug**
The memcpy_s() in the function has incorrect destination size and halts the firmware when a suitable size blob (series of multiple) is received. I found this bug with beamformer FIR and fixed it already there. I'll make a PR shortly for EQ.

**To Reproduce**
Send a blob with 4x filters of length 64 (attached - I modified the mid boost FIR example this way) with sof-ctl to FIR EQ and then start playback.

**Reproduction Rate**
100%

**Expected behavior**
Start playback with EQ blob applied.

**Impact**
Showstopper, though FIR does not currently have critical usage. This also only can happen with long enough blobs to be split into parts.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

[eq_fir_mid_long.txt](https://github.com/thesofproject/sof/files/4616496/eq_fir_mid_long.txt)

singalsu, 2020-05-12:
I just created PR #2930.

slawblauciak, 2020-06-01:
PR merged. Closing.

closed 2020-06-01 by slawblauciak