thesofproject/sof #2931: [BUG] Audio Card Did Not Detected on Swift 3 54G ALC256

LazyGeniusMan, 2020-05-13:
**Describe the bug**
SOF audio card  did'nt detected on Swift 3 SF314-54G (ALC256), instead it's only detect Intel HDA PCH. Even when i add `blacklist -snd-hda-intel` in alsa-base.conf, input and output audio is not working (there is no audio card detected).

**To Reproduce**
1. Install Manjaro 20.0.1
2. Install `sof-firmware` using pacman
3. Add `alsa-base.conf` on /etc/modprobe.d/
4. Reboot

**Expected behavior**
Audio Card is detected and Internal Mic is working

**Impact**
Can't use internal mic. if snd-hda-intel blacklisted input and output audio did not work.

**Environment**
Kernel: 5.6.11-1
SOF: 1.4.2-1
Manjaro 20.0.1

**Screenshots or console output**
`dmesg` : https://pastebin.com/raw/9DBB1smC
`ls -l /lib/firmware/intel/sof*` : 
```
/lib/firmware/intel/sof:
total 1992
drwxr-xr-x 2 root root   4096 May 10 10:37 debug
-rw-r--r-- 1 root root 282624 Jan 23 06:40 sof-apl.ri
-rw-r--r-- 1 root root 109616 Jan 23 06:40 sof-bdw.ri
-rw-r--r-- 1 root root  88164 Jan 23 06:40 sof-byt.ri
lrwxrwxrwx 1 root root     10 Jan 23 06:40 sof-cfl.ri -> sof-cnl.ri
-rw-r--r-- 1 root root  88972 Jan 23 06:40 sof-cht.ri
lrwxrwxrwx 1 root root     10 Jan 23 06:40 sof-cml.ri -> sof-cnl.ri
-rw-r--r-- 1 root root 286720 Jan 23 06:40 sof-cnl.ri
-rw-r--r-- 1 root root 109616 Jan 23 06:40 sof-hsw.ri
-rw-r--r-- 1 root root 286720 Jan 23 06:40 sof-icl.ri
-rw-r--r-- 1 root root 115472 Jan 23 06:40 sof-imx8.ri
-rw-r--r-- 1 root root 176128 Jan 23 06:40 sof-jsl.ri
-rw-r--r-- 1 root root 151552 Jan 23 06:40 sof-kbl.ri
-rw-r--r-- 1 root root 151552 Jan 23 06:40 sof-skl.ri
-rw-r--r-- 1 root root 180224 Jan 23 06:40 sof-sue.ri

/lib/firmware/intel/sof-tplg:
total 1668
-rw-r--r-- 1 root root 33227 Jan 23 06:40 sof-apl-da7219.tplg
-rw-r--r-- 1 root root 23311 Jan 23 06:40 sof-apl-demux-pcm512x.tplg
-rw-r--r-- 1 root root 10295 Jan 23 06:40 sof-apl-dmic-2ch.tplg
-rw-r--r-- 1 root root 10431 Jan 23 06:40 sof-apl-dmic-4ch.tplg
-rw-r--r-- 1 root root 10295 Jan 23 06:40 sof-apl-dmic-a2ch-b2ch.tplg
-rw-r--r-- 1 root root 10363 Jan 23 06:40 sof-apl-dmic-a2ch-b4ch.tplg
-rw-r--r-- 1 root root  5223 Jan 23 06:40 sof-apl-dmic-a2ch.tplg
-rw-r--r-- 1 root root 10363 Jan 23 06:40 sof-apl-dmic-a4ch-b2ch.tplg
-rw-r--r-- 1 root root  5223 Jan 23 06:40 sof-apl-dmic-b2ch.tplg
-rw-r--r-- 1 root root  6743 Jan 23 06:40 sof-apl-eq-dmic.tplg
-rw-r--r-- 1 root root  7223 Jan 23 06:40 sof-apl-eq-pcm512x.tplg
-rw-r--r-- 1 root root  9019 Jan 23 06:40 sof-apl-keyword-detect.tplg
-rw-r--r-- 1 root root 38003 Jan 23 06:40 sof-apl-nocodec.tplg
-rw-r--r-- 1 root root  5127 Jan 23 06:40 sof-apl-pcm512x-nohdmi.tplg
-rw-r--r-- 1 root root 25031 Jan 23 06:40 sof-apl-pcm512x.tplg
-rw-r--r-- 1 root root 19815 Jan 23 06:40 sof-apl-rt298.tplg
-rw-r--r-- 1 root root  5791 Jan 23 06:40 sof-apl-src-50khz-pcm512x.tplg
-rw-r--r-- 1 root root 11623 Jan 23 06:40 sof-apl-src-dmic.tplg
-rw-r--r-- 1 root root  5791 Jan 23 06:40 sof-apl-src-pcm512x.tplg
-rw-r--r-- 1 root root 43619 Jan 23 06:40 sof-apl-tdf8532.tplg
-rw-r--r-- 1 root root  5127 Jan 23 06:40 sof-apl-wm8804.tplg
-rw-r--r-- 1 root root 12579 Jan 23 06:40 sof-bdw-rt286.tplg
-rw-r--r-- 1 root root 12579 Jan 23 06:40 sof-bdw-rt5640.tplg
-rw-r--r-- 1 root root 13371 Jan 23 06:40 sof-byt-da7213.tplg
-rw-r--r-- 1 root root 13107 Jan 23 06:40 sof-byt-nocodec.tplg
-rw-r--r-- 1 root root 13371 Jan 23 06:40 sof-byt-rt5640.tplg
-rw-r--r-- 1 root root 13371 Jan 23 06:40 sof-byt-rt5645.tplg
-rw-r--r-- 1 root root 13371 Jan 23 06:40 sof-byt-rt5651.tplg
-rw-r--r-- 1 root root 13371 Jan 23 06:40 sof-byt-rt5682.tplg
-rw-r--r-- 1 root root 13371 Jan 23 06:40 sof-cht-max98090.tplg
-rw-r--r-- 1 root root 13107 Jan 23 06:40 sof-cht-nocodec.tplg
-rw-r--r-- 1 root root  7927 Jan 23 06:40 sof-cht-rt5682.tplg
-rw-r--r-- 1 root root  5555 Jan 23 06:40 sof-cht-src-50khz-pcm512x.tplg
-rw-r--r-- 1 root root 34187 Jan 23 06:40 sof-cml-demux-rt5682-max98357a.tplg
-rw-r--r-- 1 root root 29283 Jan 23 06:40 sof-cml-demux-rt5682.tplg
-rw-r--r-- 1 root root 39787 Jan 23 06:40 sof-cml-rt1011-rt5682.tplg
-rw-r--r-- 1 root root 34883 Jan 23 06:40 sof-cml-rt5682-kwd.tplg
-rw-r--r-- 1 root root 39787 Jan 23 06:40 sof-cml-rt5682-max98357a.tplg
-rw-r--r-- 1 root root 31123 Jan 23 06:40 sof-cml-rt5682.tplg
-rw-r--r-- 1 root root 32719 Jan 23 06:40 sof-cml-rt700.tplg
-rw-r--r-- 1 root root 34411 Jan 23 06:40 sof-cml-rt711-rt1308-mono-rt715.tplg
-rw-r--r-- 1 root root 39295 Jan 23 06:40 sof-cml-rt711-rt1308-rt715.tplg
-rw-r--r-- 1 root root 26751 Jan 23 06:40 sof-cml-src-rt5682.tplg
-rw-r--r-- 1 root root 25715 Jan 23 06:40 sof-cnl-nocodec.tplg
-rw-r--r-- 1 root root 11335 Jan 23 06:40 sof-cnl-rt274.tplg
-rw-r--r-- 1 root root 43555 Jan 23 06:40 sof-glk-da7219-kwd.tplg
-rw-r--r-- 1 root root 34831 Jan 23 06:40 sof-glk-da7219.tplg
-rw-r--r-- 1 root root 33967 Jan 23 06:40 sof-glk-rt5682.tplg
-rw-r--r-- 1 root root 45375 Jan 23 06:40 sof-hda-generic-2ch.tplg
-rw-r--r-- 1 root root 45511 Jan 23 06:40 sof-hda-generic-4ch.tplg
-rw-r--r-- 1 root root 28723 Jan 23 06:40 sof-hda-generic-idisp-2ch.tplg
-rw-r--r-- 1 root root 28859 Jan 23 06:40 sof-hda-generic-idisp-4ch.tplg
-rw-r--r-- 1 root root 16135 Jan 23 06:40 sof-hda-generic-idisp.tplg
-rw-r--r-- 1 root root 32787 Jan 23 06:40 sof-hda-generic.tplg
-rw-r--r-- 1 root root 12579 Jan 23 06:40 sof-hsw-rt5640.tplg
-rw-r--r-- 1 root root  4259 Jan 23 06:40 sof-icl-dmic-4ch.tplg
-rw-r--r-- 1 root root  7327 Jan 23 06:40 sof-icl-nocodec.tplg
-rw-r--r-- 1 root root 34883 Jan 23 06:40 sof-icl-rt5682-kwd.tplg
-rw-r--r-- 1 root root 31123 Jan 23 06:40 sof-icl-rt5682.tplg
-rw-r--r-- 1 root root 32719 Jan 23 06:40 sof-icl-rt700.tplg
-rw-r--r-- 1 root root 39295 Jan 23 06:40 sof-icl-rt711-rt1308-rt715-hdmi.tplg
-rw-r--r-- 1 root root 24643 Jan 23 06:40 sof-icl-rt711-rt1308-rt715.tplg
-rw-r--r-- 1 root root  7291 Jan 23 06:40 sof-imx8qxp-cs42888.tplg
-rw-r--r-- 1 root root  5091 Jan 23 06:40 sof-imx8qxp-nocodec-sai.tplg
-rw-r--r-- 1 root root  5091 Jan 23 06:40 sof-imx8qxp-nocodec.tplg
-rw-r--r-- 1 root root  7291 Jan 23 06:40 sof-imx8qxp-wm8960.tplg
-rw-r--r-- 1 root root 25647 Jan 23 06:40 sof-tgl-nocodec.tplg
-rw-r--r-- 1 root root 25071 Jan 23 06:40 sof-tgl-rt711-rt1308-nohdmi.tplg
-rw-r--r-- 1 root root 44607 Jan 23 06:40 sof-tgl-rt711-rt1308.tplg
-rw-r--r-- 1 root root 29283 Jan 23 06:40 sof-whl-demux-rt5682.tplg
-rw-r--r-- 1 root root 34883 Jan 23 06:40 sof-whl-rt5682-kwd.tplg
-rw-r--r-- 1 root root 31123 Jan 23 06:40 sof-whl-rt5682.tplg

```
`cat /etc/modprobe.d/*.conf | grep snd` : 
```
blacklist snd_soc_sst_acpi
blacklist snd_soc_sst_dsp
blacklist snd_soc_sst_firmware
blacklist snd_soc_sst_ipc
blacklist snd_soc_sst_match
blacklist snd_soc_sst_mfld-platform
blacklist snd_soc_skl
blacklist snd_soc_sst_byt_cht_nocodec
blacklist snd_soc_sst_atom_hifi2_platform
blacklist snd_intel_sst_core
blacklist snd_intel_sst_acpi
#blacklist snd-hda-intel
#options snd-hda-intel model=dell-headset-multi
```
`/usr/sbin/alsa-info.sh` : http://alsa-project.org/db/?f=857d3dcb9e8db782074beb4b522f38a548dcfa40
`alsamixer` : 
[image]

ranj063, 2020-05-13:
@LazyGeniusMan it doesnt look like the SOF driver is getting probed at all. Is SOF enabled in the kernel config?

ranj063, 2020-05-13:
Could you paste the output of `lsmod | grep snd'?

LazyGeniusMan, 2020-05-13:
> @LazyGeniusMan it doesnt look like the SOF driver is getting probed at all. Is SOF enabled in the kernel config?

How to do that? sorry I am very new in Linux, about a week. After a quick search I don't quite understand, is it has something to do with building custom kernel? all I get is there's config file in `/proc/config.gz`

> Could you paste the output of `lsmod | grep snd'?
```
snd_seq_dummy          16384  0
snd_hrtimer            16384  2
snd_seq                86016  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
snd_hda_codec_hdmi     73728  1
snd_hda_codec_realtek   126976  1
snd_hda_codec_generic    94208  1 snd_hda_codec_realtek
ledtrig_audio          16384  2 snd_hda_codec_generic,snd_hda_codec_realtek
snd_hda_intel          57344  3
snd_intel_dspcfg       28672  1 snd_hda_intel
snd_hda_codec         163840  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core          106496  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_hwdep              16384  1 snd_hda_codec
snd_pcm               143360  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
snd_timer              40960  3 snd_seq,snd_hrtimer,snd_pcm
snd                   110592  18 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm
soundcore              16384  1 snd
```

LazyGeniusMan, 2020-05-13:
@ranj063 
In case this is needed `zcat /proc/config.gz | grep SOF` :
```
CONFIG_EFI_SOFT_RESERVE=y
CONFIG_HAVE_ARCH_SOFT_DIRTY=y
# CONFIG_MEM_SOFT_DIRTY is not set
CONFIG_CAN_SOFTING=m
CONFIG_CAN_SOFTING_CS=m
CONFIG_SOFT_WATCHDOG=m
# CONFIG_SOFT_WATCHDOG_PRETIMEOUT is not set
CONFIG_VGACON_SOFT_SCROLLBACK=y
CONFIG_VGACON_SOFT_SCROLLBACK_SIZE=64
# CONFIG_VGACON_SOFT_SCROLLBACK_PERSISTENT_ENABLE_BY_DEFAULT is not set
CONFIG_SND_SOC_INTEL_SOF_RT5682_MACH=m
CONFIG_SND_SOC_INTEL_SOF_CML_RT1011_RT5682_MACH=m
CONFIG_SND_SOC_INTEL_SOF_DA7219_MAX98373_MACH=m
CONFIG_SND_SOC_SOF_TOPLEVEL=y
CONFIG_SND_SOC_SOF_PCI=m
CONFIG_SND_SOC_SOF_ACPI=m
CONFIG_SND_SOC_SOF_OF=m
# CONFIG_SND_SOC_SOF_DEVELOPER_SUPPORT is not set
CONFIG_SND_SOC_SOF=m
CONFIG_SND_SOC_SOF_PROBE_WORK_QUEUE=y
CONFIG_SND_SOC_SOF_INTEL_TOPLEVEL=y
CONFIG_SND_SOC_SOF_INTEL_ACPI=m
CONFIG_SND_SOC_SOF_INTEL_PCI=m
CONFIG_SND_SOC_SOF_INTEL_HIFI_EP_IPC=m
CONFIG_SND_SOC_SOF_INTEL_ATOM_HIFI_EP=m
CONFIG_SND_SOC_SOF_INTEL_COMMON=m
CONFIG_SND_SOC_SOF_MERRIFIELD_SUPPORT=y
CONFIG_SND_SOC_SOF_MERRIFIELD=m
CONFIG_SND_SOC_SOF_APOLLOLAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_APOLLOLAKE=m
CONFIG_SND_SOC_SOF_GEMINILAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_GEMINILAKE=m
CONFIG_SND_SOC_SOF_CANNONLAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_CANNONLAKE=m
CONFIG_SND_SOC_SOF_COFFEELAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_COFFEELAKE=m
CONFIG_SND_SOC_SOF_ICELAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_ICELAKE=m
CONFIG_SND_SOC_SOF_COMETLAKE_LP=m
CONFIG_SND_SOC_SOF_COMETLAKE_LP_SUPPORT=y
CONFIG_SND_SOC_SOF_COMETLAKE_H=m
CONFIG_SND_SOC_SOF_COMETLAKE_H_SUPPORT=y
CONFIG_SND_SOC_SOF_TIGERLAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_TIGERLAKE=m
CONFIG_SND_SOC_SOF_ELKHARTLAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_ELKHARTLAKE=m
CONFIG_SND_SOC_SOF_JASPERLAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_JASPERLAKE=m
CONFIG_SND_SOC_SOF_HDA_COMMON=m
CONFIG_SND_SOC_SOF_HDA_LINK=y
CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y
# CONFIG_SND_SOC_SOF_HDA_ALWAYS_ENABLE_DMI_L1 is not set
CONFIG_SND_SOC_SOF_HDA_COMMON_HDMI_CODEC=y
CONFIG_SND_SOC_SOF_HDA_LINK_BASELINE=m
CONFIG_SND_SOC_SOF_HDA=m
CONFIG_SND_SOC_SOF_XTENSA=m
CONFIG_HID_MICROSOFT=m
# CONFIG_USB_DWC2_TRACK_MISSED_SOFS is not set
CONFIG_SPEAKUP_SYNTH_SOFT=m
CONFIG_ZISOFS=y
CONFIG_SOFTLOCKUP_DETECTOR=y
# CONFIG_BOOTPARAM_SOFTLOCKUP_PANIC is not set
CONFIG_BOOTPARAM_SOFTLOCKUP_PANIC_VALUE=0
# CONFIG_IRQSOFF_TRACER is not set

```

ranj063, 2020-05-13:
@LazyGeniusMan thanks. so the snd_sof_pci module has not loaded and the kernel config seems to enable it. @plbossart any ideas?

kv2019i, 2020-05-13:
@LazyGeniusMan This is a Kabylake system:
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21)

SOF is not support these systems. SOF is enabled as the default driver for 7th Gen (Whiskey Lake) and newer in upstream. In latest upstream, the Intel SST driver was enabled for 7th Gen Kabylake systems that have a PCH digital mic (see patch "ALSA: hda: Allow SST driver on SKL and KBL platforms with DMIC").

LazyGeniusMan, 2020-05-13:
> SOF is not support these systems. SOF is enabled as the default driver for 8th Gen (Whiskey Lake) and newer in upstream. In latest upstream, the Intel SST driver was enabled for 7th Gen Kabylake systems that have a PCH digital mic (see patch "ALSA: hda: Allow SST driver on SKL and KBL platforms with DMIC").

dang, how come generation after mine worked first :(, I should've bought i3 varian for this laptop since it has 7th gen Kaby Lake (i3-7020u) or wait a few months for 8th Whiskey Lake or even bought the Ryzen varian, what a regret haha.

@kv2019i 

> As stated, DMIC with HDA configuration is supported officially only with Realtek chips. I'm gonna be honest here: with legacy stack failing to support your model too, my hands are tied and I won't be able to solve your case with desirable result.
What's left, is for me to speak with Takashi about addition of said codec to known-codecs table - more often than not, given codec belongs to entire family of chips and thus shares most - if not all - of the behavior routines. That won't help you with DMIC a bit unfortunately, as it requires link (DSP) capability to be up and running. I'll file shipping order for an exemplar of your laptop. No promises, prefer to stay honest. 

I don't quite understand, is something about Realtek chips is good or not? and what's "ALSA: hda: Allow SST driver on SKL and KBL platforms with DMIC" commit try to achieve?

kv2019i, 2020-05-13:
@LazyGeniusMan wrote:
> > As stated, DMIC with HDA configuration is supported officially only with Realtek chips. I'm 

It took me a while where is this quote from. :) I put the quote to a web search and apparently this is from here: https://bugzilla.kernel.org/show_bug.cgi?id=201251#c182 , right? 

I don't think that necessarily applies to your case. For a fully working audio, you will need a working BIOS (from your laptop vendor) and codec drivers (from Realtek, Conexant or upstream), in addition to the Intel drivers for the HDA controller and DSP Intel provides.

I think on your Swift, the ALC256 codec at least has a driver in upstream kernel, so that's good. What you need is is the HDA DSP driver for Kabylake.

>> Update intel-dspcfg with FLAG_SST_ONLY_IF_DMIC option and use it for
>> Skylake and Kabylake platforms when DMIC is present.
> 
> I don't quite understand, is something about Realtek chips is good or not? and what's "ALSA: hda: Allow SST driver on SKL and KBL platforms with DMIC" commit try to achieve here?

That patch adds logic to recognize types of systems you have (7th gen Intel CPU with a DMIC connected to PCH), and chooses the SST driver. Without the patch, the snd-hda-intel driver is chosen instead and you get audio but DMIC is not enabled (so you have no microphone). Cezary (who you quoted above) is the maintainer of SST driver in upstream.

LazyGeniusMan, 2020-05-13:
> It took me a while where is this quote from. :) I put the quote to a web search and apparently this is from here: https://bugzilla.kernel.org/show_bug.cgi?id=201251#c182 , right?

Yes, that's right. My bad for not linking it, I thought you were coming from that by saying "see patch "ALSA: hda: Allow SST driver on SKL and KBL platforms with DMIC".

> That patch adds logic to recognize types of systems you have (7th gen Intel CPU with a DMIC connected to PCH), and chooses the SST driver.
oh, I thought when they are mentioning Kaby Lake it's for 8th gen.

> Without the patch, the snd-hda-intel driver is chosen instead and you get audio but DMIC is not enabled (so you have no microphone)

So that's like my issue here, right ?

Also I wanna ask, if I buy USB Sound card, microphone or headset with USB port, will this issue will affect those device too?

kv2019i, 2020-05-13:
@LazyGeniusMan wrote:
> oh, I thought when they are mentioning Kaby Lake it's for 8th gen.

ark.intel.com is your friend to parse the CPU product names, e.g. https://ark.intel.com/content/www/us/en/ark/products/codename/82879/kaby-lake.html

>> Without the patch, the snd-hda-intel driver is chosen instead and you get audio but DMIC is not enabled (so you have no microphone)
> 
> So that's like my issue here, right ?

Indeed, that's why I was mentioning it. I recommend filing a bug to bugzilla.kernel.org which is more appropriate tracker for bugs in the SST driver.

> Also I wanna ask
> 
>     * If I buy USB Sound card, microphone or headset with USB port, will this issue affect those device too?

Correct, USB is completely separate.

>     * Also If I buy microphone or headset with jack port, will this issue affect those device ?, with no mannually created .conf file in /etc/modprobe.d/ input device is not detected. But if i add 'ptions snd-hda-intel model=dell-headset-multi` my IEM is working, both input and output. Can I assume that headset or microphone with jack port 100% will work too ?

I cannot really confirm this -- the headset jack goes via HDA controller and codec as well, so problems with the Linux codec driver could have an impact. Probably it will work, but depends on Acer's integration and the codec driver.

LazyGeniusMan, 2020-05-13:
> ark.intel.com is your friend to parse the CPU product names

Oh the 8th gen is Kaby Lake R, i didn't notice the "R" before.

> Correct, USB is completely separate.

You mean by "Correct" is the USB device will work?, I was asking "Willi this issue affect Audio Device with USB port". "Correct" for the answer kind of confused me. 

If I remember it correctly I've read something like "Audio Device with USB port come with it's own Sound Card and will use it's own Sound Card. Audio Device with jack port will use internal Sound Card (from motherboard) unless it's connected to jack port from external Sound Card/USB Sound Card", is that correct ?

plbossart, 2020-05-13:
@LazyGeniusMan USB operation is orthogonal to that of the internal PCI audio device, so no impact/correlation.

LazyGeniusMan, 2020-05-14:
> @LazyGeniusMan USB operation is orthogonal to that of the internal PCI audio device, so no impact/correlation.

Ok thanks for response.

**CONCLUSION**
Kaby Lake R (Intel 8th gen) is not supported yet. So internal mic and external mic won't work. For internal mic I haven't found any workaround, but for external mic through laptop's jack port you can make it work by adding `options snd-hda-intel model=dell-headset-multi` in `/etc/modprobe.d/*.conf`, if external mic still not detected change audio setting to duplex.

Another way is to use USB Audio Device or USB Sound Card (if you want use Audio Device with jack port without any tweak)

kv2019i, 2020-05-14:
@LazyGeniusMan wrote:
> **CONCLUSION**
> Kaby Lake R (Intel 8th gen) is not supported yet. So internal mic and external mic won't work. For 

Slight correction for the records: Not supported by this project (SOF firmware). It is supported in upstream Linux via Intel SST driver and firmware for older generations of hardware:

sound/soc/intel/skylake/skl.c

#if IS_ENABLED(CONFIG_SND_SOC_INTEL_KBL)
»       /* KBL */
»       { PCI_DEVICE(**0x8086, 0x9D71**),
»       »       .driver_data = (unsigned long)&snd_soc_acpi_intel_kbl_machines},
#endif

That matches alsa-info output from your machine:

00:1f.3 Multimedia audio controller [0401]: Intel Corporation Sunrise Point-LP HD Audio [**8086:9d71**] (rev 21)

closed 2020-05-19 by slawblauciak