thesofproject/sof #2932: [BUG] IIR EQ fails to configure with long data blob

singalsu, 2020-05-13:
**Describe the bug**
The IIR configuration max size is 1024 bytes but a large enough configuration blob that is split to parts by process components IPC fails in iir_cmd_set_data().

**To Reproduce**
Create a long configuration data blob. I made it from loudness EQ by adding to blob four more response shapes (modified example_iir_eq.m script). Then apply it with sof-ctl and playback audio.

**Reproduction Rate**
100%

**Expected behavior**
Accepted IPC and working audio playback.

**Impact**
Can be a showstopper if higher order or multi-channel EQ response is configured.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
```
[ 68610671.041667] (      143.125000) c0 eq-iir       1.2   ../audio/eq_iir/eq_iir.c:755   eq_iir_cmd()
[ 68610674.635417] (        3.593750) c0 eq-iir       1.2   ../audio/eq_iir/eq_iir.c:691   iir_cmd_set_data(), SOF_CTRL_CMD_BINARY
[ 68610984.843750] (      310.208344) c0 eq-iir       1.2   ../audio/eq_iir/eq_iir.c:755   eq_iir_cmd()
[ 68610988.437500] (        3.593750) c0 eq-iir       1.2   ../audio/eq_iir/eq_iir.c:691   iir_cmd_set_data(), SOF_CTRL_CMD_BINARY
[ 68610991.406250] (        2.968750) c0 eq-iir       1.2   ../audio/eq_iir/eq_iir.c:698   ERROR iir_cmd_set_data(), size 2064935920 is invalid
[ 68610994.322917] (        2.916667) c0 ipc                       src/ipc/handler.c:1006  ERROR ipc: comp 2 cmd 3 failed -22

```

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

[eq_iir_loudness_long.txt](https://github.com/thesofproject/sof/files/4621071/eq_iir_loudness_long.txt)

singalsu, 2020-05-13:
The PR with fix is in #2933.

slawblauciak, 2020-05-18:
@singalsu good to close?

singalsu, 2020-05-18:
@slawblauciak Sure, closing now!

closed 2020-05-18 by singalsu