thesofproject/sof #2936: [TGL]: S0i3 is blocked by AUDIO PLL being ON

sathya-nujella, 2020-05-14:
**Describe the bug**
On TGL DUT:  DSP IP shows, it is in D3, but AUDIO PLL is ON. This is preventing DUT to enter S0i3.

**To Reproduce** 
1) Flash S0ix enabled coreboot and EC on the Tiger Lake board with default DB (Max98357a + ALC5682I) -> Boot DUT -> run 'suspend_stress_test -c 1' which puts system to S0ix and wakes it 
2) Check AUDIO PLL status when system is in suspend state

[Note: During this test, not running any audio tests]

**Reproduction Rate**
100%

**Expected behavior**
Audio should not block DUT entering S0i3

**Impact**
Power

**Environment**
TGL with V0.4/V0.5 release.

More debug details:
===========
1) RTD3 seems good:
cat /sys/bus/pci/devices/0000\:00\:1f.3/power/runtime_enabled
	enabled
cat /sys/bus/pci/devices/0000\:00\:1f.3/power/runtime_status
	suspended
2)
[   24.180280] snd_sof_runtime_suspend
[   24.180299] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000
[   24.180508] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000
[   24.180626] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[   24.180641] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1000f0f successful
[   24.180647] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[   24.180753] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000000 at  00000044
[   24.180771] sof-audio-pci 0000:00:1f.3: Debug PCIW: 00000010 at  00000044
[   24.180787] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power 0
[   24.180790] sof-audio-pci 0000:00:1f.3: Current DSP power state: D3

3) ****Output of LPM registers which shows the audio problem****
Pasting only relevant Audio here:  cat /sys/kernel/debug/pmc_core/substate_status_registers

a] AudioPLL_OFF_STS               0  <<<< [meaning of this bit:  "1 means clock source is off"]
    so this AUDIO PLL is ON
b] IP status 
ADSP_D3_STS                    1  [1 meaning ---> IP is in D3]

keyonjie, 2020-05-18:
@sathya-nujella so do you mean that s0ix residencies don't get increased even no any audio is run?

sathya-nujella, 2020-05-18:
> @sathya-nujella so do you mean that s0ix residencies don't get increased even no any audio is run?

I checked with my S0ix enabling team. Here are the details.
TGL supports different S0ix states: S0i2, S0i3 and other sub states like S0i2.1/S0i2.2/S0i3.1 etc.
If system is able to successfully enter S0i2, then Low Power Mode 0 Residency counter (LPM_0_RES = S0i2 residency counter) will be incremented.
Like wise, if system enters successfully S0i3 ==> LPM_3_RES/S0i3.0 residency counter will be increased.

Current status:
with no audio activity (no pb/cap streams..  just DUT is in idle and putting system to suspend state).
a. System is able to go to S0i2 ==> S0i2 Counters increased (i.e. LPM_0_RES counter increased).

b. System not able to enter S0i3. One of the blockers is Audio. Hence S0i3 counter not increased.

keyonjie, 2020-05-19:
OK, Thanks for the details @sathya-nujella

BTW, we do need the Chrome environment to duplicate and debug the issue.

sathya-nujella, 2020-05-20:
> OK, Thanks for the details @sathya-nujella
> 
> BTW, we do need the Chrome environment to duplicate and debug the issue.

We are checking on Chrome DUT as we are able to put it in to S0ix. Not sure on other OS's. On your system, can you please put DUT to run time D3.
Check if audio pll shows up as in step-3 of bug description. But, I don't know if that is right method for this problem to debug. Please check.

jajanusz, 2020-05-21:
Which MCLK/BCLK do you use? Does issue happen when you use only XTAL for these clocks (NO PLL at all) ?

jajanusz, 2020-05-21:
There is no code responsible in FW for powering PLL, we just select source for mclks/bclks (it's xtal by default), but maybe we should switch back to XTAL when PLL is no longer used, however I need to know if issue is still here if you use only XTAL.

ranj063, 2020-05-21:
> There is no code responsible in FW for powering PLL, we just select source for mclks/bclks (it's xtal by default), but maybe we should switch back to XTAL when PLL is no longer used, however I need to know if issue is still here if you use only XTAL.

@jajanusz just to confirm, there is no active audio here. The DSP is in D3 but the audio PLL is still on.

jajanusz, 2020-05-21:
> @jajanusz just to confirm, there is no active audio here. The DSP is in D3 but the audio PLL is still on.

I know, but if the last used topology (before puasing/stopping audio) used PLL it's still selected as mclk/bclk source (instead of xtal), I was trying to figure out if it can impact if it's powered on even if it's not actively used.

sathya-nujella, 2020-05-22:
@jajanusz 
we are testing on a DUT which uses this tplg:
https://github.com/thesofproject/sof/blob/master/tools/topology/sof-tgl-max98357a-rt5682.m4

sathya-nujella, 2020-05-25:
Tried this experiment, still audio PLL is ON:
"Built SOF  fw with below two changes:
a) LPS - sof/pull/2951
b) LPRO :
Not sure if this is the right method to completly enable LPRO -- but sharing the result"
./src/arch/xtensa/configs/override/tigerlake_chrome.config
	+CONFIG_CAVS_LPRO=y
./src/arch/xtensa/configs/tigerlake_defconfig
	+CONFIG_CAVS_LPRO=y
"
Used this built SOF fw during the test..

keyonjie, 2020-05-26:
@sathya-nujella you can run 'make menuconfig' in the build_tgl_xcc folder and select it under "CAVS  --->" folder:
```
(top menu) → CAVS
                                                                                            SOF  Configuration
(46) HP memory banks count
(1) LP memory banks count
[*] Intel cAVS Low Power Sequencer for Power Management

```
after saved, you will see "CONFIG_CAVS_LPS=y" in generated/.config

sathya-nujella, 2020-05-26:
> @sathya-nujella you can run 'make menuconfig' in the build_tgl_xcc folder and select it under "CAVS --->" folder:
>
> ```
> (top menu) → [...]

Thank you @keyonjie. Let me cross check and update result again.

sathya-nujella, 2020-05-27:
> @sathya-nujella you can run 'make menuconfig' in the build_tgl_xcc folder and select it under "CAVS --->" folder:
>
> ```
> (top menu) → [...]

now enabled LPS and also cross checked by adding build error.. it is taking effect.
This is what we did: 0_6 release fw tag + sof/pull/2951 + LPS config enable.

Result is same. Audio PLL is on during S0ix (suspend/resume)
LPM_STATUS_0:	0xf57c00f4
AudioPLL_OFF_STS               0

jajanusz, 2020-06-09:
@sathya-nujella 
Can you try somehow using only XTAL-divisable frequencies so (PLL is never used) in one run?

jajanusz, 2020-06-09:
@keyonjie @sathya-nujella 
Can you try to cherry-pick this patch: https://github.com/jajanusz/sof/commit/c49d3e1c48cb2895fc797ea7725cb44896079dc4

It resets MN registers (that control MCLK/BCLK) when you free all SSPs.

jajanusz, 2020-06-09:
Also can you do checks with just one SSP being on?
I see you use:
```
# SSP 1 (ID: 7)
DAI_CONFIG(SSP, 1, 7, SSP1-Codec,
	SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
		SSP_CLOCK(bclk, 1536000, codec_slave),
		SSP_CLOCK(fsync, 48000, codec_slave),
		SSP_TDM(2, 16, 3, 3),
		SSP_CONFIG_DATA(SSP, 1, 16)))
# SSP 0 (ID: 0)
DAI_CONFIG(SSP, 0, 0, SSP0-Codec,
	SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),
		SSP_CLOCK(bclk, 2400000, codec_slave),
		SSP_CLOCK(fsync, 48000, codec_slave),
		SSP_TDM(2, 25, 3, 3),
		SSP_CONFIG_DATA(SSP, 0, 24)))
```

Can you run it with just SSP0 and then just SSP1 and see if issue is still there?

keyonjie, 2020-06-09:
Hi @jajanusz haven't tried your suggestion yet, but it looks to me that all 19.2M, 1.536M, and 2.4M is dividable by 38.4M (XTAL), so the PLL(96M) and the cardinal clock (24.576M) won't be used?

BTW, I just heard from @mrajwa that he has locally fixed this issue, so better sync with him about the solution also.

plbossart, 2020-06-09:
@keyonjie IIRC some clock sources are not available in all S0ix states so indeed we need to be careful.

sathya-nujella, 2020-06-09:
> @keyonjie @sathya-nujella
> Can you try to cherry-pick this patch: [jajanusz@b8ea60d](https://github.com/jajanusz/sof/commit/b8ea60dab0b6a331171cf20619ed862b2f894968)
> 
> It resets MN registers (that control MCLK/BCLK) when you free all SSPs.
> 
> `Edit: changed commit to b8ea60dab0b6a331171cf20619ed862b2f894968`

Sure Janusz. Let me try it and update.

keyonjie, 2020-06-10:
> @keyonjie IIRC some clock sources are not available in all S0ix states so indeed we need to be careful.

Yes, for example, using the Audio PLL (96MHz) or HP ROSC will prevent the system entering to S0i3.x here (but fine for S0i2.0~S0i2.2). When looking to the spec, what makes me concern more is that we might need to do more (e.g. power gated the LP ROSC and XTAL) before we can enter S0i3, though I am not quite sure about this.

FWIW, @mrajwa is working on the solution here.

sathya-nujella, 2020-06-10:
> Yes, for example, using the Audio PLL (96MHz) or HP ROSC will prevent the system entering to S0i3.x here (but fine for S0i2.0~S0i2.2). When looking to the spec, what makes me concern more is that we [...]

I tried the below:
Changes:  patch from this PR: linux/pull/2184   +  0_6 firmware with LPRO enabled.

Result:  Audio PLL is ON. So, we still have issue.

cat /sys/kernel/debug/pmc_core/substate_status_registers
AudioPLL_OFF_STS               0

sathya-nujella, 2020-06-11:
> @keyonjie @sathya-nujella
> Can you try to cherry-pick this patch: [jajanusz@b8ea60d](https://github.com/jajanusz/sof/commit/b8ea60dab0b6a331171cf20619ed862b2f894968)
>
> It resets MN registers (that [...]

on 0_6 release, we can not apply this patch because in 0_6 release we don't have this new ssp driver changes.
So, I directly checked out to janusz branch: jajanusz/mn-free-mclk and built firmware.
I don't see this function mn_release_mclk() getting called in sof logs.
Not sure if I miss some thing?

sathya-nujella, 2020-06-11:
> on 0_6 release, we can not apply this patch because in 0_6 release we don't have this new ssp driver changes.
> So, I directly checked out to janusz branch: jajanusz/mn-free-mclk and built firmware.
> I [...]

cross checked again,
Audio PLL is ON
cat /sys/kernel/debug/pmc_core/substate_status_registers | grep LPM
LPM_STATUS_0:   0xf57c00f4
23rd bit: AUDIO_PLL off status: 0 <<indicating audio PLL is ON

keyonjie, 2020-06-12:
Hi @jajanusz could you help to add a regular FW trace log to print out the clock power status, e.g. including CLKCTL, CLKSTS, PWRCTL and PWR_STS ?

jajanusz, 2020-06-15:
@keyonjie I can, but IMO it's wrong place to look, cos CLK* functions are for oscillators for CPU not for SSPs.

@sathya-nujella This function has no logs, but if you free'd all SSPs it should be called.

jajanusz, 2020-06-15:
> cross checked again,
> Audio PLL is ON
> cat /sys/kernel/debug/pmc_core/substate_status_registers | grep LPM
> LPM_STATUS_0: 0xf57c00f4
> 23rd bit: AUDIO_PLL off status: 0 <<indicating audio PLL is ON

@keyonjie @sathya-nujella 
Do you know at which point PLL gets turned on?
Is it already on after FW is loaded or after first SSP is configured/startedplaying/etc?

keyonjie, 2020-06-15:
> > cross checked again,
> > Audio PLL is ON
> > cat /sys/kernel/debug/pmc_core/substate_status_registers | grep LPM
> > LPM_STATUS_0: 0xf57c00f4
> > 23rd bit: AUDIO_PLL off status: 0 <<indicating audio PLL is ON
> 
> @keyonjie @sathya-nujella
> Do you know at which point PLL gets turned on in your case (by checking that bit)?
> Is it already on after FW is loaded or after first SSP is configured/startedplaying/etc?

It looks it is always on, even before the audio driver is loaded.

sathya-nujella, 2020-06-15:
Also one more experiment I did, FYI:
I removed all end like speaker/headphone/hdmi from tplg and kept only DMIC.  I still see Audio PLL ON.
==
localhost ~ # aplay -l
**** List of PLAYBACK Hardware Devices ****
localhost ~ # arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofrt5682 [sof-rt5682], device 99: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

mrajwa, 2020-06-16:
@sathya-nujella, we know that. That's why on the email thread I wrote we need to verify:

a. PMC testing screepts (are they properly aligned for TGL)? What they show is suspicious and contradicts our settings.
b. Coreboot - they may initialize something that is using this PLL

keyonjie, 2020-06-24:
Hi @sathya-nujella so looks we are good for this issue since we can already enter to s0i3.0, so let's wait for merge of thesofproject/linux#2184 and then close it?

sathya-nujella, 2020-06-24:
> Hi @sathya-nujella so looks we are good for this issue since we can already enter to s0i3.0, so let's wait for merge of [thesofproject/linux#2184](https://github.com/thesofproject/linux/pull/2184) and then close it?

sure, thanks!

keyonjie, 2020-07-20:
Hi @sathya-nujella since https://github.com/thesofproject/linux/pull/2276 is merged, can you please check and close this issue here?

sathya-nujella, 2020-07-20:
> Hi @sathya-nujella since [thesofproject/linux#2276](https://github.com/thesofproject/linux/pull/2276) is merged, can you please check and close this issue here?

Yes @keyonjie , I double confirmed with team. They have some internal WIP solution with which DUT can go to S0i3.
Audio PLL is not blocker to go to S0i3.
Also I submitted a patch to pmc driver: https://patchwork.kernel.org/patch/11624647/    fixing the bit-mapping naming.  It is reviewed and accepted by PMC driver team.. waiting for it to merge in upstream. After that, will post to Google tree.  Thanks!

closed 2020-07-20 by sathya-nujella

keyonjie, 2020-07-21:
> Yes @keyonjie , I double confirmed with team. They have some internal WIP solution with which DUT can go to S0i3.
> Audio PLL is not blocker to go to S0i3.
> Also I submitted a patch to pmc driver: [...]

It's very good to change the confusion naming in the PMC driver team, thanks a lot for doing that!