thesofproject/sof #2937: [BUG] v1.5 - Digital Microphone issues

perexg, 2020-05-14:
Issues:

* the controls were (again!) renamed from "Dmic0 Capture Volume" to "Dmic0 10 Capture Volume" ?
* the dmic volume is weaker than with 1.4.2

Control settings:

````
numid=49,iface=MIXER,name='Dmic0 10 Capture Switch'
  ; type=BOOLEAN,access=rw------,values=2
  : values=on,on
numid=48,iface=MIXER,name='Dmic0 10 Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=70,step=0
  : values=70,70
  | dBscale-min=-50.00dB,step=1.00dB,mute=1
numid=51,iface=MIXER,name='Dmic1 Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=70,step=0
  : values=70,70
  | dBscale-min=-50.00dB,step=1.00dB,mute=1
````

Testing sof-bin - v1.5 branch...

juimonen, 2020-05-14:
@perexg, crap, this is clear bug from us... We obviously have still issues testing the full stack from kernel to user space, dmic being the hard one. It is caused by firmware commit  d65495e984ec922ed7bb3d25. This needs to be fixed.

Before we might have had also amplifying eq parameters so you could have better volume then, @singalsu FYI.

singalsu, 2020-05-14:
@perexg What device model is this and what topology name does the kernel load? We seem to have a confusion with gains, sorry for that. Also does the above indicate that you have already maximized the available user volume control for capture (70. 70)?

kv2019i, 2020-05-14:
FYI @lgirdwood @mengdonglin  @fredoh9 

This seems to be a regression with this patch:
https://github.com/thesofproject/sof/pull/2660/

The patch changed ALSA mixer controls which our UCMs rely on:
https://github.com/alsa-project/alsa-ucm-conf/blob/master/ucm2/sof-hda-dsp/HiFi.conf 

... I think this calls for a 1.5 hotfix. Can you @mengdonglin check how this passed validation?

perexg, 2020-05-14:
It's Lenovo Carbon X1 7th gen (sof-cnl.ri + sof-hda-generic-idisp-4ch.tplg).

juimonen, 2020-05-14:
@fredoh9 can you take a look at this? If I remove the pipeline numbering "index" values cause issues with alsatplg. Old topologies before your commit had dmic indexes close to each other, but now when the dmic0 and dmic1 index are separated far away, alsatplg can't compile same name mixer elements... 

@ranj063 you have any idea how this should work? And what's the meaning of index values in topology? I understood it was grouping the widgets together, but now I'm puzzled...

juimonen, 2020-05-14:
hmm, I think the change was made because we had 2 "Capture Volume" control definitions (name identical) for Dmic0 and Dmic1 in conf file. I think ASoC layer is combining these for alsamixer as, so we get nice "Dmic0 Capture Volume".

But as I now see the alsa-lib conf parsing code, it is actually merging the "duplicate" definitions, so we have in essence only 1 "Capture Volume" definition. It was working before, because the Dmic0 and Dmic1 "index" definition we're close to each other (something like 7 & 8) and alsa conf parsing survives this for some reason. So it finds the single "Capture Volume" for both Dmic0 and Dmic1. Now with change in 1.5 fw the indexes are far apart (like 3 vs 8) and alsa conf parser bails out. And that's probably @fredoh9 was adding the number to the control name.

I'm little bit confused how this index thing would work before and not anymore...

Anyway it seems that if we want to have 2 separate pga's with 2 separate controls, they all _need_ to have different name. This however makes the naming little bit funny as we would get:
"Dmic0 Capture Volume 0" and "Dmic1 Capture Volume 1". You can replace the last number with foo/bar, but this kind of explains why we have pga monster names in the topologies.

If I add a ?-mark like following to the 2 section names:
SectionControlMixer.?"Capture Volume"
I get the alsatplg to create the tplg binary with 2 capture volumes and the indexing doesn't matter, but I'm not sure about the semantics of the ?-sign here... 
It says "? = do not override" in alsaconf docs, but not sure it affects as we want here.

@perexg you have any insights of the above behavior, or any preference about the naming?

perexg, 2020-05-14:
Sorry, I don't get what you're doing with the alsa-lib configs. Could you show me the pieces you refer (from the .conf files not .m4 files)?

juimonen, 2020-05-14:
@perexg sorry for being confusing... I'm just trying to understand how the conf parser works and I need to jump between alsa-utils and alsa-lib.

Attached a conf where I removed the volume control numbering (working as the old way) and it doesn't compile with alsatplg. I renamed the conf to txt as you can't attach otherwise

- However if you change the second "Capture Volume" index from 8 to 4, it compiles(????). 
- Also if you add ?-sign in front of the MixerSections like: SectionControlMixer.?"Capture Volume" {
it compiles.
[testi.txt](https://github.com/thesofproject/sof/files/4630401/testi.txt)

perexg, 2020-05-14:
You can imagine the configuration like a tree. You have this in your config:

````
SectionControlMixer."Capture Volume" {
   Index 3
}
SectionControlMixer."Capture Volume" {
   Index 8
}
````

The result after load is:

````
SectionControlMixer."Capture Volume" {
   index 8
}
````

If I normalize your unreadable configs using `alsatplg -n t.conf  -o n.conf` the problem is visible:

````
SectionControlMixer {
  'Capture Switch' {
     index 3
  }
  `Capture Volume' {
     index 8
  }
}
````

There is no 'index 3' section, because configs get merged (identical identifier "Capture Volume" in the configuration tree) - last parsed entry wins in this case.

The 'Dmic0' in SectionWidget has 'index 3', thus you need to define the SectionControlMixer controls with this 'index 3'.

The '?' or '-' operations (?='replace' or minus='remove before parse') won't help you, because you cannot link blocks from the SectionWidget to SectionControlMixer with different indexes (functional groups).

Just choose a different name for "Capture Volume" for Dmic1 like "Second Capture Volume". It's ugly, but you're in trouble because nobody in ASoC use the right ALSA control indexes (do not mix with the functional topology group). ALSA allows multiple controls with the same name with the only integer index difference like "Capture Volume,0" and "Capture Volume,1". Another way is to put Dmic0 and Dmic1 mixer controls to the one functional group (index).

I would really suggest to add the proper support for the ALSA mixer control indexes to ASoC and topology.

perexg, 2020-05-14:
Thinking more: Perhaps, it may be sufficient to add a suffix to the control mixer name specified in SectionControlMixer and strip it in the topology library when the contents is written to the output binary file. Like "Capture Volume|Dmic0" where "|Dmic0" will be stripped. If I understand correctly, the identical mixer control names may exists if they're in different functional groups, right?

juimonen, 2020-05-15:
@perexg, Yes the idea was exactly that we would have same named controls, but in different indexes. So the prefixing pga name would be the differentiator. But as you said, with current methods, to get this right, we need to have different control names also.

Anyway this current method of attaching pipeline id to control name is bad, because it changes all the time (or can change). So it needs to be changed now to some static identifier, but that requires changes in UCM configs also.

juimonen, 2020-05-15:
I made now this:
https://github.com/thesofproject/sof/pull/2942

So Dmic0 name remains as it was and Dmic1 is indexed. I think we don't currently reference Dmic1 in upstream UCM's so UCM's should work with this....

perexg, 2020-05-15:
It's just a workaround. I would really prefer a fix. Some thoughts:

1) the SectionControlMixer controls are static, so the name should be unique and adding a next extra prefix in the driver seems like a big mistake (why not define directly 'Dmic0 Volume Control') ?

2) allow private instantions of the control mixer elements for DAPM widget (enhance the config syntax and improve the alsa-lib's topology code)

Liviali155, 2020-05-15:
Matins(TPLG: sof-hda-generic-2ch.tplg)
1.4.2 The controls named  "Dmic0 Capture Volume" and  "Dmic1 Capture Volume"
1.5   The controls named  "Dmic0  10 Capture Volume" and  "Dmic1 Capture Volume"
1.4.2 The peak can up to 99% when recorded using dmic (set volume as maximum)
1.5    The peak only 20%~30% when recorded using dmic (set volume as maximum)

TGL HDA RVP(TPLG: sof-hda-generic-4ch.tplg)
1.4.2 The controls named  "Dmic0 Capture Volume" and  "Dmic1 Capture Volume"
1.5   The controls named  "Dmic0  10 Capture Volume" and  "Dmic1 Capture Volume"
1.4.2 The peak can up to 60%~70% when recorded using dmic (set volume as maximum)
1.5    The peak less 10% when recorded using dmic (set volume as maximum)

After applied PR #2942
Matins(TPLG: sof-hda-generic-2ch.tplg) and TGL HDA RVP(TPLG: sof-hda-generic-4ch.tplg)
The controls named  "Dmic0 Capture Volume" and  "Dmic1 2nd Capture Volume"
The volume issue still can be reproduced

singalsu, 2020-05-15:
The change in capture sensitivity (meaning dBSPL -> dBFs, we really should use other calibrated metric than arecord-%) can be due to several issues:

- Device with 3 or more microphones receive to firmware only volume control for channels 1 and 2. Channels 3, 4, ... operate with default gain of 0 dB. I wonder what channels get extracted by Pulseaudio when e.g. 2ch capture is done?

- We may have had in 1.4.2 an EQ in the path that provided +20 dB boost. It has been used for direct-attach DMIC capture and in some cases for HDA/SDW DMIC/AMIC capture when reports of too silent capture received

- A while ago FW volume component did not support > 0 dB gain so the EQ +20 dB gain offset was used. Now it does but we have a problem that out-of-box gain is 0 dB and it's rather silent for typical 
-26 dBFS @ 94 dBSPL microphones (equals 5% @ 94 dBSPL, so 60 dBFS conversational speech is around 0.1%, though arecord % is likely not RMS metric but peak).

- The EQ +20 dB offset is not desired solution. The UCM should know to apply out of box for direct-attach DMIC the e.g. +20 dB gain to involved PGAs.

We have some X1s in our team. Need to test what gains get applied and what really happens.

juimonen, 2020-05-15:
@Liviali155 Do you see the dmic volume control having any effect with my patch and new firmware? So do you see increase/decrease in levels? And can you also check that alsamixer dmic0 setting is also moving? So we are wondering do we have some issues with volume setting for 4 channel dmic....

perexg, 2020-05-17:
Just note: arecord uses the really dumb sample level % (so for 16-bit, 1% = 65536 / 100 = a sample value between 655 and 1311). No dBxxx related conversions.

> Device with 3 or more microphones receive to firmware only volume control for channels 1 and 2. Channels 3, 4, ... operate with default gain of 0 dB.

Wow. Another inconsistency.

> I wonder what channels get extracted by Pulseaudio when e.g. 2ch capture is done?

I think that only first two channels are used, so the volume control problem is invisible.

Liviali155, 2020-05-18:
> @Liviali155 Do you see the dmic volume control having any effect with my patch and new firmware? So do you see increase/decrease in levels? And can you also check that alsamixer dmic0 setting is also moving? So we are wondering do we have some issues with volume setting for 4 channel dmic....

Yes,  on TGL HDA RVP dmic volume control has effect,I can see increase/decrease in levels,but the peak less 10% when changed the volume to maximum,and still can capture voice when changed volume to 0,see issue https://github.com/thesofproject/linux/issues/1799

singalsu, 2020-05-19:
@Liviali155 Can you please try SOF PR #2949. It re-introduces the +20 dB EQ gain that seems to be still needed.

Liviali155, 2020-05-20:
> @Liviali155 Can you please try SOF PR #2949. It re-introduces the +20 dB EQ gain that seems to be still needed.

Tried with PR  #2949 
TGL HDA RVP:
The issue controls name still can be reproduced as below
The controls named "Dmic0 10 Capture Volume" and "Dmic1 Capture Volume"

The peak can up to 60%~70% when recorded using dmic6 (set volume as maximum)
But the peak still less than 10% when recorded using dmic7(set volume as maximum)
[image]

Matins:
The issue controls name still can be reproduced as below
The controls named "Dmic0 10 Capture Volume" and "Dmic1 Capture Volume"

The peak can up to 99% when recorded using dmic6 (set volume as maximum)
But the peak still only 20%-30% when recorded using dmic7(set volume as maximum)
[image]

juimonen, 2020-05-20:
@Liviali155 please use together the 2 PR's from @singalsu and me:
#2942 and #2949. Together these should fix naming and volume.

Liviali155, 2020-05-20:
> @Liviali155 please use together the 2 PR's from @singalsu and me:
> #2942 and #2949. Together these should fix naming and volume.

After applied with #2949,volume still has issue when arecord via dmic(0,7)

juimonen, 2020-05-20:
@Liviali155 ok this is as it supposed to be. We left the 16khz pipeline volume intentionally without eq boost as the 16khz pipe is not referred in upstream UCM's currently. @singalsuo FYI.

singalsu, 2020-05-20:
@juimonen @Liviali155 @lgirdwood I realized afterwards that it would be possible to increase the loudness of 16 kHz PCM. There's a separate macro for it so it's possible without conflicts. The macros system that was introduced is not yet complete but this seems possible.  Maybe it should be done to avoid confusion. I'll do that change for next non-RFC version.

Thanks for testing this!

singalsu, 2020-05-20:
@Liviali155 There's now PR #2961. Can you please check that it gives similar capture level for 16 kHz DMIC as the just merged #2949 change for 48 kHz did.

Liviali155, 2020-05-21:
> @Liviali155 There's now PR #2961. Can you please check that it gives similar capture level for 16 kHz DMIC as the just merged #2949 change for 48 kHz did.

Tried with PR #2961, it actually gives similar capture level for 16kHz DMIC as the just merged #2949 change for 48 kHz did.

perexg, 2020-05-26:
Could you release 1.5.1 in sof-bin with this fix?

lgirdwood, 2020-05-27:
@lbetlej @fkaczmar @mengdonglin can we do a v1.5.1 next week ?

hsanjuan, 2020-05-31:
I was going to open an issue about the lack of signed binaries for v1.5.0, but seeing this I guess I'll wait. Would it be possible to get signed binaries with this v1.5.1? Sound does not work with v1.4.2 and the latest kernels.

ClarexZhou, 2020-06-11:
fw v1.5.1 is released, find binaries from https://github.com/thesofproject/sof-bin/tree/stable-v1.5.1

lgirdwood, 2020-06-11:
@perexg fyi

slawblauciak, 2020-06-16:
Fixes merged and released. Closing.

closed 2020-06-16 by slawblauciak