thesofproject/sof #2939: [BUG][APL] SRC component frequency conversion fails for certain combinations

zrombel, 2020-05-14:
**Describe the bug**
Bug is related to PR: #2898 and is introduced to master with commit: #aded79d. 
Before bug introduction every conversion between 8kHz, 16kHz, 24kHz and 48kHz worked properly.
Now SRC conversion matrix looks as follows:

In \ Out | 8 kHz | 16 kHz | 24 kHz | 48 kHz
-- | -- | -- | -- | --
8 kHz | PASS | FAIL | FAIL | FAIL
16 kHz | FAIL | PASS | FAIL | PASS
24 kHz | FAIL | FAIL | FAIL ? | FAIL
48 kHz | PASS | PASS | FAIL | PASS

 
**To Reproduce**
Run python tests: 11_03_TestSrcSamplingRateChange

**Reproduction Rate**
100%

**Environment**
1) SOF FW branch: master, commit: aded79d
3) Test scripts branch: master, commit 5b781595
3) Platform APL

**Logs**
[fail-src-48-24-dma_trace_1.log](https://github.com/thesofproject/sof/files/4627746/fail-src-48-24-dma_trace_1.log)
[pass-src-16-48-dma_trace_1.log](https://github.com/thesofproject/sof/files/4627747/pass-src-16-48-dma_trace_1.log)
[pass-src-48-16-dma_trace_1.log](https://github.com/thesofproject/sof/files/4627748/pass-src-48-16-dma_trace_1.log)
[fail-src-8-16-dma_trace_1.log](https://github.com/thesofproject/sof/files/4627749/fail-src-8-16-dma_trace_1.log)

singalsu, 2020-05-19:
@zrombel Hi, can I keep support for 11.025 kHz, 12 kHz, 22.05 kHz in/out removed? As addition to above table I'd keep 32 kHz supported. As a nice N * 8 kHz rate, the support doesn't need much RAM.

zrombel, 2020-05-19:
11.025 kHz, 12 kHz, 22.05 kHz can be kept removed, I don't think there is any production topology using this sampling rates, and we don't cover them in python tests.

dbaluta, 2020-05-19:
@zrombel @singalsu what about 44.1KHz is that supported? This is usually a rate requested by customers. 

For us will be good to have the multiple of  11.025KHz rates (with a stress on 44.1KHz).  I know
that SRC algorithms can be pretty nasty. In this case what is the drawback of having them supported?

Is the implementation or the fact that they use a lot of resources? At least keep them under some conditional defines so that we can experiment with it.

zrombel, 2020-05-19:
As far as i know 44.1KHz sampling rate is not supported at this point.

singalsu, 2020-05-19:
@zrombel Thanks for confirm! @dbaluta Yep, 44.1 <-> 48 remains supported. Also the full set 8 - 192 kHz remains available for xt-xcc build via Kconfig.

dbaluta, 2020-05-19:
@singalsu Thanks!

zrombel, 2020-05-25:
Fix in https://github.com/thesofproject/sof/pull/2953.

closed 2020-05-25 by zrombel