thesofproject/sof #2948: [TGL] pcm read/write error when loading 16bit, 8k, 6 channel topology

peileeli, 2020-05-19:
**Describe the bug**
I have loaded a 6channel, s16_le, 8k topology. However I faced the error below:
root@intel-corei7-64:~/aud# arecord -vD hw:0,0 -r16000 -c6 -fS16_LE output.wav -d1
Recording WAVE 'output.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Channels 6
Warning: rate is not accurate (requested = 16000Hz, got = 8000Hz)
         please, try the plug plugin
Hardware PCM card 0 'sof-nocodec' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 6
  rate         : 8000
  exact rate   : 8000 (8000/1)
  msbits       : 16
  buffer_size  : 4000
  period_size  : 1008
  period_time  : 126000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1008
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 4000
  silence_threshold: 0
  silence_size : 0
  boundary     : 9007199254740992000
  appl_ptr     : 0
  hw_ptr       : 0
arecord: pcm_read:2149: read error: Input/output error

**To Reproduce**
1. Setup TGL DUT with the platform config provided.
2. Boot in Yocto Image with Dummy codec enable.
3. Execute # init3
4. arecord -vD hw:0,0 -r16000 -c6 -fS16_LE output.wav -d1

**Reproduction Rate**
All the time

**Expected behavior**
read error: Input/output error should not happen

**Impact**
arecord not able to perform

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.4.17(lts-v5.4.17-yocto-200226T060457Z)
    * SOF: tgl-v0.3 
2) Name of the topology file
    * Topology: sof-tgl-nocodec-6ch-8k-s16.m4
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL

**Screenshots or console output**
arecord: pcm_read:2149: read error: Input/output error

peileeli, 2020-05-19:
[multichannel-issue.zip](https://github.com/thesofproject/sof/files/4649678/multichannel-issue.zip)

uploaded the dmesg and sof-logger

lgirdwood, 2020-05-19:
@peileeli your topology is configured to 8k audio only. If you want to set 16k or other rates then you can either use the SRC or set the rate to 16k.

peileeli, 2020-05-20:
i'm able to play without error now. Thanks for the help @lgirdwood

closed 2020-05-20 by peileeli