thesofproject/sof #2950: [BUG][EHL][S0ix.y]S0i2.0 fails to enter with SES0 binary of EHL BIOS

peileeli, 2020-05-19:
**Describe the bug**
S0i2.0 fails to enter with SES0 binary of EHL BIOS

**To Reproduce**
1. Flash with the BIOS and Yocto package
2. Make the necessary BIOS settings for S0ix.y :
Intel Advanced Menu → Debug Setting --> Kernel Debug Serial Port --> Debug Port Disabled
--
Intel Advanced Menu→ Debug Setting → Platform Debug Consent→ Manual
Intel Advanced Menu → Debug Setting → Advanced Debug Setting → USB DbC Enable Mode→ Disabled
Intel Advanced Menu→ PCH-IO Configuration → PCI Express Configuration --> PCIE clocks --> Clock4 assignment → Disabled
Intel Advanced Menu→ PCH-IO Configuration → PCI Express Configuration --> PCIE clocks --> ClkReq for Clock4 → Disabled
3. Run echo freeze > /sys/power/state

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
S0ix test on SES0 BIOS, PSE component validation for S0ix will fail due to this issue ==> show stopper

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 
       Tag: lts-v5.4.32-yocto-200419T223245Z
       Commit:a75d4a1c03e32e9c8088a5b77dae67df6af284b2
    * SOF: tgl SOFv0.3
2) Name of the topology file
    * Topology: 
[sof-ehl-rt5660.zip](https://github.com/thesofproject/sof/files/4649985/sof-ehl-rt5660.zip)

3) Name of the platform(s) on which the bug is observed.
    * Platform: EHL

**Screenshots or console output**

 [    8.549230] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
 [    8.553600] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
 [    8.557921] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
 [    8.596643] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
 [    8.599494] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
 [    8.732952] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:1-1bb51
 [    8.740632] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:12:0 Kernel ABI 3:11:0
 [    8.748445] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
 [    8.754577] sof-audio-pci 0000:00:1f.3: firmware boot complete
 [   12.102657] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
 [   12.200430] sof-audio-pci 0000:00:1f.3: firmware boot complete
 [   12.239853] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:10:0 Kernel ABI 3:11:0
 [   12.271255] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
 [   18.461029] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
 [   18.578213] sof-audio-pci 0000:00:1f.3: firmware boot complete
 [   39.239071] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
 [   39.538879] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
 [   39.542569] sof-audio-pci 0000:00:1f.3: error: Error code=0xffffffff: FW status=0xffffffff
 [   39.546415] sof-audio-pci 0000:00:1f.3: error: iteration 0 of Core En/ROM load failed: -110
 [   39.551942] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
 [   39.851827] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
 [   39.855517] sof-audio-pci 0000:00:1f.3: error: Error code=0xffffffff: FW status=0xffffffff
 [   39.856820] sof-audio-pci 0000:00:1f.3: error: iteration 1 of Core En/ROM load failed: -110
 [   39.862084] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
 [   40.162154] sof-audio-pci 0000:00:1f.3: error: status = 0x00000000 panic = 0x00000000
 [   40.165859] sof-audio-pci 0000:00:1f.3: error: Error code=0xffffffff: FW status=0xffffffff
 [   40.169704] sof-audio-pci 0000:00:1f.3: error: iteration 2 of Core En/ROM load failed: -110
 [   40.173611] sof-audio-pci 0000:00:1f.3: error: dsp init failed after 3 attempts with err: -110
 [   40.176902] sof-audio-pci 0000:00:1f.3: error: status = 0xffffffff panic = 0xffffffff
 [   40.178223] sof-audio-pci 0000:00:1f.3: error: failed to reset DSP
 [   40.179606] sof-audio-pci 0000:00:1f.3: error: failed to boot DSP firmware after resume -110
 [   40.686243] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60010000 size 108
 [   40.688066] sof-audio-pci 0000:00:1f.3: error: status = 0x001d003c panic = 0x00000000
 [   40.690277] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
 [   40.692608] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
 [   40.694387] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00040000 mask 0x00000000
 [   40.696456] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
 [   40.698333] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.701736] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.703015] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.704527] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.706291] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.709715] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.711263] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.712808] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.714449] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.717848] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.719258] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.720805] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.722407] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.725776] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.726962] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.728606] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.730378] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.734754] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.735996] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.737563] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.738842] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.742376] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.743987] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.745734] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.747011] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.750531] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.751715] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.753968] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.755401] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.758931] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.760280] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.761916] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.763723] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   40.767533] sof-audio-pci 0000:00:1f.3: error: stream reset failed
 [   40.768913] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
 [   40.770488] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
 [   40.771859] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
 [   76.313753] sof-audio-pci 0000:00:1f.3: error: reset enter failed: core_mask 1 adspcs 0x1d003c
 [   76.313760] sof-audio-pci 0000:00:1f.3: error: dsp core reset failed: core_mask 1
 [   76.313763] sof-audio-pci 0000:00:1f.3: error: failed to power down core during suspend
 [   76.313766] sof-audio-pci 0000:00:1f.3: error: suspending dsp
 [   76.313771] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -5

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

xiulipan, 2020-05-20:
Looks very like https://github.com/thesofproject/sof/issues/2889
Happen only on certain device with certain IFWI

slawblauciak, 2020-06-02:
Awaiting update until next week. If none provided, the issue will be closed.

slawblauciak, 2020-06-09:
Closing for now as no update received.

closed 2020-06-09 by slawblauciak