thesofproject/sof #2981: [BUG] Glitches on asrc capture stress test.

jdrzaszcz, 2020-05-27:
**Describe the bug**
When running simultaneous capture and playback on external loopback (same SSP) with ASRC component on capture pipeline, with playback iteration >= 75, there are small gliches on capturead wave. Therefore, the THD value in the wave verification is worse than expected.

**Example captered stream**

Spectogram:
[image]

Wave:
[image]

**Topology**
```
                +------------------+
    +------+    | +---+  +-------+ |
    | Host |----->|Buf|->|SSP Dai|------------------+
    +------+    | +---+  +-------+ |                |
                +------------------+                |
                                                    |
                +---------------------------------+ |
    +------+    | +---+  +----+  +---+  +-------+ | |
    | Host |<-----|Buf|<-|Asrc|<-|Buf|<-|SSP Dai|<--+
    +------+    | +---+  +----+  +---+  +-------+ |
                +---------------------------------+
```

**To Reproduce**
Ru slim driver python tests from file:
test_01_asrc_simple_capture.py
e.g. 
TestAsrcCapture16000Hz16b16b2ch
TestAsrcCapture48000Hz32b32b2ch
TestAsrcCapture48000Hz16b16b2ch

**Reproduction Rate**
30 %

**Expected behavior**
Captured wave shouldn't have glitches.

**Environment**
I check it on: WHL RVP and TGL RVP,
but as I know from CI tests, it also happens on APL and ICL

Commit: 90ddae9ebe6abbb331a1833a9eacb2d0505865f8 on SOF master branch

singalsu, 2020-05-27:
Interesting, thanks for reporting @jdrzaszcz ! When doing 75+ tests, how long were each play/capture tests?

jdrzaszcz, 2020-05-27:
@singalsu On each iteration we play 3 seconds stream.

singalsu, 2020-06-04:
I'm now doing runs of 100x SPDIF ASRC capture test that includes THD+N measurement. I'm seeing variation of about 1 dB but so far no discontinuities. I'll continue tomorrow with more tests.

singalsu, 2020-06-05:
I caught the first discontinuity, in my test the 3s clip THD+N dropped to -56 dB from usual -78 dB. The waveform looks like this, different looking than what you got. But it's definitely an error if it's not originated from SPDIF transmitter source.

[image]

singalsu, 2020-06-05:
I've now run more than 1k of repeats and above has happened only once. Since it looks different than reported and with some kind of impulse train convolved I suspect it was a glitch from SPDIF source (macbook).

I will switch to diagnostic driver test where this reported issue happens. There was earlier also an issue that never happened with Linux driver.

singalsu, 2020-06-18:
With a lot of help from @jdrzaszcz I have now replicated the issue with my setup and I have an initial fix done & under test. The pull mode filter core has an overrun condition protection that can happen but the SOF buffer update is not aware of the exception and thinks requested amount was processed.

singalsu, 2020-06-30:
Update - with my test setup I seem to be able to fix the capture glitch issue but there's also for playback a stress test glitch seen for 8 kHz sample rate. It happens only with skew tracking enabled and I can made it disappear almost totally by making the loop filter unpractically slow. The jump in consumed samples originates from HW timestamping register. The samples count delta jumps while time delta remains stable.

So, I have not been able to properly fix it yet. I'll update now the PR with the work for capture, maybe it could be merged before my vacation. If yes, I could create a new issue for playback. As waveform it looks different and more severe with a rather big discontinuity.

singalsu, 2020-08-05:
There was a regression in PR #3090. While addressing the consume/produce deviation from skew there was a mistake in capture pull mode copy limit for source buffer side. It caused the copy() to process less samples than required. The new PR is #3275.

slawblauciak, 2020-08-10:
@jdrzaszcz the aforementioned PR has been already merged, can you retest and confirm if it's effective?

jdrzaszcz, 2020-08-11:
@slawblauciak I will retest it with current master and let you know.

jdrzaszcz, 2020-08-12:
@singalsu @singalsu Hi, I rerun tests with asrc on capture pipeline and I find out only few results lower than 80 THD value (less than 1 %). But on APL I got few times stream stall on playback pipeline.

lgirdwood, 2020-08-18:
@jdrzaszcz ok so are we saying a few glitches on APL only but other CAVS1.8+ platforms are OK ?

jdrzaszcz, 2020-08-18:
@lgirdwood Not exactly.  On WHL, TGL and ICL the results on capture pipeline are fine. I found out only one fail on WHL and one on TGL (but the results on TGL were a little less than 80). I think for this platforms everything is quite good.
On APL RVP i didn't saw glitches, but I found out another problem there (quite often i got stream stall on play stream - about 25 %).

jdrzaszcz, 2020-09-02:
@singalsu @slawblauciak I rerun tests with SOF master, commit: 32d32b59cb48350f8b159fd63f9f1f74741a7a0e. I run 75 playback iteration for 12 formats (3 sample rate conversion in each format except 8 kHz)
Formats:
[image]

WHL RVP - ones value lower than 80 (~65),
TGL RVP - five value lower than 80 (~ 75),
ICL RVP - two lower than 80(~75);
This means that the reproduction rate is really small, less than 1 %.
Result looks similar for both external and internal loopback.

For APL RVP I got only two times THD value lower than 80 (~60), but I still in about 25 % cases i got stream stall on play stream error.

singalsu, 2020-09-02:
Thanks @jdrzaszcz I will continue debugging this. The -75 dB THD+N value is about the best quality a C cassette player can ever get so this issue is hard to notice subjectively. I wonder how realistic the test setup where this issue happens is. It would be better to have another real I2S transmitter in the setup that our I2S receiver follows. If the jitter of test setup loop send and receive would accumulate it might put ASRC to worse jitter than normally happens. I'm wondering if using the internal loop force or normal loop is more realistic.

lgirdwood, 2021-01-12:
@singalsu ping - any update ?

lgirdwood, 2021-01-22:
@singalsu ping - maybe v1.8 ?

singalsu, 2021-01-22:
@lgirdwood Yep, unfortunately v1.8. This was confirmed to happen by @jdrzaszcz with two separate I2S interfaces without LBM quirk. So it should be a real issue. I think I should create a glitch detector component, tuned to detect tiny glitches in sine wave (now only a patch for ASRC that didn't trigger) and move it around in the pipeline topology to see where it happens.

lgirdwood, 2021-05-10:
@singalsu any update ?

kkarask, 2021-05-31:
I run 19_00_TestAsrcPlayback8000Hz32b32b2ch with playback_iterations=100 and there are glitches in wav files. Reproduction rate 5/100.

Logs:
[2021_05_31_13_22_13.zip](https://github.com/thesofproject/sof/files/6570292/2021_05_31_13_22_13.zip)

lgirdwood, 2021-06-03:
@kkarask I'm going to close this for the moment as it's not seen with the Linux driver (and wont be seen by users). Lets keep it on record.

closed 2021-06-03 by lgirdwood