thesofproject/sof #2986: [BUG] Too short volume initial ramp in format 8kHz 16b/16b 1ch

jdrzaszcz, 2020-05-28:
**Describe the bug**
When running simultaneous capture and playback on external loopback (same SSP) with Volume component on playback pipeline, inital ramp length is not equal to given initial ramp parameter.
It happens with
SofVolumeRamp = SOF_VOLUME_LINEAR, 
SofVolumeRamp = SOF_VOLUME_LINEAR_ZC, and only for format 8kHz 16b/16b 1ch

Example for SOF_VOLUME_LINEAR_ZC:
INITIAL_RAMP = 40
Expected 312.0 samples in initial ramp, but got: 214
Ramp time 0.02675 is not equal expected ramp time = 0.039 on channel 1
Format 8kHz 16b/16b 1ch: 
[image]
For format 8kHz 16b/16b 2ch looks ramp length is fine:
[image]

Example for SOF_VOLUME_LINEAR:
INITIAL_RAMP = 250
Expected 1992.0 samples in initial ramp, but got: 1332
Ramp time 0.1665 is not equal expected ramp time = 0.249 on channel 1
Format 8kHz 16b/16b 1ch: 
[image]
For format 8kHz 16b/16b 2ch looks ramp length is fine:
[image]

**Topology**
```
+------------------------------------------------------+
| +------+   +-----+   +-- --+   +-----+   +---------+ |
| | Host |-->| Buf |-->| Vol |-->| Buf |-->| SSP Dai |---+
| +------+   +-----+   +-----+   +-----+   +---------+ | |
+------------------------------------------------------+ |
                                                         |
+-----------------------------------+                    |
| +------+   +-----+   +---------+  |                    |
| | Host |<--| Buf |<--| SSP Dai |<----------------------+
| +------+   +-----+   +---------+  |
+-----------------------------------+
```

**To Reproduce**
Run slim driver python tests:
- for SOF_VOLUME_LINEAR_ZC: TestVolumeInitialRampSimple8000Hz16b16b1ch,
from file test_05_inital_ramp_parametr_simple_test.py
- for SOF_VOLUME_LINEAR: TestVolumeLinearRamp8000Hz16b16b1ch 
from file test_06_volume_linear_ramp.py

**Reproduction Rate**
100 %

**Expected behavior**
Ramp length should be equal to given initial_ramp parameter

**Environment**
I checked it with TGL RVP and WHL RVP platforms.

Commit: 46897802a4549a77e26949ef7bbf54b4c480caafon SOF master branch

singalsu, 2020-05-28:
@jdrzaszcz Thanks for through testing! The ZC mode inaccuracy is in my list but I wasn't aware of linear mode inaccuracy. Can you still add what volume transitions (as dB, or Q16.16 linear gain) did you apply in above examples?

jdrzaszcz, 2020-05-28:
@singalsu The gain was from VOL_MIN = 0 to VOL_MAX = (1 << 16)

singalsu, 2020-05-28:
Yep, for that transition the lengths should be as you expected. I think I can create next week a PR to fix these.

singalsu, 2020-06-02:
I was running in testbench volume with 1ch (SOF_VOLUME_LINEAR, INITIAL_RAMP = 250) and got correct ramp length. I'll check next SOF_VOLUME_LINEAR_ZC on testbench. Also I'll try to set up a real device for 1ch.

singalsu, 2020-06-02:
@jdrzaszcz I'm not getting wrong ramp length in testbench with SOF_VOLUME_LINEAR. Can you add this patch to your setup and see if the stream parameters seen by volume are correct.

```
diff --git a/src/audio/volume/volume.c b/src/audio/volume/volume.c
index 9e72835f1a1d..cfd5e8a54657 100644
--- a/src/audio/volume/volume.c
+++ b/src/audio/volume/volume.c
@@ -830,6 +830,8 @@ static int volume_prepare(struct comp_dev *dev)
        cd->ramp_finished = true;
        cd->vol_ramp_frames = dev->frames / (dev->period / VOL_RAMP_UPDATE_US);
        cd->channels = sinkb->stream.channels;
+       comp_info(dev, "volume_prepare(): channels = %d, vol_ramp_frames = %d",
+                 cd->channels, cd->vol_ramp_frames);
        for (i = 0; i < cd->channels; i++) {
                cd->volume[i] = cd->vol_min;
                volume_set_chan(dev, i, cd->tvolume[i], false);

```
For 1 ch 8 kHz it should print 1 and 8.

jdrzaszcz, 2020-06-02:
@singalsu I checked and it looks like volume see correct parameters.
volume_prepare(): channels = 1, vol_ramp_frames = 8
[dma_trace_1.log](https://github.com/thesofproject/sof/files/4717132/dma_trace_1.log)

singalsu, 2020-06-02:
@jdrzaszcz If the test tone frequency is e.g. 1 kHz, is the output of the test in Audacity also 1 kHz? Wrong sample rate in wav produced by the test system could also cause this. 

But I'm definitely getting the issue with ZC mode and I'm doing a fix for it now.

singalsu, 2020-06-02:
@jdrzaszcz I made PR #3013 . Can you please test if it fixes SOF_VOLUME_LINEAR_ZC mode issues. I have not been able to replicate SOF_VOLUME_LINEAR (non-ZC) issues yet, not as large deviation as you got.

jdrzaszcz, 2020-06-03:
@singalsu Frequency in Audacity looks fine for 8 kHz  sample rate.
I  have run the volume inital_rampt tests with PR #3013 and it works for both SOF_VOLUME_LINEAR and default SOF_VOLUME_LINEAR_ZC  ramp mode.

singalsu, 2020-06-03:
@jdrzaszcz Excellent news, thank you! @mengdonglin Could someone in PRC do for PR #3013 the volume component related checks to make sure it doesn't cause regressions.

I will also test today it here more before removing [RFC]. Volume is such complicated creature for controls so it needs a of attention when fundamental changes are done.