thesofproject/sof #2991: [BUG][TGL][SDW]"ipc error for 0x60010000 size 20" occured when captured via arecord0,4(dmic).

sinahuang, 2020-05-28:
**Describe the bug**
" Unable to install hw params:" display when captured via arecord0,4(dmic) and dmesg shows "ipc error for 0x60010000 size 20" .

**To Reproduce** 
1. Boot up system..
2.  Arecord -Dhw:0,4 -r16000 -c4 -fs16_le 00.wav -vv -i

**Reproduced rate**
100%
 
**Expected result**
Arecord should work successfully.

**Actual result**
Dmesg shows "error: ipc error for 0x60010000 size 20" and arecord cannot work.

```
ubuntu@ubuntu-Tiger-Lake-Client-Platform:~$ arecord -Dhw:0,4 -r48000 -c4 -fs16_le 33.wav -vv -i
Recording WAVE '33.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Channels 4
Warning: rate is not accurate (requested = 48000Hz, got = 16000Hz)
         please, try the plug plugin
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 64
CHANNELS: 4
RATE: 16000
PERIOD_TIME: 124500
PERIOD_SIZE: 1992
PERIOD_BYTES: 15936
PERIODS: (4 5)
BUFFER_TIME: 499500
BUFFER_SIZE: 7992
BUFFER_BYTES: 63936
TICK_TIME: 0
```
dmesg log:
```
[   67.149439] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[   67.149440] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[   67.149653] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 4 dir 1
[   67.149655] sof-audio-pci 0000:00:1f.3: generating page table for 00000000ab0c7676 size 0xf9c0 pages 16
[   67.149661] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   67.149677] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   67.149678] sof-audio-pci 0000:00:1f.3: period_bytes:0x3e40
[   67.149679] sof-audio-pci 0000:00:1f.3: periods:4
[   67.149690] sof-audio-pci 0000:00:1f.3: stream_tag 2
[   67.149696] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   67.149861] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   67.149865] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[   67.149866] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[   67.149868]  DMIC16kHz: ASoC: hw_params FE failed -22
[   67.149870] sof-audio-pci 0000:00:1f.3: pcm: free stream 4 dir 1
[   67.149995] sof-audio-pci 0000:00:1f.3: pcm: close stream 4 dir 1
[   67.646722] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   67.646757] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   68.146669] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
```
**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev  commit: 0d22f66
FW: https://github.com/thesofproject/sof/commits/master commit: 46897802
Tplg: Same with FW branch, sof-tgl-rt711-rt1308.tplg 
Platform:TGL-U RVP with codec ALC711 and RT1308 in SDW mode.

[dmesg.log](https://github.com/thesofproject/sof/files/4694696/dmesg.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/4694697/sof_trace.log)

sinahuang, 2020-05-28:
Might be related to issue  https://github.com/thesofproject/sof/issues/2445

plbossart, 2020-05-28:
@sinahuang you are trying to record 48kHz on the 16kHz link, this cannot possibly work.

sinahuang, 2020-05-29:
Error still can be seen when arecord with 16kHz link.
```
ubuntu@ubuntu-Tiger-Lake-Client-Platform:~$ arecord -Dhw:0,4 -r16000 -c4 -fs16_le 33.wav -vv -i
Recording WAVE '33.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Channels 4
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 64
CHANNELS: 4
RATE: 16000
PERIOD_TIME: 124500
PERIOD_SIZE: 1992
PERIOD_BYTES: 15936
PERIODS: (4 5)
BUFFER_TIME: 499500
BUFFER_SIZE: 7992
BUFFER_BYTES: 63936
TICK_TIME: 0
```

sinahuang, 2020-05-29:
Issue can be fixed by PR https://github.com/thesofproject/sof/pull/2994.

slawblauciak, 2020-06-02:
PR has been merged. Closing.

closed 2020-06-02 by slawblauciak