thesofproject/sof #3009: [BUG] [TGL] ipc error when tirgger wov pipline with s24_le and s32_le

ClarexZhou, 2020-06-02:
**Describe the bug**
Unable to install hw params and "ipc error for 0x60010000 size 20" is dmesg when tirgger wov pipline with s24_le and s32_le

**To Reproduce**
arecord -Dhw:0,100 -M -N -f s24_le -r 16000 -i -vvv -c2 --period-size=68000 ./tmp.wav --dump-hw-params

**Reproduction Rate**
3/3

**Expected behavior**
wov pipline should be triggered successfully

**Impact**
wov trigger failed with ipc error.

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: https://github.com/thesofproject/linux/commits/topic/sof-dev commit:5cb79ee6f
Revert regression on USB-C 6a7c533d4a1854f54901a065d8c672e890400d8a
Revert regression on PINCTRL cd0a32371db73d0b50536a7ca4f036abddff0d1d
+ https://github.com/thesofproject/linux/pull/2062+ hack commit https://github.com/thesofproject/linux/commit/6769df9d4d48cfae008a8192aa5b723bc8edb804
    * SOF: https://github.com/thesofproject/sof/tree/tgl-006-drop-stable commit: f45bf945962813e6e
2) Name of the topology file
    * Topology: sof-tgl-max98373-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL chrome volteer with 98373 in I2S mode

**Screenshots or console output**
```

ubuntu@TGL-Volteer-01:~$ arecord -Dhw:0,100 -M -N -f s24_le -r 16000 -i -vvv -c2 --period-size=68000 ./tmp.wav --dump-hw-params
Recording WAVE './tmp.wav' : Signed 24 bit Little Endian, Rate 16000 Hz, Stereo
HW Params of device "hw:0,100":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: 16000
PERIOD_TIME: [1500 3999000]
PERIOD_SIZE: [24 63984]
PERIOD_BYTES: [192 255936]
PERIODS: [2 160]
BUFFER_TIME: [3000 19998000]
BUFFER_SIZE: [48 319968]
BUFFER_BYTES: [192 1279872]
TICK_TIME: ALL
--------------------
arecord: set_params:1363: Unable to install hw params:
ACCESS:  MMAP_INTERLEAVED
FORMAT:  S24_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 1999500
PERIOD_SIZE: 31992
PERIOD_BYTES: 255936
PERIODS: 2
BUFFER_TIME: 3999000
BUFFER_SIZE: 63984
BUFFER_BYTES: 511872
TICK_TIME: 0
```
no error sof logger trace.
Dmesg
```
[  155.916096] sof-audio-pci 0000:00:1f.3: pcm: open stream 100 dir 1
[  155.916098] sof-audio-pci 0000:00:1f.3: period min 192 max 256000 bytes
[  155.916098] sof-audio-pci 0000:00:1f.3: period count 2 max 160
[  155.916099] sof-audio-pci 0000:00:1f.3: buffer max 1280000 bytes
[  155.916238] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 100 dir 1
[  155.916240] sof-audio-pci 0000:00:1f.3: generating page table for 0000000075d0612b size 0x7cf80 pages 125
[  155.916246] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  155.916261] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  155.916279] sof-audio-pci 0000:00:1f.3: period_bytes:0x3e7c0
[  155.916282] sof-audio-pci 0000:00:1f.3: periods:2
[  155.916294] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  155.916301] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  155.916598] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  155.916601] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[  155.916603] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[  155.916605]  DMIC16k: ASoC: hw_params FE failed -22
[  155.916608] sof-audio-pci 0000:00:1f.3: pcm: free stream 100 dir 1
[  155.916729] sof-audio-pci 0000:00:1f.3: pcm: close stream 100 dir 1
[  156.815606] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  156.815638] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```
[dmesg_wov_006.txt](https://github.com/thesofproject/sof/files/4715717/dmesg_wov_006.txt)
[full_sof_wov_006.txt](https://github.com/thesofproject/sof/files/4715718/full_sof_wov_006.txt)

ClarexZhou, 2020-06-02:
@keyonjie  please help to check first.
Error not seen with s16_le format.

keyonjie, 2020-06-03:
Hi @ClarexZhou , we should specify buffer-size, not period-size.
see it here: https://github.com/thesofproject/sof-docs/blob/master/developer_guides/linux_driver/third_party/keyword_detect/keyword_detect.rst#run-keyword-detection-pipeline

arecord -Dhw:0,100 -M -N -f s24_le -r 16000 -i -vvv -c2 --buffer-size=68000 ./tmp.wav

ClarexZhou, 2020-06-03:
yes. buffer size is working. Close issue

closed 2020-06-03 by ClarexZhou