thesofproject/sof #3017: [TGL] Need PDM decimation coefficients with 0dB

sathya-nujella, 2020-06-03:
Need:
Currently DMIC PDM decimation coefficients are generated with -0.4dB (likely linear gain of 0.95).
Customer request decimation coefficients with 0dB. 

Reason:
"Having 0dB decimation" means that user space/CRAS would like to get DMIC data without any gain applied on it.
Example usage: record a concert (with a peak SPL of close to 120dB), any static gain adjustment to the microphone means direct clipping.

sathya-nujella, 2020-06-03:
Hi @singalsu ,
Sorry for late in getting back on this request. Customer confirmed the need.

plbossart, 2020-06-03:
> Need:
> Currently DMIC PDM decimation coefficients are generated with -0.4dB (likely linear gain of 0.95).
> Customer request decimation coefficients with 0dB.
>
> Reason:
> "Having 0dB decimation" means [...]

What would a 0.4dB attenuation do here? I just don't get the concern.

plbossart, 2020-06-03:
Then I don't understand what the ask was. Whether we re-add 0.4dB in the coefficients or it's done by CRAS, some clipping will happen...

cujomalainey, 2020-06-03:
@sathya-nujella Jimmy is @chiang831

chiang831, 2020-06-04:
I have discussed with Leo and confirmed  that previously what we have concern is the 9dB gain. Having 0.4 dB attenuation is fine. You don't need to remove it from legacy code.
Thanks!

singalsu, 2020-06-04:
@chiang831 @cujomalainey Thanks for update! But it ever need the filters characteristic in PDM bits decimation HW is quite well programmable. The amplification that you have seen is due to specially made DMIC configuration to optionally boost the signal in previous firmware. While in SOF the DMIC path has been always this near 0 dB. The gain is better to apply in FW components or CRAS for usual VoIP etc. capture sensitivity.

sathya-nujella, 2020-06-04:
Thank you all for inputs. Based on discussion here, I am closing this one.

closed 2020-06-04 by sathya-nujella