thesofproject/sof #3035: [BUG][v1.5.1]"ERROR eq_iir_init_coef(), ch 0 is set to bypass" shows during the check-audio-equalizer test

YvonneYang2, 2020-06-09:
**Describe the bug**
"ERROR eq_iir_init_coef(), ch 0 is set to bypass" shows during the check-audio-equalizer test. Issue caused by commit: 88707ca87e782631. 

**To Reproduce**
1. Boot up system with topology has a EQ component in it
2. cd sof-test and run " export TPLG=/lib/firmware/intel/sof-tplg/sof-cml-rt1011-rt5682.tplg"
3. cd test-case and run ./check-audio-equalizer.sh
https://github.com/thesofproject/sof-test/blob/master/test-case/check-audio-equalizer.sh

**Reproduction Rate**
5/5

**Expected behavior**
no error occurred.

**Impact**
Dmesg shows “error: ipc error for 0x60010000 size 20”
Sof-trace shows "ERROR eq_iir_init_coef(), ch 0 is set to bypass"

**Environment**
Kernel :topic/sof-dev (1af84d0e)
fw:  https://github.com/thesofproject/sof/commits/v1.5.1, commit 88707ca8
tplg: same as fw sof-cml-eq-rt1011-rt5682.tplg rename to sof-cml-rt1011-rt5682.tplg
platform: cml chromebook with i2s rt5682 

**error sof-trace**
```
         TIMESTAMP              DELTA C# COMPONENT          LOCATION                       CONTENT
[193805865.885417] (193805872.000000) c0 eq-iir             ../audio/eq_iir/eq_iir.c:419   ERROR eq_iir_init_coef(), ch 0 is set to bypass
[193805873.125000] (        7.239583) c0 eq-iir             ../audio/eq_iir/eq_iir.c:419   ERROR eq_iir_init_coef(), ch 1 is set to bypass
```

**dmesg** 
```
[  108.433805] sof-audio-pci 0000:00:1f.3: periods:4
[  108.433816] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  108.433823] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  108.434151] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  108.434155] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[  108.434156] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[  108.434159]  Speakers: ASoC: hw_params FE failed -22
[  108.434161]  SSP1-Codec: ASoC: hw_free BE SSP1-Codec
[  108.434166]  Speakers: ASoC: hw_free FE Speakers
```

[eror_logger.log](https://github.com/thesofproject/sof/files/4750269/eror_logger.log)
[logger.log](https://github.com/thesofproject/sof/files/4750270/logger.log)
[dmesg_eq.log](https://github.com/thesofproject/sof/files/4750271/dmesg_eq.log)

abonislawski, 2020-06-09:
Are you sure about 88707ca?
There is no trace from this commit
trace_ipc_error("ipc: invalid size 0x%x", hdr->size);

YvonneYang2, 2020-06-09:
Tested it with fw compile via xcc, issue cannot be reproduced.
Close it for it's gcc specific issue.

closed 2020-06-09 by YvonneYang2

singalsu, 2020-06-09:
The code should work as gcc build too, despite lower optimization level, I'll still see what happens. Even if the process would require to close this I'll still keep this as action for myself to study what happens.

singalsu, 2020-06-09:
I'm not able to repeat this same error on a CML device with gcc build. The fail with too long FIR filter is normal and it's known.

```
[327276916.614583] (        5.208333) c0 eq-fir             ../audio/eq_fir/eq_fir.c:334  ERROR Filter length 0 exceeds limitation for build.
[327276919.947917] (        3.333333) c0 eq-fir       7.37  ../audio/eq_fir/eq_fir.c:774  ERROR eq_fir_prepare(): eq_fir_setup failed.
[327276924.270833] (        4.322917) c0 pipe                   src/audio/pipeline.c:598  ERROR pipeline_prepare(): ret = -22, dev->comp.id = 34
[327276927.968750] (        3.697917) c0 ipc                       src/ipc/handler.c:312  ERROR ipc: pipe 7 comp 34 prepare failed -22

```
Also re-confirm that with xt-xcc build the above check for filter length (to prevent larger fail of SOF)  does not trigger and the test case passes, no errors in trace. OK to close in my opinion too.

singalsu, 2020-06-09:
Actually there's a minor bug in above "Filter length 0", the printed value (not set yet) is not the same as tested value. I'll make a small PR to fix the error message.

(Thanks for well prepared error report @YvonneYang2 )