thesofproject/sof #3047: [BUG][ICL]

fredoh9, 2020-06-12:
Moved the issue from linux project, https://github.com/thesofproject/linux/issues/2193

In recentl PR tests, ICL KMOD tests are failed. This is found only on ICL_RVP_HDA, but very consistent.
From daily regression report,

- linux hash: 2839e03e, sof hash: b5043671==> no issue
- linux hash: c5e14df5, sof hash: 700ca2ae ==> issue found for the first time

Module loading /unloading are working fine. But dsp power status are always active after module reloading.
```
2020-06-12 16:47:15 UTC [REMOTE_INFO] ===== Starting iteration 1 of 2 =====
2020-06-12 16:47:15 UTC [REMOTE_INFO] wait dsp power status to become suspended
2020-06-12 16:47:32 UTC [REMOTE_INFO] dsp is not suspended after 15s, end test
2020-06-12 16:47:32 UTC [REMOTE_INFO] Test Result: FAIL!
```

KMOD test alone worked fine.

Looks volume test causing DSP status to active.

1. check-runtime-pm-status.sh ==> ==> PASS
2. volume-basic-test.sh ==> PASS
3. check-runtime-pm-status.sh ==> Failed

```
[  105.950102] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[  106.453635] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[  106.453642] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  106.453665] sof-audio-pci 0000:00:1f.3: unknown ROM status value ffffffff
[  106.453751] sof-audio-pci 0000:00:1f.3: invalid header size 0xffffffff. FW oops is bogus
[  106.453755] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0xffffffff trace 0xffffffff
[  106.453786] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  106.453788] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  106.453814] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
```

[dmesg.log](https://github.com/thesofproject/linux/files/4772491/dmesg.log)
[sof-logger.log](https://github.com/thesofproject/linux/files/4772490/sof-logger.log)

- linux hash: c5e14df5, sof hash: b5043671 ==> worked fine.

After reverting below commit, found working fine. @singalsu can you take a look at? not sure why fails with ICL_RVP_HDA.

17f281de9 Audio: Improve volume gain ramp length accuracy

slawblauciak, 2020-06-16:
Are you sure that the bisect is 100% accurate? What's the reproduction rate here?

fredoh9, 2020-06-17:
> Are you sure that the bisect is 100% accurate? What's the reproduction rate here?

it's 100% reproducible. after revert the commit it worked fine.

aiChaoSONG, 2020-06-19:
@fredoh9 I run kmod scripts on one ICL HDA device, issue not reproduced, seem it is fix, but no idea who and what PR fix this issue. Can you help to check?

No need to close linux issue and open a new sof issue, you can direct transfer issue from one repo to another.   try to search "Transfer issue" on this web page, you should know how to transfer.

fredoh9, 2020-06-19:
@aiChaoSONG I can check this again tomorrow.
Good to know we can transfer a github issue from different repo. I will try that in next time.

fredoh9, 2020-06-19:
Indeed, I quickly checked again a ICL_RVP_HDA with today's build. it is working fine.

slawblauciak, 2020-06-19:
Good to close then?

plbossart, 2020-06-19:
> Indeed, I quickly checked again a ICL_RVP_HDA with today's build, it is working fine.

can we re-enable the CI tests @fredoh9 and close after CI tests report success?

fredoh9, 2020-06-19:
Sure, I just submitted the patch to enable KMOD tests on ICL. I will closely monitor the result after merging it.

fredoh9, 2020-06-19:
BTW, I did git bisect also.
After picking this, a2d1e1efdc554becec70a6c1d95491bac59800ef, it worked fine.

fredoh9, 2020-06-22:
confirmed, working fine with latest build.
https://sof-ci.01.org/linuxpr/PR2220/build3978/devicetest/

closed 2020-06-22 by fredoh9