thesofproject/sof #3049: [BUG]

ZaheenJ, 2020-06-14:
**Describe the bug**
The mic only produces static/white noise. It was a certain update that caused the mic to stop working properly, but I can't quite remember what was updated. I remember the kernel, and some alsa and topology packages being updated. I tried to downgrade the kernel, but that didn't fix it.

**To Reproduce**
I'm not sure if it is the kernel upgrade to 5.7 or some alsa component upgrades.

**Reproduction Rate**
This always happens on boot.

**Expected behavior**
Mic to work properly

**Impact**
Annoyance, I have to use windows or my phone for calls.

**Environment**
OS: Arch
Kernel: [5.7.2](https://www.archlinux.org/packages/core/x86_64/linux/)
SOF: [1.5.1](https://www.archlinux.org/packages/extra/any/sof-firmware/)
**dmesg | grep sof**
```
[    0.507191] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    0.507196] software IO TLB: mapped [mem 0x6df9c000-0x71f9c000] (64MB)
[    3.159570] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.160445] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.160455] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.160627] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.160740] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.167251] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.238891] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.238893] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.351961] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:1-88707
[    3.351963] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:13:0
[    3.351964] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.352309] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[    3.352330] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[    3.361383] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:16:0 Kernel ABI 3:13:0
[    3.361384] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    3.361400] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.362306] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.362307] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.363531] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.363534] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.364435] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.364437] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.365597] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    3.365598] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    3.365599] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    3.365600] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    3.365601] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    3.365602] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    3.366545] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    3.366548] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    3.368319] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    3.368320] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    3.369215] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    3.377851] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    3.665277] input: sof-hda-dsp Headphone Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input23
[    3.665325] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input24
[    3.665376] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input25
[    3.665422] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input26
```
**journalctl -b | grep sof**
Same as dmesg but with additional line:
`Jun 14 10:36:43 bigblue alsactl[680]: Found hardware: "sof-hda-dsp" "Realtek ALC3254" "HDA:8086280b,80860101,00100000 HDA:10ec0295,10280949,00100002 cfg-dmics:2" "" ""
`
**arecord -l**
```
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
It looks correct to me, and I think it looked like this when the mic was working. Does this mean it is a pulseaudio problem?
Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

singalsu, 2020-06-15:
Which of the capture devices are impacted. 0, 6, or 7?

Do you have a headphone (3 pole plug) or headset with microphone (4 pole plug) or some line-out cable connected to 3.5 mm plug. And what device is this?

ZaheenJ, 2020-06-15:
I am not sure what you mean by 0, 6, or 7, I am sorry. I only have one mic in this scenario, and it is my laptop's built in mic. I have [speakers](https://www.cnet.com/products/altec-acs-44-speaker-s/) with no mic connected via a 3.5 mm cable. Before my mic stopped working, I always had two mic options which were headset microphone and the build in one, even though my speakers didn't have a microphone. 
Here is an output of a fully working audio with mic, I am not sure if this is helpful:
```
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: use msi interrupt mode
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:12:0
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: firmware boot complete
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:12:0
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
Apr 28 14:29:38 bigblue kernel: sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC3254: line_outs=1 (0x14/0x0/0x0/0x0/0x0) type:speaker
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0:    inputs:
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0:      Headphone Mic=0x1b
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0:      Headset Mic=0x19
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
Apr 28 14:29:38 bigblue kernel: snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
Apr 28 14:30:02 bigblue plasma_session[2529]: org.kde.plasma.session: Starting autostart service  "/etc/xdg/autostart/pulseaudio.desktop" ("/usr/bin/start-pulseaudio-x11")
Apr 28 14:30:02 bigblue dbus-daemon[1117]: [system] Activating via systemd: service name='org.freedesktop.RealtimeKit1' unit='rtkit-daemon.service' requested by ':1.63' (uid=1000 pid=2601 comm="/usr/bin/pulseaudio --daemonize=no ")
[... cut ...]
Apr 28 14:30:02 bigblue pulseaudio[2601]: W: [pulseaudio] alsa-ucm.c: UCM file does not specify 'PlaybackChannels' or 'CaptureChannels'for device Speaker, assuming stereo duplex.
Apr 28 14:30:02 bigblue pulseaudio[2601]: W: [pulseaudio] alsa-ucm.c: UCM file does not specify 'PlaybackChannels' or 'CaptureChannels'for device Headphones, assuming stereo duplex.
Apr 28 14:30:02 bigblue pulseaudio[2601]: E: [pulseaudio] main.c: Source echoCancel_source does not exist.
Apr 28 14:30:02 bigblue pulseaudio[2601]: E: [pulseaudio] main.c: Sink echoCancel_sink does not exist.
Apr 28 14:30:12 bigblue systemd[1799]: pulseaudio.socket: Succeeded.
```

slawblauciak, 2020-06-16:
What device is this? Would be useful to have full dmesg and sof-logger traces supplied.

lgirdwood, 2020-06-16:
> I am not sure what you mean by 0, 6, or 7,

@ZaheenJ these are the PCM device numbers from your ```arecord -l```

You can capture from thiose uisng th e```arecord -Dhw:0,x``` where x is the PCM number.

singalsu, 2020-06-17:
@ZaheenJ Can you try Ubuntu 20.04 live image (try without installing option) and from Settings -> Sound -> Input device. Try the options there like "Internal Microphone - Built-in Audio". The level meter  there should follow your speech level if there's sufficient Volume level set from the widget below.

If you are willing to debug the low level audios the arecord command can be used for doing test recordings on the Linux distribution install where you see the problem. To avoid pulseaudio to impact recording you can disable it temporarily by

`sudo mv /usr/bin/pulseaudio /usr/bin/pulseaudio.disabled
`
Reboot. If you don't belong to group audio you need to run the commands with sudo.

```
sudo arecord -l # Check the device names that they match the next commands
sudo arecord -Dhw:0,0 -f dat -d 5 rechw00.wav
sudo arecord -Dhw:0,6 -f dat -d 5 rechw06.wav
sudo arecord -Dhw:0,7 -f dat -d 5 rechw07.wav
sudo mv /usr/bin/pulseaudio.disabled /usr/bin/pulseaudio

```
Make some recognizable noise while recording the 5s clips. Then reboot to get pulseaudio back normally. You can check with e.g. Audacity if the recordings made contained audio from the microphone that you wanted to use.

ZaheenJ, 2020-06-18:
I am sorry for not responding here, but I have good news. I don't think it was a SOF problem because an update to [alsa-ucm-conf](https://www.archlinux.org/packages/extra/any/alsa-ucm-conf/) fixed everything. I also had another unrelated problem before all of these updates, everything worked but sometimes I had a problem where my speakers and mic fail and the system hanged. Now that doesn't happen anymore either. I think that was because I updated to SOF 1.5.1, I use the aur package. Sorry for this bug

closed 2020-06-18 by ZaheenJ

singalsu, 2020-06-18:
@ZaheenJ Thanks for resolve, update and close of issue. Please continue reporting issues if things do not work with SOF!