thesofproject/sof #3058: [BUG] Internal Speakers and HDMI not working on Acer TravelMate P614 (Realtek ACL295)

stakaz, 2020-06-16:
**Describe the bug**
- The integrate speakers of the Laptop Acer TravelMate P614-51T-G2 do not produce any sound. 
- HDMI cannot be used to play sound
- headphone jack works as expected
- microphone works after tweaking but is a bit noisy

**To Reproduce**
Fresh install of Arch Linux and sof-firmware
I have added a sink and a source to `/etc/pulse/default.pa` (see below), the microphone works after this addition but the speakers still do not work.

**Reproduction Rate**
10/10

**Expected behavior**
Fully functioning sound-card behavior

**Impact**
no sound without headphones and no HDMI output

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.7.2.arch1-1
    * SOF: 1.5.1-1
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: Arch Linux

**Screenshots or console output**

`/etc/modprobe.d/blacklist.conf`
```
blacklist snd_hda_intel
blacklist snd_sok_skl
```
leaving this out does not change anything.

`dmesg | grep sof`
```
[    1.029329] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    1.029331] software IO TLB: mapped [mem 0x6937d000-0x6d37d000] (64MB)
[    2.896872] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.897000] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.897008] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.897151] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.897239] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.903514] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    2.964037] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    2.964039] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.052338] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:1-88707
[    3.052340] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:13:0
[    3.052341] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.052673] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[    3.052689] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[    3.114206] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:16:0 Kernel ABI 3:13:0
[    3.114207] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    3.114211] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.115066] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.115068] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.116175] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.116177] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.116947] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.116949] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.118077] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    3.118079] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    3.118080] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    3.118082] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    3.118083] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    3.118084] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    3.118859] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[... 446 lines cut ...]
### when no local session needs us anymore.
.ifexists module-console-kit.so
load-module module-console-kit
.endif
.ifexists module-systemd-login.so
load-module module-systemd-login
.endif

### Enable positioned event sounds
load-module module-position-event-sounds

### Cork music/video streams when a phone stream is active
load-module module-role-cork

### Modules to allow autoloading of filters (such as echo cancellation)
### on demand. module-filter-heuristics tries to determine what filters
### make sense, and module-filter-apply does the heavy-lifting of
### loading modules and rerouting streams.
load-module module-filter-heuristics
load-module module-filter-apply

### Make some devices default
#set-default-sink output
#set-default-source input
```

stakaz, 2020-06-16:
The speakers work well on Windows and shutting down windows without fast startup did not resolved the issue.

paulstelian97, 2020-06-16:
> The speakers work well on Windows and shutting down windows without fast startup did not resolved the issue.

You may try other combinations -- booting into Windows and then rebooting without a full shutdown into Linux. Fresh boot Linux without having booted Windows (try after a force-power-off, like holding down the button 4 seconds, that can also change hardware state).

stakaz, 2020-06-16:
Hello, unfortunately none of the windows/linux booting methods helped. Is is that common to have such issues? I thought it is much more likely to have a linux driver that does not work (hopefully yet).

paulstelian97, 2020-06-16:
Well the next thing to check is whether it's SOF or PulseAudio at fault. You should download some .wav file, say it is called file.wav and it is in your home folder.

Then, you'd try the command `aplay -Dhw:0,0 file.wav` or `aplay -Dhw:0,1 file.wav`. If either works, this is a problem in the UCM configuration. You may need to prepend `sudo` to these commands if you're not in the `audio` group.

Let us know if aplay does output any sound. Of course you have to close pulseaudio otherwise aplay will fail to actually do anything. `sudo killall pulseaudio`

stakaz, 2020-06-16:
Ok, for the sake of clarity I have uninstalled pulseaudio and depending packages. The total result is the same: headphones: yes, speaker no. However I was only able to play sound with 
```
mycomputer: ~ $ aplay  /usr/share/sounds/alsa/Front_Left.wav 
Playing WAVE '/usr/share/sounds/alsa/Front_Left.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```
while trying to set a hardware card ended in different errors
```
mycomputer: ~ $ aplay  -Dhw:0,0 /usr/share/sounds/alsa/Front_Center.wav 
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available
mycomputer: ~ $ aplay  -Dhw:0,1 /usr/share/sounds/alsa/Front_Center.wav 
aplay: main:830: audio open error: Invalid argument
mycomputer: ~ $ aplay  -Dhw:0,2 /usr/share/sounds/alsa/Front_Center.wav 
aplay: main:830: audio open error: No such file or directory
mycomputer: ~ $ aplay  -Dhw:0,3 /usr/share/sounds/alsa/Front_Center.wav 
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available
mycomputer: ~ $ aplay  -Dhw:0,4 /usr/share/sounds/alsa/Front_Center.wav 
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
aplay: set_params:1349: Channels count non available
```

Here are again some outputs
```
mycomputer: ~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
mycomputer: ~ $ dmesg | grep sof
[    1.020178] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    1.020180] software IO TLB: mapped [mem 0x6937d000-0x6d37d000] (64MB)
[    3.193677] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.193820] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.193831] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    3.193990] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.194100] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.200865] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    3.347962] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    3.347965] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.436076] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:1-88707
[    3.436077] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:13:0
[    3.436078] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.436385] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[    3.436402] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[    3.475326] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:16:0 Kernel ABI 3:13:0
[    3.475326] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    3.475330] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.476158] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.476160] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.477275] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.477277] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.478046] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.478048] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.479264] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[... 15 lines cut ...]
```

```
mycomputer: ~ $ dmesg | grep snd 
[    2.977503] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.977637] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.505095] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC295: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    3.505096] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.505097] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    3.505098] snd_hda_codec_realtek ehdaudio0D0:    mono: mono_out=0x0
[    3.505098] snd_hda_codec_realtek ehdaudio0D0:    inputs:
[    3.545251] snd_hda_codec_realtek ehdaudio0D0: ASoC: sink widget AIF1TX overwritten
[    3.545255] snd_hda_codec_realtek ehdaudio0D0: ASoC: source widget AIF1RX overwritten
[    3.545295] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> DMIC 6 mapping ok
[    3.545299] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> DMIC16kHz 7 mapping ok
[    3.545304] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDA Analog 0 mapping ok
[    3.545310] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDA Digital 1 mapping ok
[    3.545314] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI1 3 mapping ok
[    3.545318] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI2 4 mapping ok
[    3.545321] skl_hda_dsp_generic skl_hda_dsp_generic: snd-soc-dummy-dai <-> HDMI3 5 mapping ok
```
alsamixer
[image]

Once again I have commented out the blacklisted modules but the result remained the same.

stakaz, 2020-06-16:
What I thick could be important is this information:

```
[    3.505095] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC295: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    3.505096] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.505097] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
```
which suggests that there are no speaker outputs but the line_out is remapped to speakers.

lgirdwood, 2020-06-17:
@juimonen are we missing anything here from UCM or config ?

stakaz, 2020-06-18:
Hey there, just a short question: it is a very recent notebook and I bought it one week ago. So, fro what can be seen now, it is more likely that it will be fixed in the near future or is it more like "another not-working hardware in linux world"? Just an impression of how it feels I would say, it could be fixed soon but I am not an expert of course.

stakaz, 2020-06-18:
And here is a list with loaded modules

```
mycomputer: ~ $ lsmod
Module                  Size  Used by
ccm                    20480  3
fuse                  139264  3
rfcomm                 90112  16
hid_logitech_hidpp     45056  0
cmac                   16384  7
algif_hash             16384  3
algif_skcipher         16384  3
af_alg                 32768  14 algif_hash,algif_skcipher
hid_logitech_dj        28672  0
bnep                   28672  2
uvcvideo              114688  0
btusb                  65536  0
btrtl                  24576  1 btusb
videobuf2_vmalloc      20480  1 uvcvideo
videobuf2_memops       20480  1 videobuf2_vmalloc
btbcm                  20480  1 btusb
videobuf2_v4l2         28672  1 uvcvideo
btintel                32768  1 btusb
videobuf2_common       57344  2 videobuf2_v4l2,uvcvideo
rtsx_usb_sdmmc         32768  0
rtsx_usb_ms            24576  0
bluetooth             712704  43 btrtl,btintel,btbcm,bnep,btusb,rfcomm
mmc_core              188416  1 rtsx_usb_sdmmc
videodev              274432  3 videobuf2_v4l2,uvcvideo,videobuf2_common
memstick               24576  1 rtsx_usb_ms
ecdh_generic           16384  2 bluetooth
mc                     61440  4 videodev,videobuf2_v4l2,uvcvideo,videobuf2_common
rtsx_usb               28672  2 rtsx_usb_sdmmc,rtsx_usb_ms
usbhid                 65536  1 hid_logitech_dj
ecc                    36864  1 ecdh_generic
snd_soc_skl_hda_dsp    28672  2
snd_hda_codec_hdmi     73728  1
snd_soc_hdac_hdmi      40960  1 snd_soc_skl_hda_dsp
x86_pkg_temp_thermal    20480  0
intel_powerclamp       20480  0
snd_hda_codec_realtek   135168  1
joydev                 28672  0
mousedev               24576  0
coretemp               20480  0
iTCO_wdt               16384  0
snd_soc_dmic           16384  1
snd_hda_codec_generic    98304  1 snd_hda_codec_realtek
iTCO_vendor_support    16384  1 iTCO_wdt
mei_hdcp               24576  0
kvm_intel             327680  0
snd_sof_pci            24576  4
hid_multitouch         32768  0
snd_sof_intel_byt      20480  1 snd_sof_pci
snd_sof_intel_ipc      20480  1 snd_sof_intel_byt
hid_generic            16384  0
intel_rapl_msr         20480  0
snd_sof_intel_hda_common    90112  1 snd_sof_pci
kvm                   835584  1 kvm_intel
snd_soc_hdac_hda       24576  1 snd_sof_intel_hda_common
snd_sof_xtensa_dsp     16384  2 snd_sof_intel_hda_common,snd_sof_intel_byt
snd_sof_intel_hda      20480  1 snd_sof_intel_hda_common
irqbypass              16384  1 kvm
intel_wmi_thunderbolt    20480  0
acer_wmi               28672  0
wmi_bmof               16384  0
snd_sof               126976  4 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_byt,snd_sof_intel_ipc
snd_hda_codec         167936  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_codec_realtek,snd_soc_hdac_hda,snd_soc_skl_hda_dsp
crct10dif_pclmul       16384  1
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
[... 73 lines cut ...]
acer_wireless          20480  0
wmi                    36864  3 intel_wmi_thunderbolt,acer_wmi,wmi_bmof
intel_hid              24576  0
int3400_thermal        20480  0
acpi_thermal_rel       16384  1 int3400_thermal
sparse_keymap          16384  2 intel_hid,acer_wmi
drm                   581632  19 drm_kms_helper,i915
crypto_user            16384  0
agpgart                53248  2 intel_gtt,drm
ip_tables              36864  0
x_tables               53248  1 ip_tables
ext4                  798720  2
crc32c_generic         16384  0
crc16                  16384  2 bluetooth,ext4
mbcache                16384  1 ext4
jbd2                  139264  1 ext4
serio_raw              20480  0
atkbd                  36864  0
libps2                 20480  1 atkbd
xhci_pci               20480  0
crc32c_intel           24576  4
xhci_hcd              286720  1 xhci_pci
i8042                  32768  1 acer_wmi
serio                  28672  4 serio_raw,atkbd,i8042
```

juimonen, 2020-06-18:
@stakaz did I read it correctly that you manually configured the sinks in pulse config file? these hw0,0 and hw0,6? If you did so, we're there some issues pulseaudio autodetecting alsa outputs/inputs?

stakaz, 2020-06-18:
Yes, at least the microphone was not working without this setting. But the headphones are detected and I can use them without the sink setting. But it does not change the problem with no HDMI/no speakers output... 

THe difference is, when not explicitly set the sink I have a device where I can switch between Stereo and Multi-Channel settings but with the sink set, there only one fixed output possible.

juimonen, 2020-06-18:
@stakaz ok, it might be then as @lgirdwood hinted that there's no correct version of alsa-ucm in your setup (these are the config files that "glue" pulseaudio and sof together, so pulseaudio finds correct sinks&sources)...

If you check for example this, I think it was with arch linux:
https://github.com/thesofproject/sof/issues/3049

stakaz, 2020-06-18:
@juimonen Thanks, after installing the last version of alsa-ucm things became much better. I now see all the correct sinks and sources of my device. The microphone works, HDMI works and headphones work as well, but still no sound through the speakers.

```
mycomputer: ~ $ dmesg | grep sof
[    1.008724] PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
[    1.008726] software IO TLB: mapped [mem 0x6937d000-0x6d37d000] (64MB)
[    2.905026] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.905147] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.905158] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.905286] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    2.905371] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.916747] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    2.958617] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    2.958618] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    3.045650] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:5:1-88707
[    3.045652] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:16:0 Kernel ABI 3:13:0
[    3.045653] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[    3.045960] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 3 size 0x1c
[    3.045977] sof-audio-pci 0000:00:1f.3: warning: unknown ext header type 4 size 0x10
[    3.048256] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:16:0 Kernel ABI 3:13:0
[    3.048257] sof-audio-pci 0000:00:1f.3: warn: topology ABI is more recent than kernel
[    3.048259] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    3.049097] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    3.049098] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    3.050222] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    3.050223] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    3.050966] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    3.050967] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    3.051967] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    3.051968] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    3.051969] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    3.051970] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    3.051972] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    3.051973] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    3.052757] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    3.052758] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    3.053525] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    3.053526] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    3.054306] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    3.061759] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    3.126641] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input17
[    3.126679] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    3.126711] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    3.126743] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20
mycomputer: ~ $ dmesg | grep snd
[    2.249653] Modules linked in: i2c_i801 snd_timer snd soundcore thunderbolt(+) pcc_cpufreq(-) acpi_cpufreq(-) libarc4 iwlwifi i915 cfg80211 i2c_algo_bit drm_kms_helper mei_me mei cec rc_core intel_gtt intel_lpss_pci syscopyarea intel_lpss sysfillrect idma64 processor_thermal_device rfkill sysimgblt intel_rapl_common fb_sys_fops intel_soc_dts_iosf mac_hid i2c_hid hid battery evdev int3403_thermal ac int340x_thermal_zone acer_wireless wmi int3400_thermal intel_hid acpi_thermal_rel sparse_keymap drm crypto_user agpgart ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 serio_raw atkbd libps2 xhci_pci crc32c_intel xhci_hcd i8042 serio
[    2.393071] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.393204] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    2.550549] snd_soc_skl 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    2.550608] snd_soc_skl 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    3.076750] snd_hda_codec_realtek ehdaudio0D0: autoconfig for ALC295: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    3.076751] snd_hda_codec_realtek ehdaudio0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    3.076752] snd_hda_codec_realtek ehdaudio0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[... cut ...]
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

stakaz, 2020-06-18:
I do not need the blacklist anymore and no manual settings for pulseaudio (except disabling `load-module module-suspend-on-idle` to avoid noise in the idle state).

stakaz, 2020-06-18:
Anything I can test with `HDAJackRetask`?

[image]

stakaz, 2020-06-18:
It seams somehow that I cannot override anything through HDAJackRetask.

stakaz, 2020-06-18:
Could something in the `HiFi.conf` be wrong? This one seems to be used.

```
/usr/share/alsa/ucm2/hda-dsp/HiFi.conf
# Usecase for stereo playback Speaker and Headset, Recording on DMIC and Headset MIC.
# For Audio in HDA mode on Intel cAVS platforms

SectionVerb {

	EnableSequence [
		cset "name='mch_cap_out mo mch_cap_in mi Switch' 1"
	]

	Value {
		TQ "HiFi"
	}
}

SectionDevice."Headphones" {
	Comment "Headphones"

	ConflictingDevice [
		"Speaker"
	]

	EnableSequence [
		cset "name='Headphone Playback Switch' 1,1"
		cset "name='Speaker Playback Switch' 0,0"
	]

	Value {
		PlaybackPriority 200
		PlaybackPCM "hw:${CardId},7"
		PlaybackMixerElem "Headphone"
		PlaybackMasterElem "Master"
		PlaybackVolume "Headphone Playback Volume"
		PlaybackSwitch "Headphone Playback Switch"
		JackControl "Front Headphone Jack"
		JackHWMute "Speaker"
	}
}

SectionDevice."Speaker" {
	Comment "Speaker"

	ConflictingDevice [
		"Headphones"
	]

	EnableSequence [
		cset "name='Headphone Playback Switch' 0,0"
		cset "name='Speaker Playback Switch' 1,1"
	]

	Value {
		PlaybackPriority 100
		PlaybackPCM "hw:${CardId},7"
		PlaybackMixerElem "Speaker"
		PlaybackMasterElem "Master"
		PlaybackVolume "Speaker Playback Volume"
		PlaybackSwitch "Speaker Playback Switch"
	}
}

SectionDevice."Headset" {
	Comment "Headset Microphone"

	ConflictingDevice [
		"Mic"
	]

	EnableSequence [
		cset "name='media0_out mo codec0_in mi Switch' 1"
	]

	Value {
		CapturePriority 200
		CapturePCM "hw:${CardId},7"
	}
}

SectionDevice."Mic" {
	Comment "Digital Microphone"

	ConflictingDevice [
		"Headset"
	]

	EnableSequence [
		cset "name='mch_cap_out mo mch_cap_in mi Switch' 1"
	]

	Value {
		CapturePriority 100
		CapturePCM "hw:${CardId},13"
	}
}

<hda-dsp/Hdmi1.conf>
<hda-dsp/Hdmi2.conf>
```

However I wounder why the `7` is used for the `PlaybackPCM` because `aplay -l` gives

```
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

stakaz, 2020-06-18:
Oh no, probably it is this one:

```
# /usr/share/alsa/ucm2/sof-hda-dsp/HiFi.conf
# Usecase for stereo playback Speaker and Headset, Recording on DMIC and Headset MIC.
# For Audio in I2S mode on Intel SKYLAKE/KABYLAKE platforms

SectionVerb {
	EnableSequence [
		cset "name='media0_out mo dmic01_hifi_in mi Switch' 1"
	]

	Value {
		TQ "HiFi"
	}
}

SectionDevice."Headphones" {
	Comment "Headphones"

	ConflictingDevice [
		"Speaker"
	]

	EnableSequence [
		cset "name='HPO L Switch' 1"
		cset "name='HPO R Switch' 1"
		cset "name='Headphone Jack Switch' 1"
		cset "name='Speaker Playback Switch' 0,0"
	]

	Value {
		PlaybackPriority 200
		PlaybackPCM "hw:${CardId}"
		JackControl "Headphone Jack"
		JackHWMute "Speaker"
	}
}

SectionDevice."Speaker" {
	Comment "Speaker"

	ConflictingDevice [
		"Headphones"
	]

	EnableSequence [
		cset "name='SPO Switch' 1"
		cset "name='Speaker Playback Switch' 1,1"
		cset "name='Speaker Switch' 1"
		cset "name='HPO L Switch' 0"
		cset "name='HPO R Switch' 0"
	]

	Value {
		PlaybackPriority 100
		PlaybackPCM "hw:${CardId}"
	}
}

SectionDevice."Headset" {
	Comment "Headset Microphone"

	ConflictingDevice [
		"Mic"
	]
	EnableSequence [
		cset "name='media0_out mo codec0_in mi Switch' 1"
		cset "name='ADC0 Capture Switch' 1,1"
		cset "name='ADC 0 Mux' 0"
		cset "name='Mic Jack Switch' 1"
		cset "name='media0_out mo dmic01_hifi_in mi Switch' 0"
	]

	DisableSequence [
		cset "name='media0_out mo dmic01_hifi_in mi Switch' 1"
		cset "name='media0_out mo codec0_in mi Switch' 0"
	]
	Value {
		CapturePriority 200
		CapturePCM "hw:${CardId},1"
		JackControl "Mic Jack"
	}
}

SectionDevice."Mic" {
	Comment "Digital Stereo Microphone"

	ConflictingDevice [
		"Headset"
	]

	EnableSequence [
		cset "name='media0_out mo dmic01_hifi_in mi Switch' 1"
		cset "name='media0_out mo codec0_in mi Switch' 0"
		cset "name='ADC0 Capture Switch' 0,0"
		cset "name='Mic Jack Switch' 0"
		cset "name='ADC 0 Mux' 0"
		
	]

	DisableSequence [
		cset "name='media0_out mo codec0_in mi Switch' 1"
		cset "name='media0_out mo dmic01_hifi_in mi Switch' 0"
		cset "name='ADC0 Capture Switch' 1,1"
		cset "name='Mic Jack Switch' 1"
		cset "name='ADC 0 Mux' 2"
	]
	Value {
		CapturePriority 100
		CapturePCM "hw:${CardId},1"
	}
}

<skylake-rt286/Hdmi1.conf>
<skylake-rt286/Hdmi2.conf>
```

stakaz, 2020-06-19:
Could #1899 this be somehow related? Because I have a Kaby Lake processor `i7-10510U`. I wanted to compile sof-fimrware from master branch but as I see now, it is not possible, is it? 

One repetitive remark: I have tested the sound on a system completely without pulseaudio and it has the same issue, so it is likely **not** the pulseaudio which causes the fault.

paulstelian97, 2020-06-19:
> Could #1899 this be somehow related? Because I have a Kaby Lake processor `i7-10510U`. I wanted to compile sof-fimrware from master branch but as I see now, it is not possible, is it?
>
> One [...]

That CPU is apparently Comet Lake. I for one am not sure about the state of implementation for CML. It seems to use the Skylake implementation on your platform but I'm not sure whether it's correct. I'll leave someone from Intel comment on this.

stakaz, 2020-06-19:
> That CPU is apparently Comet Lake, not Kaby Lake.

Sorry, to hasty copy/paste ;)

> It seems to use the Skylake implementation on your platform but I'm not sure whether it's correct. I'll leave someone from Intel comment on this.

Just in case, could this be the reason for such a "no speaker" behavior? After all, everything else is working properly...

paulstelian97, 2020-06-19:
I don't know if the Skylake implementation is supposed to work on CML or there needs to be separate support. Maybe there isn't (again, I will let someone from Intel to say for sure, I'm working on the ARM-based i.MX platforms so I don't know how the Intel side works exactly). If the implementation is incompatible this might lead to behaviors like not loading at all or not connecting to the speakers.

Do note that you also need a driver for the speaker itself, SOF just reuses the existing HDA driver for this.

@singalsu Do we have proper topologies available for this platform? It does seem to create the devices but I can't guarantee it creates them properly.

Also, @stakaz It might be useful for the purpose of troubleshooting to get firmware logs, using the sof-logger tool. It's in the repository in tools/logger. It might need root in order to extract those logs. Maybe there is a silent error that causes the errors in aplay (or even non-silent). The kernel cannot intercept firmware errors to display them in dmesg.

plbossart, 2020-06-19:
@stakaz let's step back a bit here.

To debug playback on speakers, the first order of business is to test with the 'legacy' driver. Please use 'blacklist snd-pci-sof', remove 'blacklist snd-hda-intel' and force the snd-hda-intel driver to be probed with 'options snd-intel-dspcfg dsp_driver=1'

Then try to play with the legacy driver. If you don't get the expected sound, then it's a codec configuration issue that will happen whether you use SOF or not. If you do get sound, then it's a problem with SOF not doing the same sequence as legacy.

Can you please try with the legacy driver?
Thanks!

stakaz, 2020-06-19:
Unfortunately (fortunately for sofproject ;) ) it does **not** work with legacy driver.

I have also tested all possible pin reconnections through hdajackretask but nothing worked.

Here some output. Could you help me with where to ask about the codec problem now? 

Here is some output again
```
mycomputer: ~ $ dmesg | grep snd
[    2.386319] snd_hda_intel 0000:00:1f.3: enabling device (0000 -> 0002)
[    2.399166] Modules linked in: snd_compress ac97_bus snd_pcm_dmaengine snd_hda_intel snd_intel_dspcfg snd_hda_codec input_leds mac80211 pcspkr fjes(-) snd_hda_core snd_hwdep snd_pcm ofpart libarc4 cmdlinepart snd_timer intel_spi_pci intel_spi snd spi_nor i915 mtd soundcore i2c_i801 thunderbolt e1000e(+) iwlwifi mei_me i2c_algo_bit mei cfg80211 drm_kms_helper cec intel_lpss_pci rc_core intel_lpss intel_gtt idma64 syscopyarea processor_thermal_device sysfillrect i2c_hid intel_rapl_common sysimgblt rfkill fb_sys_fops intel_soc_dts_iosf hid battery int3403_thermal int340x_thermal_zone ac acer_wireless evdev mac_hid wmi intel_hid int3400_thermal sparse_keymap acpi_thermal_rel drm crypto_user agpgart ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 serio_raw atkbd libps2 xhci_pci crc32c_intel xhci_hcd i8042 serio
[    2.503456] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    2.569574] snd_hda_codec_realtek hdaudioC0D0: autoconfig for ALC295: line_outs=1 (0x17/0x0/0x0/0x0/0x0) type:speaker
[    2.569576] snd_hda_codec_realtek hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    2.569578] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=1 (0x21/0x0/0x0/0x0/0x0)
[    2.569579] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[    2.569580] snd_hda_codec_realtek hdaudioC0D0:    inputs:
```

```
mycomputer: ~ $ pacmd list-sinks 
1 sink(s) available.
  * index: 0
        name: <alsa_output.pci-0000_00_1f.3.analog-stereo>
        driver: <module-alsa-card.c>
        flags: HARDWARE HW_MUTE_CTRL HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY DYNAMIC_LATENCY
        state: IDLE
        suspend cause: (none)
        priority: 9039
        volume: front-left: 65536 / 100% / 0.00 dB,   front-right: 65536 / 100% / 0.00 dB
                balance 0.00
        base volume: 65536 / 100% / 0.00 dB
        volume steps: 65537
        muted: no
        current latency: 1837.69 ms
        max request: 344 KiB
        max rewind: 344 KiB
        monitor source: 0
        sample spec: s16le 2ch 48000Hz
        channel map: front-left,front-right
                     Stereo
        used by: 0
        linked by: 0
        configured latency: 1837.50 ms; range is 0.50 .. 1837.50 ms
        card: 0 <alsa_card.pci-0000_00_1f.3>
        module: 6
        properties:
                alsa.resolution_bits = "16"
                device.api = "alsa"
                device.class = "sound"
                alsa.class = "generic"
                alsa.subclass = "generic-mix"
                alsa.name = "ALC295 Analog"
                alsa.id = "ALC295 Analog"
                alsa.subdevice = "0"
                alsa.subdevice_name = "subdevice #0"
                alsa.device = "0"
                alsa.card = "0"
                alsa.card_name = "HDA Intel PCH"
                alsa.long_card_name = "HDA Intel PCH at 0xa323c000 irq 168"
                alsa.driver_name = "snd_hda_intel"
                device.bus_path = "pci-0000:00:1f.3"
                sysfs.path = "/devices/pci0000:00/0000:00:1f.3/sound/card0"
                device.bus = "pci"
                device.vendor.id = "8086"
                device.vendor.name = "Intel Corporation"
                device.product.id = "02c8"
                device.form_factor = "internal"
                device.string = "front:0"
                device.buffering.buffer_size = "352800"
                device.buffering.fragment_size = "176400"
                device.access_mode = "mmap+timer"
                device.profile.name = "analog-stereo"
                device.profile.description = "Analog Stereo"
                device.description = "Built-in Audio Analog Stereo"
                alsa.mixer_name = "Realtek ALC295"
                alsa.components = "HDA:10ec0295,1025141d,00100002 HDA:8086280b,80860101,00100000"
                module-udev-detect.discovered = "1"
                device.icon_name = "audio-card-pci"
        ports:
                analog-output-speaker: Speakers (priority 10000, latency offset 0 usec, available: unknown)
                        properties:
                                device.icon_name = "audio-speakers"
                analog-output-headphones: Headphones (priority 9900, latency offset 0 usec, available: no)
                        properties:
                                device.icon_name = "audio-headphones"
        active port: <analog-output-speaker>
```

plbossart, 2020-06-19:
@stakaz yes it's disappointing but at the same time it's progress if you've identified a codec configuration issue. it's not unusual, it's just back luck that it happened on your device.

Since the SOF team has limited expertise on the pin retasking, I would report this issue on the alsa-devel mailing list, cc: Takashi Iwai <tiwai@suse.de> with the subject [HDAudio][ALC295] speakers not working on Acer Travelmate P614

https://mailman.alsa-project.org/mailman/listinfo/alsa-devel

Be sure to include a link to the alsa-info upload for reference. You are already running a very recent kernel but you may want to give v5.8-rc1 a try, in case this is fixed already? Usually when you show that you have tried the latest code it helps get more attention.

stakaz, 2020-06-19:
Many thanks for the quick reply.

>codec configuration issue. it's not unusual, it's just back luck that it happened on your device.

From what you know, is such an issue usually get fixed at some point? 

Shell I compile the v5.8-rc1 kernel or can I get from it somewhere through arch repositories?

> Since the SOF team has limited expertise on the pin retasking, I would report this issue on the alsa-devel mailing list, cc: Takashi Iwai tiwai@suse.de with the subject [HDAudio][ALC295] speakers not working on Acer Travelmate P614

I will do it after the kernel try.

> Be sure to include a link to the alsa-info upload for reference. 

What do you mean by that?

Probably you can close this issue now? Or do you have some more hints for me?

plbossart, 2020-06-19:
You're welcome @stakaz 

Fixing the issue is just like other Linux issues, maintainers and domain experts will need to understand what happens and ask you to test patches. So it's useful for you to compile the kernel yourself in preparation for feedback and asks for tests.

alsa-info is a script that will upload the info on your device to the alsa-project site. You can use the reference to the uploaded file for everyone to see the data.

You can get this issue open in case you need guidance or feedback. When you feel you can use the alsa-devel mailing list you can close. Best of luck!

slawblauciak, 2020-06-23:
Closing, not related to the SOF firmware or the driver.

closed 2020-06-23 by slawblauciak