thesofproject/sof #3059: [FEATURE] Enhance build_tools.sh

dbaluta, 2020-06-16:
**Is your feature request related to a problem? Please describe.**

1) Add -h option to sof/scripts/build-tools.sh so that running ./scripts/build-tools.sh -h will call print_usage function.

2) Refactor build-tools.sh options so that for example when called with "-f" the script will only build the fuzzer not all the toools

**Describe alternatives you've considered**
I don't really like when using "-f" with build-tools.sh to compile the topologies for example.

**Additional context**

build-tools.sh script can be found in ./scripts/build-tools.sh

dbaluta, 2020-06-16:
Cc: @crazoes care to have a look at this task?

paulstelian97, 2020-06-16:
If you want -f to only build the tools, you will need a separate switch for building the topologies and I don't recall there being any. So you say one should be introduced for this as well?

dbaluta, 2020-06-16:
@paulstelian97 Yes, that's true

crazoes, 2020-06-17:
Topologies are getting built as part of building tools which happen by default for building fuzzer and test topologies. Should only topology not be built? Should we skip building tools(probes, logger, ctl, topology, test) altogether if not specified by let's say `-T`?

dbaluta, 2020-06-17:
@crazoes in my opinion script should compile everything without parameters and then add a parameter for each tool: (probes, logger, ctl, topology, test, fuzzer, etc).

closed 2020-08-04 by dbaluta