thesofproject/sof #3063: [BUG][CI][CHT] Build failure on CHT: .text will not fit in region sof_text_start

aiChaoSONG, 2020-06-18:
[100%] Linking C executable sof
/home/test/xcc/install/tools/RD-2012.5-linux/XtensaTools/bin/xt-ld: sof section `.text' will not fit in region `sof_text_start'
CMakeFiles/sof.dir/build.make:1820: recipe for target 'sof' failed
make[3]: *** [sof] Error 2
CMakeFiles/Makefile2:396: recipe for target 'CMakeFiles/sof.dir/all' failed
make[2]: *** [CMakeFiles/sof.dir/all] Error 2
CMakeFiles/Makefile2:1658: recipe for target 'src/arch/xtensa/CMakeFiles/bin.dir/rule' failed
make[1]: *** [src/arch/xtensa/CMakeFiles/bin.dir/rule] Error 2
Makefile:723: recipe for target 'bin' failed
make: *** [bin] Error 2

ClarexZhou, 2020-06-18:
Bi-sect result: issue occurred from https://github.com/thesofproject/sof/commit/a2d1e1efdc554becec70a6c1d95491bac59800ef

ClarexZhou, 2020-06-18:
@bkokoszx  could you please help to check first?

xiulipan, 2020-06-22:
The TEXT size on CHT is just a little out range.
After hack with memory.h, we can got the current TEXT size.
```
writing module 0 sof-cht

        Totals  Start           End             Size
        TEXT    0xff2c0000      0xff2d4047      0x14047 (IRAM_SIZE is 0x14000)
        DATA    0xff300010      0xff328000      0x27ff0
        BSS     0xff303920      0xff307788      0x3e68
```

marc-hb, 2020-06-24:
> Bi-sect result: issue occurred from a2d1e1e

Yet everything is green in PR #2897  ?!

@xiulipan do you think this went over the limit only when PR #2897 was rebased and combined on top of some other code addition?

xiulipan, 2020-06-29:
@marc-hb Yes, this is the issue with merge.
When #2897 run with test, it would be on code base A.
But when it is merged, it would be on code base B. I manually test with base A and B and the issue is there.

That would be an ideal solution, but the question here is you will never know what is the point for the rebase. The rebase happen when you click the merge button. At that time, the PR is merged into the master branch.

We indeed have some after merge check. In both Jenkins and Travis. But we only have feedback from Travis and internal warning message in Jenkins. Will try to make the Jenkins push check expose to GitHub.

marc-hb, 2020-07-15:
Was this fixed by #3080 ? If yes please close.

aiChaoSONG, 2020-07-15:
colse for fixed by #3080

closed 2020-07-15 by lgirdwood